norlab-ulaval/libpointmatcher #61: Linking issue

simone-fontana, 2015-01-16:
I'am trying to compile my own software that uses libpointmatcher. However, when linking, I get the following error I couldn't solve in any way:
```
Linking CXX executable /home/simone/catkin_ws/devel/lib/dem_simulator/dem_simulator_test
/home/simone/libpointmatcher/build/libpointmatcher.so: undefined reference to `boost::filesystem::detail::initial_path(boost::system::error_code*)'
/home/simone/libpointmatcher/build/libpointmatcher.so: undefined reference to `boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*)'
/home/simone/libpointmatcher/build/libpointmatcher.so: undefined reference to `boost::filesystem::path::extension() const'
/home/simone/libpointmatcher/build/libpointmatcher.so: undefined reference to `boost::filesystem::absolute(boost::filesystem::path const&, boost::filesystem::path const&)'
/home/simone/libpointmatcher/build/libpointmatcher.so: undefined reference to `boost::filesystem::path::parent_path() const'
/home/simone/libpointmatcher/build/libpointmatcher.so: undefined reference to `boost::filesystem::path::operator/=(boost::filesystem::path const&)'
collect2: error: ld returned 1 exit status
```
I am using the following CMakeLists.txt: 

```
cmake_minimum_required (VERSION 2.8)
project(dem_simulator)

find_package(catkin_simple REQUIRED)
find_package(Boost COMPONENTS system filesystem REQUIRED)
find_package(libpointmatcher REQUIRED)

catkin_simple(ALL_DEPS_REQUIRED)

add_definitions(--std=c++11)

include_directories(${CMAKE_CURRENT_BINARY_DIR} ${Boost_INCLUDE_DIRS} ${libpointmatcher_INCLUDE_DIRS})

cs_add_library(${PROJECT_NAME} src/dem_simulator.cc)
cs_add_executable(dem_simulator_test src/main.cc)

target_link_libraries(dem_simulator_test ${PROJECT_NAME} ${catkin_LIBRARIES} ${libpointmatcher_LIBRARIES} ${Boost_LIBRARIES})

cs_install()
cs_export()
```

pomerlef, 2015-01-16:
which version of boost are you using?

simone-fontana, 2015-01-16:
I am using libboost 1.54

tsandy, 2015-01-19:
Just FYI. I also am having this problem. Also with Boost 1.54. If I can help at all from ETH, let me know :)

pomerlef, 2015-02-19:
ok, I'm investigating this issue now. It seems that boost is not link to the library but only to the example applications, which make the all thing work within our build hierarchy but most probably not for external projects. It seems that drhalftone was reporting the same problem.

What is the output of  `${libpointmatcher_LIBRARIES}` on your side?

In your CMakeLists.txt, you can add:
```
message("libpointmatcher_LIBRARIES= ${libpointmatcher_LIBRARIES}")
```

tsandy, 2015-02-20:
Hi Francois,
I've already solved the problem on my machine. Sorry I forgot to send a
message with the fix. If I remember correctly, all I had to do was add a
link to the Boost libraries for 'pointmatcher'.
Line 215 of CMakeLists: target_link_libraries(pointmatcher
${yaml-cpp_LIBRARIES} ${libnabo_LIBRARIES} ${Boost_LIBRARIES})

Hopefully this works for everyone.

Best,
Tim

pomerlef, 2015-02-20:
@simone-fontana the last commit 48b1a61ddd1d1c19ea6440b6d401c373fc3e9a39 should fix your problem. I also add a small `CMakeFile.txt` inspired from your example along with an executable to make sure that all information is properly included using `find_package()`. If you could try on your side, it would be great to confirm everything before I close this issue.

The example project is in `/examples/demo_external_link/`.

@tsandy thanks for the tip. I cleaned the `CMakeList.txt` to track the external libraries more easily. There was also some difference whether the library was static or dynamic. If you could update and confirm that it cover your case too, it would be great!

Also, if everything work on your side, could you give me your system spec. It seems that you're on a more recent system than me and would like to add that in the table on the top of [tutorial page](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/Compilation.md). All the commands to quickly write the versions are at the end of that page.

Cheers!

drhalftone, 2015-02-21:
I'm still having problems.  Here is the error message:

/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -headerpad_max_install_names -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk -mmacosx-version-min=10.7 -Wl,-rpath,/Qt/5.4/clang_64/lib -o LAUPointMatcher.app/Contents/MacOS/LAUPointMatcher main.o laupointmatcherwidget.o moc_laupointmatcherwidget.o   -F/Qt/5.4/clang_64/lib -L/usr/local/lib/ -lboost_thread-mt -lboost_filesystem-mt -lboost_system-mt -lboost_program_options-mt -lboost_date_time-mt -lboost_chrono-mt -framework QtWidgets -framework QtGui -framework QtCore -framework DiskArbitration -framework IOKit -framework OpenGL -framework AGL 
Undefined symbols for architecture x86_64:
  "PointMatcher<float>::DataPoints::Label::Label(std::string const&, unsigned long)", referenced from:
      LAUPointMatcherWidget::LAUPointMatcherWidget(QWidget*) in laupointmatcherwidget.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [LAUPointMatcher.app/Contents/MacOS/LAUPointMatcher] Error 1
22:54:26: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project LAUPointMatcher (kit: Desktop Qt 5.4.0 clang 64bit)
When executing step "Make"

Here is the Qt Project file that I'm using:

QT       += core gui widgets

TARGET    = LAUPointMatcher
TEMPLATE  = app

SOURCES  += main.cpp\
            laupointmatcherwidget.cpp

HEADERS  += laupointmatcherwidget.h

INCLUDEPATH += /Users/dllau/Libraries/libpointmatcher/pointmatcher \
               /usr/local/include/eigen3/ \
               /usr/local/include/

LIBS     += -L/usr/local/lib/ -lboost_thread-mt -lboost_filesystem-mt -lboost_system-mt -lboost_program_options-mt -lboost_date_time-mt -lboost_chrono-mt
            /Users/dllau/Libraries/libpointmatcher/build/libpointmatcher.a

Here is the header for the object that is triggering the undefined symbols:

#include <QWidget>
#include "PointMatcher.h"

class LAUPointMatcherWidget : public QWidget
{
    Q_OBJECT

public:
    LAUPointMatcherWidget(QWidget *parent = 0);
    ~LAUPointMatcherWidget();

private:
    PointMatcher<float>::DataPoints::Label label;
};

closed 2015-02-25 by pomerlef