norlab-ulaval/libpointmatcher #62: Undefined symbols for architecture x86_64:

drhalftone, 2015-01-24:
I'm trying to compile in Mac OS X 10.10 against Boost 1.57, which I installed using homebrew. Here is the relevant part of the compiler log:

Linking CXX shared library libpointmatcher.dylib
Undefined symbols for architecture x86_64:
  "boost::filesystem::path::operator/=(boost::filesystem::path const&)", referenced from:
      PointMatcherIO<float>::FileInfoVector::localToGlobalFileName(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in IO.cpp.o
      PointMatcherIO<double>::FileInfoVector::localToGlobalFileName(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in IO.cpp.o
  "boost::filesystem::detail::initial_path(boost::system::error_code*)", referenced from:
      PointMatcherSupport::validateFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in IO.cpp.o
  "boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*)", referenced from:
      PointMatcherIO<float>::FileInfoVector::localToGlobalFileName(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in IO.cpp.o
      PointMatcherIO<double>::FileInfoVector::localToGlobalFileName(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in IO.cpp.o
  "boost::filesystem::absolute(boost::filesystem::path const&, boost::filesystem::path const&)", referenced from:
      PointMatcherSupport::validateFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in IO.cpp.o
  "boost::system::system_category()", referenced from:
      boost::mutex::unlock() in Logger.cpp.o
      boost::unique_lock<boost::mutex>::lock() in Logger.cpp.o
      boost::mutex::lock() in Logger.cpp.o
      boost::mutex::mutex() in Logger.cpp.o
      __GLOBAL__I_a in Logger.cpp.o
      __GLOBAL__I_a in Exceptions.cpp.o
      __GLOBAL__I_a in DataPoints.cpp.o
      ...
  "boost::system::generic_category()", referenced from:
      __GLOBAL__I_a in Logger.cpp.o
      __GLOBAL__I_a in Exceptions.cpp.o
      __GLOBAL__I_a in DataPoints.cpp.o
      __GLOBAL__I_a in Matches.cpp.o
      __GLOBAL__I_a in ICP.cpp.o
      __GLOBAL__I_a in Registry.cpp.o
      __GLOBAL__I_a in Registrar.cpp.o
      ...
  "boost::filesystem::path::parent_path() const", referenced from:
      PointMatcherIO<float>::FileInfoVector::FileInfoVector(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in IO.cpp.o
      PointMatcherIO<double>::FileInfoVector::FileInfoVector(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in IO.cpp.o
  "boost::filesystem::path::extension() const", referenced from:
      PointMatcher<float>::DataPoints::load(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in IO.cpp.o
      PointMatcher<double>::DataPoints::load(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in IO.cpp.o
      PointMatcher<float>::DataPoints::save(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const in IO.cpp.o
      PointMatcher<double>::DataPoints::save(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const in IO.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libpointmatcher.1.2.1.dylib] Error 1
make[1]: *** [CMakeFiles/pointmatcher.dir/all] Error 2
make: *** [all] Error 2
22:21:13: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project libpointmatcher (kit: Desktop Qt 5.4.0 clang 64bit)
When executing step "Make"
22:21:13: Elapsed time: 00:08.

drhalftone, 2015-01-29:
I think I know what the problem is, but don't know how to fix it.  I think CMake is not using the Clang++ compiler provided by Xcode 6.1.1.  Anyone know how to do this?

drhalftone, 2015-01-29:
Okay, I found a solution, and as a person who really hates CMake, I'm not surprised its as convoluted a process as it is.  I completely erased the libpointmatcher source code and re-downloaded it fresh.  I then added "-G Xcode" to the CMake command to create an Xcode project.  Here, I could actually see that there were no libraries being linked.  So I aded a, "Link Binary with Libraries," phase under, "Build Phases."  I had to add it because it wasn't already there.  I then manually added all of my boost libraries from their brew directory.  I then compiled the code without issue.

samlcharreyron, 2015-01-29:
Where are your Boost 1.57 libraries located? Homebrew should symlink them into /usr/local/ which would be picked up by CMake but sometimes that isn't the case. You can fix links with:
    
    brew <link/unlink> boost

drhalftone, 2015-01-29:
Cmake says it detected boost in the brew locations. I found another git project complaining about Cmake and xcode 6.1.1 issue with Cmake not detecting the latest clang compiler which lead me to my conclusions.

drhalftone, 2015-01-29:
I will try this and get back to you shortly.

drhalftone, 2015-01-29:
I unlinked and then re-linked boost to no avail.

drhalftone, 2015-02-10:
Found the error in the CMakeLists.txt file for the entire project.  The issue was fixed with line 215:

target_link_libraries(pointmatcher ${yaml-cpp_LIBRARIES} ${libnabo_LIBRARIES})

Notice that it is missing the Boost libraries.  So I modified the line to read:

target_link_libraries(pointmatcher ${yaml-cpp_LIBRARIES} ${libnabo_LIBRARIES} ${Boost_LIBRARIES})

And now its creating the dylib files.  Now the problem happens at 100% completion with the following error when compiling the unit tests:

Apple LLVM version 6.0 (clang-600.0.56) (based on LLVM 3.5svn)
Target: x86_64-apple-darwin14.1.0
Thread model: posix
 "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld" -demangle -dynamic -arch x86_64 -macosx_version_min 10.10.0 -o utest -search_paths_first -headerpad_max_install_names CMakeFiles/utest.dir/utest.cpp.o CMakeFiles/utest.dir/ui/IO.cpp.o CMakeFiles/utest.dir/ui/DataFilters.cpp.o CMakeFiles/utest.dir/ui/Matcher.cpp.o CMakeFiles/utest.dir/ui/Outliers.cpp.o CMakeFiles/utest.dir/ui/ErrorMinimizers.cpp.o CMakeFiles/utest.dir/ui/Transformations.cpp.o CMakeFiles/utest.dir/ui/DataPoints.cpp.o CMakeFiles/utest.dir/ui/Inspectors.cpp.o CMakeFiles/utest.dir/ui/Loggers.cpp.o ../contrib/libgtest.a ../libpointmatcher.1.2.1.dylib -framework OpenCL /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_program_options-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_chrono-mt.dylib ../contrib/yaml-cpp-pm/libyaml-cpp-pm.a /Users/dllau/Libraries/libnabo/build/libnabo.a /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_program_options-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_chrono-mt.dylib -lc++ -lSystem /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../lib/clang/6.0/lib/darwin/libclang_rt.osx.a
Undefined symbols for architecture x86_64:
  "PointMatcherIO<float>::loadPLY(std::__1::basic_istream<char, std::__1::char_traits<char> >&)", referenced from:
      IOTest_loadPLY_Test::TestBody() in IO.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [utest/utest] Error 1
make[1]: *** [utest/CMakeFiles/utest.dir/all] Error 2
make: *** [all] Error 2

closed 2015-02-11 by drhalftone