norlab-ulaval/libpointmatcher #63: libpointmatcher.dylib doesn't seem to include symbol found in PointMatcher.h

drhalftone, 2015-02-10:
When I compile the code inside of Xcode, which I created using CMake.  Everything seems to work fine, but when I try to use the libpointmatcher.dylib in Qt using QMake, I get the following error:

Undefined symbols for architecture x86_64:
  "PointMatcher<float>::DataPoints::Label::Label(std::string const&, unsigned long)", referenced from:
      LAUPointMatcherObject::LAUPointMatcherObject(LAUScan, LAUScan, QObject*) in laupointmatcherobject.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Is there something in the CMake set up that means I can't use libpointmatcher with any other build system?

drhalftone, 2015-02-10:
Here is the complete compiler listing:

12:24:04: Running steps for project LAUScoliosis...
12:24:04: Configuration unchanged, skipping qmake step.
12:24:04: Starting: "/usr/bin/make" -j16
/Qt/5.4/clang_64/bin/qmake -spec macx-clang CONFIG+=debug CONFIG+=x86_64 -o Makefile ../LAUScoliosis/LAUScoliosis.pro
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -headerpad_max_install_names -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk -mmacosx-version-min=10.7 -Wl,-rpath,/Qt/5.4/clang_64/lib -o LAUScoliosis.app/Contents/MacOS/LAUScoliosis main.o lauscan.o lauscanmergedialog.o lauiterativeclosestpointobject.o laulaplaciansmootherobject.o laupointmatcherobject.o laufiducialwidget.o qrc_LAUScoliosisWidget.o qrc_shaders.o qrc_lauscan.o moc_lauscan.o moc_lauscanmergedialog.o moc_lauiterativeclosestpointobject.o moc_laulaplaciansmootherobject.o moc_laupointmatcherobject.o moc_laufiducialwidget.o   -F/Qt/5.4/clang_64/lib -L/usr/local/lib -ltiff -lboost_container-mt -lboost_system-mt /Users/dllau/Libraries/libpointmatcher/build/libpointmatcher.a -framework QtOpenGL -framework QtWidgets -framework QtGui -framework QtCore -framework DiskArbitration -framework IOKit -framework OpenGL -framework AGL 
Undefined symbols for architecture x86_64:
  "PointMatcher<float>::DataPoints::Label::Label(std::string const&, unsigned long)", referenced from:
      LAUPointMatcherObject::LAUPointMatcherObject(LAUScan, LAUScan, QObject*) in laupointmatcherobject.o
  "std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::find(char, unsigned long) const", referenced from:
      boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char> >::parse(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in libpointmatcher.a(DataPoints.cpp.o)
      int boost::io::detail::upper_bound_from_fstring<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::ctype<char> >(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::value_type, std::__1::ctype<char> const&, unsigned char) in libpointmatcher.a(DataPoints.cpp.o)
  "std::__1::__vector_base_common<true>::__throw_length_error() const", referenced from:
      std::__1::vector<PointMatcher<float>::DataPoints::Label, std::__1::allocator<PointMatcher<float>::DataPoints::Label> >::vector(unsigned long, PointMatcher<float>::DataPoints::Label const&) in libpointmatcher.a(DataPoints.cpp.o)
      PointMatcher<float>::DataPoints::allocateFields(PointMatcher<float>::DataPoints::Labels const&, PointMatcher<float>::DataPoints::Labels&, Eigen::Matrix<float, -1, -1, 0, -1, -1>&) const in libpointmatcher.a(DataPoints.cpp.o)
      std::__1::vector<PointMatcher<double>::DataPoints::Label, std::__1::allocator<PointMatcher<double>::DataPoints::Label> >::vector(unsigned long, PointMatcher<double>::DataPoints::Label const&) in libpointmatcher.a(DataPoints.cpp.o)
      PointMatcher<double>::DataPoints::allocateFields(PointMatcher<double>::DataPoints::Labels const&, PointMatcher<double>::DataPoints::Labels&, Eigen::Matrix<double, -1, -1, 0, -1, -1>&) const in libpointmatcher.a(DataPoints.cpp.o)
      void std::__1::vector<PointMatcher<double>::DataPoints::Label, std::__1::allocator<PointMatcher<double>::DataPoints::Label> >::__push_back_slow_path<PointMatcher<double>::DataPoints::Label const>(PointMatcher<double>::DataPoints::Label const&) in libpointmatcher.a(DataPoints.cpp.o)
      void std::__1::vector<PointMatcher<float>::DataPoints::Label, std::__1::allocator<PointMatcher<float>::DataPoints::Label> >::__push_back_slow_path<PointMatcher<float>::DataPoints::Label const>(PointMatcher<float>::DataPoints::Label const&) in libpointmatcher.a(DataPoints.cpp.o)
      std::__1::vector<boost::io::detail::format_item<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<boost::io::detail::format_item<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >::__append(unsigned long, boost::io::detail::format_item<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in libpointmatcher.a(DataPoints.cpp.o)
      ...
  "std::__1::__basic_string_common<true>::__throw_length_error() const", referenced from:
      std::__1::basic_stringbuf<char, std::__1::char_traits<char>, std::__1::allocator<char> >::str() const in libpointmatcher.a(DataPoints.cpp.o)
  "std::__1::locale::use_facet(std::__1::locale::id&) const", referenced from:
      void boost::io::detail::put<char, std::__1::char_traits<char>, std::__1::allocator<char>, boost::io::detail::put_holder<char, std::__1::char_traits<char> > const&>(boost::io::detail::put_holder<char, std::__1::char_traits<char> > const&, boost::io::detail::format_item<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char> >::string_type&, boost::basic_format<char, std::__1::char_traits<char>, std::__1::allocator<char> >::internal_streambuf_t&, std::__1::locale*) in libpointmatcher.a(DataPoints.cpp.o)
[... cut ...]
      construction vtable for std::__1::basic_iostream<char, std::__1::char_traits<char> >-in-std::__1::basic_stringstream<char, std::__1::char_traits<char>, std::__1::allocator<char> > in libpointmatcher.a(DataPoints.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [LAUScoliosis.app/Contents/MacOS/LAUScoliosis] Error 1
12:24:06: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project LAUScoliosis (kit: Desktop Qt 5.4.0 clang 64bit)
When executing step "Make"
12:24:06: Elapsed time: 00:02.

pomerlef, 2015-02-10:
ok, it seems that there is a list of problem using libpointmatcher under MacOS environment. I've access to a Mac with the OS 10.10.2 (14C109). I never developed on it but I can give it a try for a day to see how far I go. It should happen during the week. Meanwhile, if you have any other details that would help me, don't hesitate to post them.

drhalftone, 2015-02-10:
Searching online, the only hint I can seem to find is perhaps cmake is linking to the wrong std library, either libc++ or libstdc++.  From what I know, this has been an issue for a while since Apple made the change from one to the other.

drhalftone, 2015-02-10:
That last comment looks like a dead end. 

You might have noticed that I fixed the issue from the previous post.  I just had to edit the CMAKE file, line 215, to explicitly include the boost libraries.  But even with the correctly built dynamic library, the unit test is failing to build. And I still have the issue described above using the correctly built library.  But at least now, I'm seeing this error universally across CMAKE and unix make files, CMAKE with Xcode project, and Qt using Qmake to link the library.  So we can probably eliminate any issue that is compiler system dependent.

pomerlef, 2015-02-16:
Hi,

I compile successfully the library today with CMake and could run all the unit tests. 

To work, the library need to be build as a static one (i.e., libpointmatcher.a). There seems to be a difference between dynamic libraries and bundles on MacOS for which I'm not quite familiar with. Through me reading, it seems that using static link is the easiest way to keep it simple.

Could you try this on your side? When using CMake to generate the makefile, use the following line:
```
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DSHARED_LIBS=OFF ${SRC_DIR}
```

This will force the creation of a static library. You can confirm that everything is working using:

```
utest/utest --path ../examples/data/
```

If that work for you, I'll make sure that this is the default behavior on MacOS. Also, I updated the [documentation for MacOS build] (https://github.com/ethz-asl/libpointmatcher/blob/master/doc/CompilationMac.md).

Lets start with that, then we can check for the linking problem in an external project.

Cheers!

drhalftone, 2015-02-17:
It now compiles completely and successfully runs through its unit tests.  So everything seems to be working fine.  I am unable to link the library to my Qt code, though. I'm trying to create a QObject class that I can use to wrap around an ICP object.  So this object has the following private variables:

private:

    PointMatcher<float>::DataPoints stationaryPoints;

    PointMatcher<float>::DataPoints movingPoints;

    Eigen::MatrixXf scanToMatrix(LAUScan scan);

};

And just including these two lines of code causes the following error during compilation:

Undefined symbols for architecture x86_64:
"PointMatcher<float>::DataPoints::DataPoints()", referenced from:
LAUPointMatcherObject::LAUPointMatcherObject(LAUScan, LAUScan, QObject*) in laupointmatcherobject.o
"PointMatcher<float>::ICPChainBase::setDefault()", referenced from:
vtable for PointMatcher<float>::ICP in laupointmatcherobject.o
"PointMatcher<float>::ICPChainBase::loadAdditionalYAMLContent(YAML_PM::Node&)", referenced from:
vtable for PointMatcher<float>::ICP in laupointmatcherobject.o
"PointMatcher<float>::ICPChainBase::ICPChainBase()", referenced from:
PointMatcher<float>::ICP::ICP() in laupointmatcherobject.o
"PointMatcher<float>::ICPChainBase::~ICPChainBase()", referenced from:
PointMatcher<float>::ICP::~ICP() in laupointmatcherobject.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [LAUScoliosis.app/Contents/MacOS/LAUScoliosis] Error 1
19:58:46: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project LAUScoliosis (kit: Desktop Qt 5.4.0 clang 64bit)
When executing step "Make"
19:58:46: Elapsed time: 00:06.

Clearly this is not the libraries fault.  I must not be doing everything I need to do to include a cmake built library into my QMake project.  Any suggestions would be appreciated.  But thank you very much for fixing the compilation issues because I would have continued to suspect that the problem was in libpointmatcher.

drhalftone, 2015-02-17:
I sent you the wrong error message.  Here is the right one:

Undefined symbols for architecture x86_64:
"PointMatcher<float>::DataPoints::DataPoints()", referenced from:
LAUPointMatcherObject::LAUPointMatcherObject(LAUScan, LAUScan, QObject*) in laupointmatcherobject.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [LAUScoliosis.app/Contents/MacOS/LAUScoliosis] Error 1
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -headerpad_max_install_names -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk -mmacosx-version-min=10.7 -Wl,-rpath,/Qt/5.4/clang_64/lib -o LAUScoliosis.app/Contents/MacOS/LAUScoliosis main.o lauscan.o lauscanmergedialog.o lauiterativeclosestpointobject.o laulaplaciansmootherobject.o laupointmatcherobject.o laufiducialwidget.o qrc_LAUScoliosisWidget.o qrc_shaders.o qrc_lauscan.o moc_lauscan.o moc_lauscanmergedialog.o moc_lauiterativeclosestpointobject.o moc_laulaplaciansmootherobject.o moc_laupointmatcherobject.o moc_laufiducialwidget.o -F/Qt/5.4/clang_64/lib -L/usr/local/lib -ltiff /usr/local/lib/libboost_system-mt.dylib /Users/dllau/Libraries/libnabo/build/libnabo.a -framework QtOpenGL -framework QtWidgets -framework QtGui -framework QtCore -framework DiskArbitration -framework IOKit -framework OpenGL -framework AGL
20:02:05: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project LAUScoliosis (kit: Desktop Qt 5.4.0 clang 64bit)
When executing step "Make"
20:02:05: Elapsed time: 00:03.

drhalftone, 2015-02-17:
One last question.  Is there any chance that you are connecting to the libstdc++ library instead of the libc++ library?  When Apple made Clang the default compiler, it also made libc++ the standard library over libstdc++.  There were a lot of problems during the transition, like the linking issue I am having right now with some programmers using the older library.

pomerlef, 2015-02-17:
We don't use specific calls for MacOS build. Given that [link](http://stackoverflow.com/questions/19774778/when-is-it-necessary-to-use-use-the-flag-stdlib-libstdc) the default since Maverick is libc++ .

Here is the full line called to compile one of the demo file. On my system /usr/bin/c++ is pointing to clang++:

```
/usr/bin/c++   -O2 -g -DNDEBUG -Wl,-search_paths_first -Wl,-headerpad_max_install_names   CMakeFiles/pmicp.dir/icp.cpp.o  -o pmicp  ../libpointmatcher.a /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_program_options-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_chrono-mt.dylib ../contrib/yaml-cpp-pm/libyaml-cpp-pm.a /Users/francoispomerleau/Research/Code/libnabo/build//libnabo.a
```

If you want to compare on your system, you can use the following:
```
make VERBOSE=true
```

I'm not very used to development on MacOS, so it's less clear for me what's the impact of those libraries. I can tell that the whole library do not rely on c++11 functionalities as it caused some problem on Windows in the past.

pomerlef, 2015-02-17:
ok, I'm slowly catching up with MacOS development tools. You can use `otool -L` to see against what your executable is linked.

Here is an example with `pmicp`:

```
otool -L ./examples/pmicp 
./examples/pmicp:
	/usr/local/lib/libboost_thread-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/lib/libboost_filesystem-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/lib/libboost_system-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/lib/libboost_program_options-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/lib/libboost_date_time-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/lib/libboost_chrono-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1213.0.0)
```

which show that it's linked to `libc++.1.dylib`.

pomerlef, 2015-02-17:
I'll rename this issue so it is clearer and close it with the commit e020f2a063ae84229ca112ec011855aefab4f60d. The option of building dynamic library is disabled for when the `APPLE` variable is defined by cmake. So, libpointmatcher will only be build as a static library for simplicity.

I'll close this issue. To keep things separated, if you have further problem with linking libpointmatcher to an external project, please open another issue.

Cheers!

closed 2015-02-17 by pomerlef

pomerlef, 2015-02-20:
@drhalftone you were right about some dependency problems for an external project. The latest commit should fix the problem and there is an example on how to add libpointmatcher to an external project.

Now, after compiling libpointmatcher, you can do:
```
cd /examples/demo_external_link/
mkdir build
cd build
cmake ..
make
```

to confirm that a minimal example can work.

For more details, please follow what's going on in Issue #61 .

Cheers!