norlab-ulaval/libpointmatcher #65: Linking libpointmatcher.a to a Qt project...

drhalftone, 2015-02-21:
This is related to issue #61, but I created a new issue because I wanted to include a specific Qt project that I'm trying to link to the static library.  For those not familiar with Qt, you need two files to create a project. The first is the project file which ends in .pro.  Think of this as an alternative to a CMAKE file:

QT       += core gui widgets
TARGET    = LAUPointMatcher
TEMPLATE  = app
SOURCES  += main.cpp
INCLUDEPATH += /Users/dllau/Libraries/libpointmatcher/pointmatcher \
               /usr/local/include/eigen3/ \
               /usr/local/include/
LIBS     += -L/usr/local/lib/ -lboost_thread-mt -lboost_filesystem-mt \
               -lboost_system-mt -lboost_program_options-mt \
               -lboost_date_time-mt -lboost_chrono-mt \
            /Users/dllau/Libraries/libpointmatcher/build/libpointmatcher.a

Now we need a main.cpp file to hold our code.  So here is what I have for that:

#include <QApplication>
#include <QWidget>
#include "PointMatcher.h"

class LAUPointMatcherWidget : public QWidget
{
    Q_OBJECT

public:
    LAUPointMatcherWidget(QWidget *parent = 0) : QWidget(parent) { ; }
    ~LAUPointMatcherWidget() { ; }

private:
    PointMatcher<float>::DataPoints::Label label;
};

int main(int argc, char *argv[])
{
    QApplication a(argc, argv);
    LAUPointMatcherWidget w;
    w.show();

    return a.exec();
}

I'd be forever indebted to anyone can get this to compile and link and run on a Mac OS X system running OS 10.10 Yosemite.

pomerlef, 2015-02-21:
Looking at your compilation line in issue #61, it seems that you're not linking to libpointmatcher at all.  Here are the only library I see:
```
clang++ ...stuff...  -L/usr/local/lib/ -lboost_thread-mt -lboost_filesystem-mt -lboost_system-mt -lboost_program_options-mt -lboost_date_time-mt -lboost_chrono-mt ...stuff...
```

I would try something like that:
```
LIBS     +=  -L/Users/dllau/Libraries/libpointmatcher/build/ -lpointmatcher
```

If that add the library to your compilation line, I suspect that you will miss libnabo also.
```
LIBS     +=  -L/Users/dllau/Libraries/libnabo/build/ -lnabo
```

It might worth it to investigate how to generate a QT project from cmake. There seems to have some example out there. 

If you manage to make it work. I would like to add a simple QT configuration in the tutorial list eventually. So that it's clear to other too.

drhalftone, 2015-02-21:
Let me address the CMAKE suggestion first.  Yes, it is possible to use CMAKE instead of QMAKE; however, while CMAKE did in fact address a desperate need at one time, its since grown just as unruly as the make process it was intended to replace. QMAKE is a perfectly easy to use build system.  I believe the issues we are seeing now are very much tied to CMAKE.

When using QMAKE, the following line from the PRO file:

LIBS += /Users/dllau/Libraries/libpointmatcher/build/libpointmatcher.a

will link the library with absolute path and filename.  Using the -L and -l is problematic because it tends to create new filenames by adding lib to the front and .dylib to the back. So I find it preferable to specify the library without the -L and -l flags.  I was surprised the first time someone else showed it to me. I also found that in situations where there is both a dynamic and a static library available, QMAKE will take the dynamic library over the static when I really want the static.  This method of absolutely specifying the library avoids that issue.

While I did try it both ways just to make sure that wasn't the problem, I do know it being linked either way because its listed in my compiler output as being one of the linked libraries as listed below.  In this instance, I have my libnabo installed in the /usr/local/lib directory. You can see it listed in the library section of the compiler output.

18:40:37: Running steps for project LAUPointMatcher...
18:40:37: Configuration unchanged, skipping qmake step.
18:40:37: Starting: "/usr/bin/make" 
/Qt/5.4/clang_64/bin/qmake -spec macx-clang CONFIG+=debug CONFIG+=x86_64 -o Makefile ../LAUPointMatcher/LAUPointMatcher.pro
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -headerpad_max_install_names -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk -mmacosx-version-min=10.7 -Wl,-rpath,/Qt/5.4/clang_64/lib -o LAUPointMatcher.app/Contents/MacOS/LAUPointMatcher main.o   -F/Qt/5.4/clang_64/lib -L/usr/local/lib/ -lnabo -lboost_thread-mt -lboost_filesystem-mt -lboost_system-mt -lboost_program_options-mt -lboost_date_time-mt -lboost_chrono-mt /Users/dllau/Libraries/libpointmatcher/build/libpointmatcher.a -framework QtWidgets -framework QtGui -framework QtCore -framework DiskArbitration -framework IOKit -framework OpenGL -framework AGL 
Undefined symbols for architecture x86_64:
  "PointMatcher<float>::DataPoints::Label::Label(std::string const&, unsigned long)", referenced from:
      LAUPointMatcherWidget::LAUPointMatcherWidget(QWidget*) in main.o
  "vtable for LAUPointMatcherWidget", referenced from:
      LAUPointMatcherWidget::~LAUPointMatcherWidget() in main.o
      LAUPointMatcherWidget::LAUPointMatcherWidget(QWidget*) in main.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [LAUPointMatcher.app/Contents/MacOS/LAUPointMatcher] Error 1
18:40:38: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project LAUPointMatcher (kit: Desktop Qt 5.4.0 clang 64bit)
When executing step "Make"
18:40:38: Elapsed time: 00:01.

Really the only path forward I see is to recompile all the code in Qt and completely avoid CMAKE as much as possible.  If anyone has had a similar problem linking to CMAKE build library to anything not CMAKE and has a suggestion about how to do it, I'd love to hear it.  I have never been able to link against PCL, VTK, ITK, or any other Kitware product ever.

drhalftone, 2015-02-22:
Forgive my last post.  I cut and pasted the wrong compilation results.  Here are the correct results showing the undefined symbols:

19:13:35: Running steps for project LAUPointMatcher...
19:13:35: Configuration unchanged, skipping qmake step.
19:13:35: Starting: "/usr/bin/make" 
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -headerpad_max_install_names -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk -mmacosx-version-min=10.7 -Wl,-rpath,/Qt/5.4/clang_64/lib -o LAUPointMatcher.app/Contents/MacOS/LAUPointMatcher main.o moc_laupointmatcherwidget.o   -F/Qt/5.4/clang_64/lib -L/usr/local/lib/ -lnabo -lboost_thread-mt -lboost_filesystem-mt -lboost_system-mt -lboost_program_options-mt -lboost_date_time-mt -lboost_chrono-mt /Users/dllau/Libraries/libpointmatcher/build/libpointmatcher.a -framework QtWidgets -framework QtGui -framework QtCore -framework DiskArbitration -framework IOKit -framework OpenGL -framework AGL 
Undefined symbols for architecture x86_64:
  "PointMatcher<float>::DataPoints::Label::Label(std::string const&, unsigned long)", referenced from:
      LAUPointMatcherWidget::LAUPointMatcherWidget(QWidget*) in main.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [LAUPointMatcher.app/Contents/MacOS/LAUPointMatcher] Error 1
19:13:35: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project LAUPointMatcher (kit: Desktop Qt 5.4.0 clang 64bit)
When executing step "Make"
19:13:35: Elapsed time: 00:00.

pomerlef, 2015-02-25:
I installed QtCreator and messed around with your example code. This compile, link and run on my side:

.pro
```
QT       	+= core gui
greaterThan(QT_MAJOR_VERSION, 4): QT += widgets
TARGET    	 = LAUPointMatcher
TEMPLATE  	 = app
SOURCES  	+= main.cpp

INCLUDEPATH += 	/Users/francoispomerleau/Research/Code/libpointmatcher/pointmatcher \
                                /Users/francoispomerleau/Research/Code/libnabo/ \
                                /usr/local/Cellar/eigen/3.2.4/include/eigen3 \
				/usr/local/include/

#THIS WAS ADDED!
CONFIG          += c++11 

LIBS     	+= /usr/local/lib/libboost_thread-mt.dylib \
                                /usr/local/lib/libboost_filesystem-mt.dylib \
                                /usr/local/lib/libboost_system-mt.dylib \
                                /usr/local/lib/libboost_program_options-mt.dylib \
                                /usr/local/lib/libboost_date_time-mt.dylib \
                                /usr/local/lib/libboost_chrono-mt.dylib \
                                /Users/francoispomerleau/Research/Code/libpointmatcher/build/libpointmatcher.a \
                                /Users/francoispomerleau/Research/Code/libnabo/build/libnabo.a \
                                /Users/francoispomerleau/Research/Code/libpointmatcher/build/contrib/yaml-cpp-pm/libyaml-cpp-pm.
```

main.cpp
```
#include <QApplication>
#include <QWidget>
#include "PointMatcher.h"

class LAUPointMatcherWidget : public QWidget
{
    //Q_OBJECT

	public:
        //LAUPointMatcherWidget(QWidget *parent = 0) : QWidget(parent) { ; }
        //~LAUPointMatcherWidget();

	private:
        PointMatcher<float>::DataPoints::Label label;
};

int main(int argc, char *argv[])
{
	QApplication a(argc, argv);
	LAUPointMatcherWidget w;
	w.show();

	return a.exec();
}
```

I hope that help to shrink down the problem a bit.

drhalftone, 2015-02-25:
You are the man!  Adding “CONFIG c++11” fixed the problem.

You have no idea how happy you have made me, and how you have instilled a faith in humanity that I don’t think I have very felt.  I’m going to be very busy for the next few months digging into this library!

drhalftone, 2015-02-25:
I spent time this evening chiseling away at the project file to see how I could make it smaller and use the libraries from their installed location.  Here is what I reduced the project file down to and was still able to compile, link, and execute:

QT       += core gui widgets
TARGET    = LAUPointMatcher
TEMPLATE  = app
CONFIG   += c++11
HEADERS  += laupointmatcherwidget.h
SOURCES  += main.cpp
INCLUDEPATH += /usr/local/include/ \
               /usr/local/include/eigen3 \
               /usr/local/include/pointmatcher
LIBS     += -L/usr/local/lib/ -lpointmatcher -lboost_system-mt

I'm sure I may need to pull in additional boost libraries as I expand how much of the Point Matcher library I actually use, but that's an easy fix.

pomerlef, 2015-02-25:
Libpointmatcher can compile on with or without c++11 and it seems to be the same for the [cmake external example](https://github.com/ethz-asl/libpointmatcher/tree/master/examples/demo_cmake) relying on boost.

So it's still unclear who required this flag, but if it solve the problem, I'll close this issue.

closed 2015-02-25 by pomerlef