norlab-ulaval/libpointmatcher #67: RemoveNaNDataPointsFilter crashes on Mac OS X...

drhalftone, 2015-02-28:
I finally have libpointmatcher up and running and am able to do absolutely awesome alignments if I manually remove NANs, but when I allow NANs in my point clouds and enable the RemoveNaN filter, the software consistently crashes. Here is the exception message from the application log:

libc++abi.dylib: terminating with uncaught exception of type PointMatcher<float>::ConvergenceError: no outlier to filter

drhalftone, 2015-02-28:
I also found a second situation that creates a similar error.  This time, I tried to align a point cloud with itself.  So my guess is that the distances between points is going to be zero for all points, and the outliers filter (outlierFilters:  - TrimmedDistOutlierFilter: ratio: 0.2) can't decide which points to keep and which to delete.  Seems like you would have to make a random decision to keep points and remove others in cases of a tie.

pomerlef, 2015-04-20:
could you send one test file so we could replicate on our side?

drhalftone, 2015-04-20:
I’ll send you one as soon as I get the chance (in a day or two).

pomerlef, 2015-09-09:
If there is no follow-up on this, I'll close it.

closed 2015-09-29 by pomerlef