norlab-ulaval/libpointmatcher #70: yaml-cpp-pm isn't compiled anymore under some build conditions

HannesSommer, 2015-03-07:
I strongly suspect that this line 
```cmake
 get_property(yaml-cpp-pm_LIB TARGET yaml-cpp-pm PROPERTY LOCATION)
```
(https://github.com/ethz-asl/libpointmatcher/commit/a9b72eb323#diff-af3b638bc2a3e6c650974192a53c7291R190)

Isn't exactly what we want, because it resolves to a path and not to a target name. It seems not implicitly create a target dependency for ```pointmatcher``` anymore:
```
[ 37%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/TransformationCheckersImpl.cpp.o
[ 39%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/InspectorsImpl.cpp.o
make[2]: *** No rule to make target `contrib/yaml-cpp-pm/libyaml-cpp-pm.a', needed by `libpointmatcher.so.1.2.1'.  Stop.
```
(taken from http://jenkins.ros.org/job/prerelease-indigo-libpointmatcher/ARCH_PARAM=amd64,UBUNTU_PARAM=trusty,label=prerelease/1/console).

It can be reproduced with a ```make pointmatcher``` when ```yaml-cpp-pm``` has not been built or its output removed before. Apparently this is what is issued on jenkins.ros.org. (it works with ```make```, because ```yaml-cpp-pm``` is within the default targets and _before_ ```pointmatcher```.

The simple solution before a9b72eb323:

```cmake
set (EXTERNAL_LIBS ${EXTERNAL_LIBS} yaml-cpp-pm )
```
would make cmake to let ```libpointmatcher``` depend on ```yaml-cpp-pm``` (in the sense of cmake's target dependencies), in combination with 
```cmake
target_link_libraries(pointmatcher ${EXTERNAL_LIBS})
```
(not changed with a9b72eb323)

@pomerlef , what was the reason for this location property based solution? 

One can fix it with adding an explicit dependency. E.g. :
```cmake
add_dependencies(pointmatcher yaml-cpp-pm)
```

But this needs to come after the ```pointmatcher``` target's declaration. So maybe we should introduce another collecting variable for dependencies? E.g. ```POINTMATCHER_EXTRA_DEPS``` ?

HannesSommer, 2015-03-07:
As a suggested solution I've just created #71 .

pomerlef, 2015-03-09:
there was a problem with the find_package(libpointmatcher) command so I cleaned all dependencies and made that change meanwhile. It appear that the problem would be related to  `yaml-cpp-pm ` not being installed #66 with `make install`.

Would it be more convienient to roll back to:
```
set (EXTERNAL_LIBS ${EXTERNAL_LIBS} yaml-cpp-pm )
```
instead of your patch?

HannesSommer, 2015-03-09:
Ah, you mean if yaml-cpp-pm is properly installed, then we don't need to resolve the location anymore? Makes sense... Yes, I would go for that. I'm not that comfortable with relying on the properties. I think they are not that frequently used and will therefore at some point in future break with our expectations. Furthermore it is wrong to use the targets location here, as this will not be the installed location, right? So as soon as someone deletes the build folder it doesn't work anymore. Like it was the problem in libnabo before 62a384d99e943 (see https://github.com/ethz-asl/libnabo/blame/master/CMakeLists.txt#L174 ). 

So then we would rollback here (partially), so use exactly the line 
```
set (EXTERNAL_LIBS ${EXTERNAL_LIBS} yaml-cpp-pm )
```
again. 

But that means we have to solve #66. It will be only important for people using the static libpointmatcher version (the dynamic library will already be linked to yaml-cpp-pm). But still this is important as we allow static pointmatcher library. We should not forget then that we need to fix the path in the ```libpointmatcherConfig.cmake ```  for yaml-cpp-pm as well, like in https://github.com/ethz-asl/libpointmatcher/blob/master/CMakeLists.txt#L342. This is kind of destroying the concept of EXTERNAL_LIBS again. I'm afraid the best is to not treat yaml-cpp-pm as EXTERNAL_LIBS but give it a special treatment again. Otherwise you have to remove yaml-cpp-pm from EXTERNAL_LIBS before you write it in the installed version of libpointmatcherConfig.cmake. 

Oh my, this is all ugly. The ideal would be of course to just support new versions of yaml ;) and get rid of this backed in yaml version (#42).   But who has the time for that? Has anybody ever looked into the effort that would need? Maybe it is just a few lines we are trying to avoid :). That would be a bad joke...

HannesSommer, 2015-03-09:
Just had a look: Doesn't seem to be too bad. But we are also not too lucky. I guess 3h. Especially if one goes the cheap way to still support the old API (for environments with no yaml-cpp 0.5). That would be keeping the old code and switch between old and new with the preprocessor.

pomerlef, 2015-03-09:
My hope is to have an intern this summer for that kind of stuff. For now, I need to reduce my supporting time a bit...

My fear with yaml library is that they are still at version 0.5, which means that they didn't want to commit on their current API... But I agree that we should follow the latest version.

Another solution would be to add yaml source in libpointmatcher and to not consider it as a dependency at all.

HannesSommer, 2015-03-10:
Yes, packing all the yaml objects into pointmatcher is a nice option as well :). I looked into that already. The problem is that this would mean reading through the CMakeLists.txt of yaml-cpp-mp, which is a bit bigger than what one would expect. But most likely there is nothing important in it. (compiler flags and co.). There is unit test coverage for the yaml functionality of libpointmatcher? If yes, one could just try..

pomerlef, 2015-03-19:
This will be considere close for now

closed 2015-03-19 by pomerlef