norlab-ulaval/libpointmatcher #72: Could not compile in Xcode 6

jayden118, 2015-03-11:
Dear all,

I have compiled and intalled the libpointmatcher in my mac os machine successfully. However, I want to link to my current xcode projects and trying to compile the libpointmatcher project in xcode. I followed the instruction in advanced section but failed to compile the code in xcode. The errors are as follows:

[image]

May I know how to solve this issue? 

Thank you very much.

Regards,
Jayden

pomerlef, 2015-03-11:
My guess is that your problem is related to #66. As a temporary fix, you can copy `libyaml-cpp-pm.a` in your install folder. In you case, you should be able to find the library in:
```
./libpointmatcher/build/contrib/yaml-cpp-pm
```
and your install folder [1]:
```
/usr/local/lib
```
[1] confirm that libpointmacher.a is there, I'm not sure where `make install` will send it in your case.

This would be a temporary fix, while we arrange the whole yaml dependency problem.

jayden118, 2015-03-12:
Dear pomerlef,

Thank you very much. Now everything is working fine.

Cheers!

Regards,
Jayden

closed 2015-03-12 by jayden118