norlab-ulaval/libpointmatcher #73: Building on Windows

braddodson, 2015-03-14:
I'm trying to follow the instructions to build on Windows and I'm failing when trying to run CMake for libnado.

In particular, I have this error:
C:\repos\libnado-build\CMakeFiles\CMakeTmp\CheckSymbolExists.c(2): fatal error C1083: Cannot open include file: 'unistd.h': No such file or directory [C:\repos\libnado-build\CMakeFiles\CMakeTmp\cmTryCompileExec3210956335.vcxproj]

I think this means it's trying to do a unix build.
I'm using Visual Studio 2012 Express on Win 8.1 (although I have VS2013 installed also, which might be affecting it). I have cmake 3.2.0-rc2 if that could potentially make a difference.

pomerlef, 2015-03-14:
Just to make sure. You receive this error while compiling or while running cmake?

This problem is related to the flag `POSIX_TIMERS` which should not exist in your case. Could you report the result of 
```
grep POSIX_TIMERS CMakeCache.txt
```
from you `build` folder in your libnabo.

braddodson, 2015-03-14:
The error is while running CMake. I installed CMake 2.8.11.2 just to be sure. I also realized that I have to add grep to the PATH before running CMake (I figured the installer would do this). That gets me further, but I'm still getting an error about the BOOST_ROOT:

CMake Error at C:/Program Files (x86)/CMake 2.8/share/cmake-2.8/Modules/FindBoost.cmake:1106 (message):
  Unable to find the requested Boost libraries.

  Unable to find the Boost header files.  Please set BOOST_ROOT to the root
  directory containing Boost or BOOST_INCLUDEDIR to the directory containing
  Boost's headers.
Call Stack (most recent call first):
  CMakeLists.txt:46 (find_package)

I've tried setting BOOST_ROOT to the path of the boost repository and also the boost\build folder (which contains lib and include folders). CMake seems to remove the BOOST_ROOT variable everytime I click configure.

Thanks for your help!

pomerlef, 2015-03-14:
As you can see, the instructions are dating a bit. I did that for a company a while ago and I don't have access to a Windows machine close by. So your inputs are quite appreciated, but I can only roughly guide you.

I updated the info for `grep` on [Compiling and Installing libpointmatcher on Windows](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/CompilationWindows.md#install-grep-tool).

Just to make sure I'm following right. You don't have the problem with `Cannot open include file: 'unistd.h'` and it's either downgrading cmake to 2.8 or adding `grep` to your `PATH` that make it work.

As for boost on Windows, I recall struggling with it. Have a look to this post:
http://stackoverflow.com/questions/17108593/cannot-specify-boost-root-for-cmake

pomerlef, 2015-04-20:
@braddodson is that issue resolved?

braddodson, 2015-04-23:
After a lot of annoyance, I did finally succeed in building a .lib for pointmatcher. I can't quite remember all of the things I had to do, but some stuff I remember:
- If you use a new version of boost, you also need a new version of CMake which looks for that version number
- If you have VS2013 installed, CMake will use it despite you specifying otherwise, so you need to choose to generate for it.

I never did succeed in building exe's to run the code, and since I wanted to reference from C# I realized that the interop was going to be difficult (since it uses C++ classes).

I ended up just translating the parts I needed to C#. I've published those to github here: https://github.com/braddodson/pointmatcher.net

Perhaps someone will find them useful. Thanks for your help!

pomerlef, 2015-04-23:
thanks, I'll link to this issue in the documentation for Windows

closed 2015-04-23 by pomerlef