norlab-ulaval/libpointmatcher #75: Build_map.cpp runtime error

jayden118, 2015-03-16:
Hi all,

I'm trying to run the build_map.cpp in terminal but it shows up this error:

[image]

From my understanding I guess these two line cause the problem. 
[image]

May I know how to solve this issue?

Thank you in advanced.

Regards,
Jayden

pomerlef, 2015-03-17:
good catch. This example is outdated (2012). I'll fix it tomorrow.

jayden118, 2015-03-17:
Hi pomerlef,

I'll wait for your solution. Thank you very much.

Regards,
Jayden

pomerlef, 2015-03-17:
Many things were outdated in that example. It should be solved with the latest commit.

pomerlef, 2015-03-18:
I will assume that it's working. Just reopen if there is something

closed 2015-03-18 by pomerlef

jayden118, 2015-03-19:
Hi powerlef,

Yes. The updated code is working perfectly now. 

Thank you very much.

Regards,
Jayden