norlab-ulaval/libpointmatcher #76: crash merging two 3-D scans together...

drhalftone, 2015-03-30:
I'm using Point Matcher to merge a scan with its mirror image.  Doing so shows me where the plane of symmetry through the scan exits.  The error that I am getting is as follows:

* KDTreeMatcher: initialized with knn=1, epsilon=3.16, searchType=1 and maxDist=inf
Applying 1 DataPoints filters - 2326 points in
* MinDistDataPointsFilter - 2326 points out (-0%)
Applied 1 filters - 2326 points out (-0%)
PointMatcher::icp - reference pre-processing took 0.000998562 [s]
Applying 1 DataPoints filters - 2326 points in
* MinDistDataPointsFilter - 2326 points out (-0%)
Applied 1 filters - 2326 points out (-0%)
PointMatcher::icp - reading pre-processing took 0.00028519 [s]
libc++abi.dylib: terminating with uncaught exception of type PointMatcherSupport::TransformationError: RigidTransformation: Error, rotation matrix is not orthogonal.

Here is my YAML file:

readingDataPointsFilters:
  - MinDistDataPointsFilter:
      minDist: 0.1

referenceDataPointsFilters:
  - MinDistDataPointsFilter:
      minDist: 0.1

matcher:
  KDTreeMatcher:
    knn: 1
    epsilon: 3.16

outlierFilters:
  - TrimmedDistOutlierFilter:
      ratio: 0.65

errorMinimizer:
  PointToPointErrorMinimizer

transformationCheckers:
  - CounterTransformationChecker:
      maxIterationCount: 1500
  - DifferentialTransformationChecker:
      minDiffRotErr: 0.00001
      minDiffTransErr: 0.00001
      smoothLength: 4   
      
inspector:
  PerformanceInspector

logger:
  FileLogger

Any my point cloud data:

   8.4321906e+02   8.2779266e+02   8.1271967e+02   7.9854205e+02   7.9012988e+02   7.8206256e+02   7.7703668e+02   7.6574438e+02   7.9717096e+02   7.8567175e+02   7.8383630e+02   7.6975543e+02   7.6117877e+02   7.5024780e+02   7.4180878e+02   7.7265179e+02   7.5876123e+02   7.5065198e+02   7.3674994e+02   7.2618207e+02   7.0375812e+02   7.0109021e+02   6.9030280e+02   6.8488495e+02   6.7909662e+02   7.6128931e+02   7.4472089e+02   7.3119910e+02   7.2276453e+02   7.0923053e+02   6.9307080e+02   6.7968683e+02   6.6889014e+02   6.7940985e+02   6.6015186e+02   7.5052478e+02   7.3419067e+02   7.2032837e+02   7.0710089e+02   6.9337262e+02   6.8024603e+02   6.6715808e+02   6.5410522e+02   6.3610675e+02   6.2080505e+02   6.0327649e+02   5.9062482e+02   5.7798938e+02   5.6536731e+02   5.5446558e+02   5.4181592e+02   5.3322009e+02   5.2046600e+02   5.1612958e+02   5.0498929e+02   4.9626102e+02   7.4454449e+02   7.2834058e+02   7.1231836e+02   7.0206616e+02   6.8845123e+02   6.7276495e+02   6.5723944e+02   6.4441858e+02   6.2913849e+02   6.1400433e+02   6.0090698e+02   5.8831409e+02   5.7348444e+02   5.6097778e+02   5.5232806e+02   5.3973523e+02   5.3119574e+02   5.1849756e+02   5.0809497e+02   4.9904388e+02   4.8832660e+02   4.7922113e+02   4.6444525e+02   7.3138788e+02   7.1547028e+02   7.0488055e+02   6.8919965e+02   6.7587427e+02   6.6314636e+02   6.4784283e+02   6.3473187e+02   6.2217017e+02   6.0963251e+02   5.9474622e+02   5.7999542e+02   5.6717743e+02   5.5263776e+02   5.4634308e+02   5.3182849e+02   5.2147870e+02   5.1259210e+02   5.0197336e+02   4.9124051e+02   4.8255612e+02   4.7187540e+02   4.5902109e+02   4.4615433e+02   4.3708450e+02   4.2590082e+02   4.1635101e+02   4.0481219e+02   3.9670917e+02   3.8341150e+02   7.2899579e+02   7.1313025e+02   6.9973126e+02   6.8416492e+02   6.7095288e+02   6.5833710e+02   6.4523267e+02   6.3014343e+02   6.1769055e+02   6.0283179e+02   5.9000720e+02   5.7768469e+02   5.6537543e+02   5.5088196e+02   5.4206812e+02   5.2974780e+02   5.1742999e+02   5.0708035e+02   5.0006033e+02   4.8752463e+02   4.8075287e+02   4.6627866e+02   4.5359689e+02   4.4090158e+02   4.3168448e+02   4.2068069e+02   4.0953043e+02   3.9680176e+02   3.8375201e+02   3.7545139e+02   3.6547556e+02   3.5226480e+02   3.4345822e+02   3.3421048e+02   3.2344983e+02   7.2361353e+02   7.0786511e+02   6.9229340e+02   6.7913013e+02   6.5946954e+02   6.4230615e+02   6.3165985e+02   6.1688800e+02   6.0425165e+02   5.8971619e+02   5.7958142e+02   5.6705524e+02   5.5501398e+02   5.4473663e+02   5.3437311e+02   5.2225726e+02   5.1176172e+02   5.0117493e+02   4.9087790e+02   4.7860648e+02   4.6776929e+02   4.5718396e+02   4.4478262e+02   4.3236591e+02   4.2183737e+02   4.0778391e+02   4.0004095e+02   3.8593048e+02   3.7934106e+02   3.6802197e+02   3.5501880e+02   3.4222806e+02   3.3477792e+02   3.2298782e+02   3.1509030e+02   3.0029602e+02   3.0673404e+02   2.9096445e+02   2.7403052e+02   7.0866284e+02   6.9323981e+02   6.7798981e+02   6.6514478e+02   6.5017358e+02   6.3535944e+02   6.2487341e+02   6.1280939e+02   6.0026978e+02   5.8583008e+02   5.7531628e+02   5.6335809e+02   5.5095947e+02   5.4078613e+02   5.3052563e+02   5.1809589e+02   5.0609348e+02   4.9566318e+02   4.8705188e+02   4.7489059e+02   4.6452338e+02   4.5368597e+02   4.4139249e+02   4.2908295e+02   4.1675497e+02   4.0287085e+02   3.9529620e+02   3.8278354e+02   3.7327603e+02   3.6218457e+02   3.5093814e+02   3.3806650e+02   3.2797446e+02   3.1760095e+02   3.1123206e+02   2.9907034e+02   2.8439432e+02   2.8820035e+02   2.7263596e+02   2.5843903e+02   6.9909442e+02   6.8387958e+02   6.7112408e+02   6.5619415e+02   6.4361169e+02   6.3108453e+02   6.1860901e+02   6.0414240e+02   5.9180829e+02   5.7951520e+02   5.6962946e+02   5.5550159e+02   5.4510303e+02   5.3288501e+02   5.2283069e+02   5.1268604e+02   5.0042523e+02   4.9015143e+02   4.8169547e+02   4.6783041e+02   4.5731030e+02   4.4703986e+02   4.3325623e+02   4.2284531e+02   4.0913141e+02   3.9672952e+02   3.8610327e+02   3.7534528e+02   3.6748666e+02   3.5634714e+02   3.4532721e+02   3.3537372e+02   3.2656683e+02   3.1378525e+02   3.0458728e+02   2.9273755e+02   2.8070340e+02   2.7087885e+02   2.7263596e+02   2.5597299e+02   2.4289680e+02   6.9191809e+02   6.8153955e+02   6.6425836e+02   6.5171881e+02   6.3923706e+02   6.2680957e+02   6.1234460e+02   6.0210309e+02   5.8782635e+02   5.7562909e+02   5.6536438e+02   5.5134222e+02   5.4149902e+02   5.3112921e+02   5.2112067e+02   5.0727618e+02   4.9678137e+02   4.8503339e+02   4.7480862e+02   4.6448611e+02   4.5406439e+02   4.4039371e+02   4.2681503e+02   4.1627939e+02   4.0595493e+02   3.9396594e+02   3.8313782e+02   3.7391486e+02   3.6445413e+02   3.5369379e+02   3.4405200e+02   3.3145694e+02   3.2398621e+02   3.1243854e+02   3.0330121e+02   2.9028616e+02   2.7953784e+02   2.6756198e+02   2.5956198e+02   2.4808173e+02   2.3733429e+02   2.2320586e+02   6.8952594e+02   6.7451941e+02   6.5968121e+02   6.4724347e+02   6.3267517e+02   6.1825977e+02   6.0816833e+02   5.9394592e+02   5.8185352e+02   5.6979993e+02   5.5967755e+02   5.4579645e+02   5.3564258e+02   5.2542285e+02   5.1556317e+02   5.0186636e+02   4.9151797e+02   4.7991534e+02   4.6792178e+02   4.5593958e+02   4.4757260e+02   4.3409738e+02   4.2206888e+02   4.1037006e+02   4.0150784e+02   3.8935995e+02   3.8046890e+02   3.6847922e+02   3.6031888e+02   3.4812170e+02   3.3997131e+02   3.3023294e+02   3.2023257e+02   3.0884729e+02   2.9837122e+02   2.8803906e+02   2.7720676e+02   2.6627206e+02   2.5624991e+02   2.4479370e+02   2.3424399e+02   2.2320586e+02   2.0992577e+02   1.9928603e+02  -1.5644124e+02  -1.6852643e+02  -1.8254979e+02  -1.9702029e+02  -2.1180492e+02   6.8234961e+02   6.6749927e+02   6.5510406e+02   6.4053052e+02   6.3048792e+02   6.1398486e+02   6.0399207e+02   5.8986731e+02   5.7787164e+02   5.6397076e+02   5.5588635e+02   5.4209930e+02   5.3023657e+02   5.1839966e+02   5.0658572e+02   4.9645651e+02   4.8665948e+02   4.7322250e+02   4.6141757e+02   4.4962256e+02   4.3963821e+02   4.2780106e+02   4.1766171e+02   4.0577393e+02   3.9547253e+02   3.8536807e+02   3.7750345e+02   3.6561835e+02   3.5756204e+02   3.4679501e+02   3.3869611e+02   3.2754016e+02   3.1765195e+02   3.0503159e+02   2.9601340e+02   2.8436197e+02   2.7487564e+02   2.6406082e+02   2.5311215e+02   2.4167007e+02   2.3115370e+02   2.2031271e+02   2.0992577e+02   1.9828584e+02  -1.5644124e+02  -1.6907359e+02  -1.8254979e+02  -1.9780275e+02  -2.1263683e+02  -2.2708377e+02   6.7577130e+02   6.6340417e+02   6.4652191e+02   6.3213928e+02   6.2392603e+02   6.0970996e+02   5.9772766e+02   5.8171008e+02   5.7189880e+02   5.6008466e+02   5.4830389e+02   5.3285632e+02   5.2483057e+02   5.1137646e+02   4.9803574e+02   4.8813370e+02   4.7815710e+02   4.6652969e+02   4.5338293e+02   4.4181918e+02   4.3350708e+02   4.2045532e+02   4.1020349e+02   3.9855139e+02   3.8721365e+02   3.7677023e+02   3.6653122e+02   3.5760794e+02   3.5232404e+02   3.4175363e+02   3.3334021e+02   3.2239941e+02   3.1507132e+02   3.027870 [... cut ...]  -1.1530000e+03  -1.1570000e+03  -1.1610000e+03  -1.1650000e+03  -1.1730000e+03  -1.1770000e+03  -1.1850000e+03  -1.1890000e+03  -1.1970000e+03  -1.2020000e+03  -1.2190000e+03  -1.2190000e+03  -1.2270000e+03  -1.2270000e+03  -1.2320000e+03  -1.2320000e+03  -1.2320000e+03  -1.2410000e+03  -1.2450000e+03  -1.2450000e+03  -1.4040000e+03  -1.2640000e+03  -1.3920000e+03  -1.3980000e+03  -1.3920000e+03  -1.4040000e+03  -1.4150000e+03  -1.4390000e+03  -1.4520000e+03  -1.4580000e+03  -1.4580000e+03  -1.

pomerlef, 2015-04-20:
I'll close that one since a workaround was found and the bug could not be reproduce on our side.

closed 2015-04-20 by pomerlef

artemiialessandrini, 2019-12-20:
@drhalftone Hi, could you please provide more details how to solve this issue?