norlab-ulaval/libpointmatcher #77: Last libnabo version compatibility

smichaud, 2015-04-08:
I tried to follow the installation steps from the readme and I get error (see below). 
When I use an older version of libnabo it works. You might consider binding a working version of libnabo with libpointmatcher.

/home/smichaud/Libraries/libnabo/nabo/kdtree_cpu.cpp:302: undefined reference to `GOMP_parallel_start'
/home/smichaud/Libraries/libnabo/nabo/kdtree_cpu.cpp:302: undefined reference to `GOMP_parallel_end'
/usr/local/lib/libnabo.a(kdtree_cpu.cpp.o): In function `Nabo::KDTreeUnbalancedPtInLeavesImplicitBoundsStackOpt<double, Nabo::IndexHeapSTL<int, double> >::knn(Eigen::Matrix<double, -1, -1, 0, -1, -1> const&, Eigen::Matrix<int, -1, -1, 0, -1, -1>&, Eigen::Matrix<double, -1, -1, 0, -1, -1>&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, int, double, unsigned int) const':
/home/smichaud/Libraries/libnabo/nabo/kdtree_cpu.cpp:334: undefined reference to `GOMP_parallel_start'
/home/smichaud/Libraries/libnabo/nabo/kdtree_cpu.cpp:334: undefined reference to `GOMP_parallel_end'
collect2: ld returned 1 exit status
make[2]: *** [examples/align_sequence] Error 1
make[1]: *** [examples/CMakeFiles/align_sequence.dir/all] Error 2
make: *** [all] Error 2

HannesSommer, 2015-04-09:
The old libnabo did not use GOMP. It isn't a simple incompatibility between the two masters. Apparently on your system libpointmatcher does not get linked against gomp. Something goes wrong with propagating this dependency from libnabo to libpointmatcher. I tried to reproduce but could not. Inspecting the code did not help either. 
If you like to help fixing it, could you please post the content of your ```/usr/local/share/libnabo/cmake/libnaboConfig.cmake``` and maybe the output of ```make VERBOSE=1``` in your libpointmatcher build folder?
Otherwise you could keep to the old libnabo and work without OpenMP support in libnabo till this is solved.

smichaud, 2015-04-09:
My bad, I was using my forked version of libpointmatcher (not up to date) with the libnabo from the master.  I just recompiled all the most recent sources for both and it works perfectly. I'll merge the changes in my repo. Sorry for the inconvenience.

closed 2015-04-09 by pomerlef