norlab-ulaval/libpointmatcher #78: Building on windows

gn02596703, 2015-04-10:
I'm trying to follow the instructions to build on Windows using MSVC 2012.  In the last step. Cmake comes up error in CMakeList.txt line 274 (add_dependencies called with incorrect number of arguments). 
I try to delete the line and Cmake can work. But when I cd to my build folder and enter msbuild /m:2 libpointmatcher.s as instructions. The compiler shows 4 errors and they all mentioned error c2582 'operator =' function is unavailable in 'PointMatcherIO<T>::SupportedLabel'. 
Should I recover the deleted line??

pomerlef, 2015-04-10:
That line was added after testing on Windows. Can you add the following under line 274 and report the results:
```
message("EXTRA_DEPS: ${EXTRA_DEPS}")
```

I've the feeling that the variable is empty in your specific configuration.

@HannesSommer do you know what happen if `EXTRA_DEPS` is empty?

pomerlef, 2015-04-10:
@HannesSommer empty quotes seems to work

```
add_dependencies(pointmatcher "")
```

as oppose to just an empty variable which would produce the error of missing arguments:
```
add_dependencies(pointmatcher )
```

HannesSommer, 2015-04-11:
Yes, @pomerlef this seems to be one of the wired truths about cmake :). I would suggest

```
if(EXTRA_DEPS)
	add_dependencies(pointmatcher ${EXTRA_DEPS})
endif(EXTRA_DEPS)
```
I pushed this suggestion as PR #79.

gn02596703, 2015-04-11:
Thanks I add 

if(EXTRA_DEPS)
    add_dependencies(pointmatcher ${EXTRA_DEPS})
endif(EXTRA_DEPS)
and cmake can work.

 The compiler still shows 4 errors and they all mentioned error c2582 'operator =' function is unavailable in 'PointMatcherIO::SupportedLabel'. It seems that the problem caused by MSVC. Should I try to use under Ubuntu??

HannesSommer, 2015-04-14:
The compiler complain is independent of the CMAKE issue. 

Yes, I guess the problem is cause by MSVC seeing a true assignment (not a copy construction) somewhere, where other compilers don't, e.g. gcc. 
A quick remedy for you should be to remove the three ```const``` in https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/IO.h#L81-L83.

To suggest a real fix i would need more detailed the compiler output. Especially where it is caused.

pomerlef, 2015-04-14:
@gn02596703 Could you give us the line number generating the error c2582?

Also, which version of MSVC are you using? Please report any version different from https://github.com/ethz-asl/libpointmatcher/blob/master/doc/CompilationWindows.md

gn02596703, 2015-04-14:
@HannesSommer @pomerlef 
Thanks
I removed the three const mentioned above and it can work.
My MSVC version is Visual Studio 2012 and other dependencies are same as the document.
The error log is at
 Microsoft Visual Studio 11.0\VC\include\utility line <49> " operator =" function is unavailable in 'PointMatcherIO::SupportedLabel'

pomerlef, 2015-04-14:
Good job @HannesSommer ! 

Following that forum: http://stackoverflow.com/questions/11309416/stdvectora-error-c2582-operator-function-is-unavailable-in

We would need to explicitly define the default constructor `SupportedLabel(const SupportedLabel& label)` even though the error blames the assignment operator.

Would that make sense?

HannesSommer, 2015-04-18:
Hm, that would be a big surprise to me. But then again we are talking about a Microsoft compiler ;). We could easily try. 
If I understand correctly (http://en.cppreference.com/w/cpp/container/vector, "Template parameters") then, prior to c++11, the std::vector required its template parameter to be a CopyAssignable (which a class with const members, like SupportedLabel will never be, unless providing a fake assign operator - which sounds like a bad idea ;) ). Starting with c++11 this is only required for some operations on vectors. So if we depend on c++11 the code is probably fine and the MSVC complier is wrong. We could then (preprocessor) condition the ```const``` for the Microsoft compiler until they do the right thing.
If we target prior c++11 these ```const```s can not correctly be combined with std::vector. So we need an alternative for the pair of them.
@pomerlef, Do you remember the reasoning behind this combination? 

My PR #81 suggests removing the ```const```s there and make the vector returned by ```getSupportedExternalLabels``` const instead. It also implements the list as a singleton. This should work prior c++11. But it is only thread safe since c++11. Will ```getSupportedExternalLabels``` be called in parallel? If yes, we should remove this singleton concept but keep the ```const```s.

I don't have an overview regarding use cases or purpose of the ```SupportedLabels``` - so I'm not sure whether it is suitable.

pomerlef, 2015-04-20:
@gn02596703 A patch was send to the main branch conserning your problem. Could you `git pull` the latest sources and confirm that it's working on your side before I close this.

gn02596703, 2015-04-24:
@pomerlef  I tried to compile the new version. The only error is that it can't open nabo.lib.  Error c2582 didn't come up. I will check whether my setting caused the error.

pomerlef, 2015-09-09:
We moved forward with modified CMakeList.txt following issue #104, #105 and #111. This issue is most probably obsolete.

closed 2015-09-09 by pomerlef