norlab-ulaval/libpointmatcher #80: Building on Windows

tim30235, 2015-04-15:
The step "Build libpointmatcher"
I got some cmake error but don't know how to fix it
[image]
[image]
is there anything wrong??

pomerlef, 2015-04-20:
The error lines in your error doesn't fit the current sources.

Please update to the latest version. Some of those problems were already resolved recently.

We only do limited support for Windows. On my side I can only access to an Ubuntu and Mac systems. But, if you have other problems, don't hesitate to post.

tim30235, 2015-04-22:
Hi~
I've tried the latest version and got the same error messages.
"Unable to find the requested Boost libraries." while building on the other notebook with Win7.
Should I change any path or dir in CMakeList.txt ?

HannesSommer, 2015-04-22:
To me it seems to be mostly warnings from a very recent CMake (this should interest us at some point). 

The only error (concerning yaml-cpp) is caused in https://github.com/ethz-asl/libpointmatcher/blob/master/contrib/yaml-cpp-pm/CMakeLists.txt#L182. Apparently the target renaming has not been executed completely.

PR #84 should solve this, but I can't test it.

pomerlef, 2015-04-22:
Please `git pull` again. The patch was integrated to the main branch.

closed 2015-09-09 by pomerlef