norlab-ulaval/libpointmatcher #86: Tutorial on a way to retrieve the final mean distance

galou, 2015-05-05:
First, thanks for the great implementation of ICP!

I'm missing a tutorial on a way to retrieve the final mean distance, taking into account outliers. I'm willing to write this tutorial myself but I need a hint how to actually do it.

I tried to use PointToPlaneWithCovErrorMinimizer but getCovariance returns 3.40282e+38 on all diagonal elements when calling it after calling icp.

pomerlef, 2015-05-08:
Hi galou,

I can help you but you will need to define what is "the final mean distance". There are many distances used in the library, and they might not be what you expect. The library is modular so the matching distance is not forced to be in Euclidean space, outliers can be weighted (soft rejection) instead of binary (hard rejection), etc. Also, there is currently no memory in the ICP class (I need to confirm that), so it would be close to: "knowing the transformation, can I extract a list of matched points and compute the Euclidean distance of the matches with a non zero weight."

Is it what you want? I can work on an example code doing so in the following days.

galou, 2015-05-12:
What I meant was, there is an internal criterion to stop the iterations if a value is smaller than a certain threshold. I would need this value. I should have written "final error" instead of "final mean distance".

I want to use it because I want to compare two places that can be far from each other (especially the rotation). So I compute the ICP with different angles and I would then take the transform with the smallest "place difference" value if I had access to it.

Basically, what you write about computing this value a posteriori would be a solution, I just need to be sure that the same conditions (filters, error function,...) are used. Thanks for offering to provide an example!

pomerlef, 2015-05-19:
Hi Galou,

The modules looking at convergence (i.e., transformationChecker) only look at the transformation parameters and evaluate if they stop changing. I think you won't be able to define if one point cloud is better aligned than the other just by looking at those values.

I suggest you to do the following for a more precise evaluation. Just add this code snippet in the example [icp.cpp](https://github.com/ethz-asl/libpointmatcher/blob/master/examples/icp.cpp#L145) to test it on your side.

``` c++
148     // START demo
149     // Test for retrieving paired point mean distance
150     // ref: point cloud used as reference
151     // data_out: aligned point cloud (using the transformation outputted by icp)
152     // icp: icp object used to aligned the point clouds
153     
154     // initiate the matching with unfiltered point cloud
155     icp.matcher->init(ref);
156     // extract closest points 
157     const PM::Matches matches = icp.matcher->findClosests(data_out);
158     // weight paired points
159     const PM::OutlierWeights outlierWeights = icp.outlierFilters.compute(data_out, ref, matches);
160     // generate tuples of matched points and remove pairs with zero weight
161     const PM::ErrorMinimizer::ErrorElements matchedPoints = icp.errorMinimizer->getMatchedPoints( data_out, ref, matches, outlierWeights);
162     
163     
164     // extract relevant information for convenience
165     const int dim = matchedPoints.reading.getEuclideanDim();
166     const int nbMatchedPoints = matchedPoints.reading.getNbPoints(); 
167     const PM::Matrix matchedRead = matchedPoints.reading.features.topRows(dim);
168     const PM::Matrix matchedRef = matchedPoints.reference.features.topRows(dim);
169     
170     // compute mean distance
171     const PM::Matrix dist = (matchedRead - matchedRef).colwise().norm(); // replace that by squaredNorm() to save computation time
172     const float meanDist = dist.sum()/nbMatchedPoints;
173     cout << "meanDist: " << meanDist << endl;
174     
175     // END demo
```

There would be a faster way to do so, but the functionality was not enforced through all the errorMinimizer module so it could failed with difference yaml configuration. I will add an enhancement issue to standardize this functionality. For now, it's not my highest priority.

You are more than welcome to write a tutorial about it.

Cheers!

closed 2015-05-19 by pomerlef

pomerlef, 2015-05-19:
I forgot, please `git pull` and `make` the library before  trying the example.