norlab-ulaval/libpointmatcher #96: Android build

bhack, 2015-05-21:
Is this really buildabile for Android target? I see some boost dependencies in libnao and here that could be a little bit heave to resolve under Android.

pomerlef, 2015-05-21:
Where have you seen that we are supporting Android builds?

Libpointmatcher is developed on Ubuntu, is supported on MacOs and was ported once in the past on Windows. There is no plan to port it on mobile devices unless a new developer join actively the project to do so.

bhack, 2015-05-21:
No it wasn't declared. I only see that that there are no main problem to run this on android other than boost dependencies.  I will try to build it with https://github.com/MysticTreeGames/Boost-for-Android

closed 2015-05-21 by bhack

EthanGreen75, 2017-10-19:
@bhack hello, did you build this successfully on android ? I want to build it too.