norlab-ulaval/libpointmatcher #101: Missing DESTINATION for TARGETS error when building libpointmatcher for Windows

kwill, 2015-07-05:
I have been able to build all of the dependencies successfully on Windows. Now that I am trying to build libpointmatcher itself, I am getting the following error:

````
-- libnabo found, version 1.0.6
-- using built-in yaml-cpp, version 0.3.0
   -- text-based configuration enabled
CMake Error at CMakeLists.txt:262 (install):
  install Library TARGETS given no DESTINATION!
````

Please let me know what other information I can provide.

pomerlef, 2015-07-06:
Could you send the output of that command (from your build directory):
```
cmake -LAH
```
It's either the variable `CMAKE_INSTALL_PREFIX` or `INSTALL_LIB_DIR` that is empty. Given that the goal is to produce a `.sln` file, it shouldn't matter.

It might worth to consider having a `.sln` directly on Github. How other projects release their sources on Windows?

kwill, 2015-07-14:
Thanks for the feedback, LAH output follows and I can already identify several issues. I'm not sure what the typical or best approach is, but I have had some issues with prebaked `.sln` files in the past, so building them myself seems safer. Also, your instructions are very clear, which makes it easier.

```
CMake Error at CMakeLists.txt:25 (MESSAGE):
  32 bits compiler detected.  Libpointmatcher is only supported in 64 bits.

-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
CMake Error at C:/Program Files (x86)/CMake/share/cmake-3.2/Modules/FindBoost.cmake:1182 (message):
  Unable to find the requested Boost libraries.

  Unable to find the Boost header files.  Please set BOOST_ROOT to the root
  directory containing Boost or BOOST_INCLUDEDIR to the directory containing
  Boost's headers.
Call Stack (most recent call first):
  CMakeLists.txt:106 (find_package)

-- libnabo found, version 1.0.6
CMake Warning (dev) at CMakeLists.txt:191 (get_property):
  Policy CMP0026 is not set: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The LOCATION property should not be read from target "yaml-cpp-pm".  Use
  the target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

This warning is for project developers.  Use -Wno-dev to suppress it.

-- using built-in yaml-cpp, version 0.3.0
   -- text-based configuration enabled
CMake Error at CMakeLists.txt:262 (install):
  install Library TARGETS given no DESTINATION!

CMake Warning (dev) at CMakeLists.txt:330 (get_target_property):
  Policy CMP0026 is not set: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The LOCATION property should not be read from target "pointmatcher".  Use
  the target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Boost_INCLUDE_DIR (ADVANCED)
   used as include directory in directory C:/Data/Projects/libpointmatcher
   used as include directory in directory C:/Data/Projects/libpointmatcher
   used as include directory in directory C:/Data/Projects/libpointmatcher/examples
   used as include directory in directory C:/Data/Projects/libpointmatcher/examples
   used as include directory in directory C:/Data/Projects/libpointmatcher/examples
   used as include directory in directory C:/Data/Projects/libpointmatcher/examples
   used as include directory in directory C:/Data/Projects/libpointmatcher/examples
   used as include directory in directory C:/Data/Projects/libpointmatcher/examples
   used as include directory in directory C:/Data/Projects/libpointmatcher/utest
EIGEN_INCLUDE_DIR
   used as include directory in directory C:/Data/Projects/libpointmatcher
   used as include directory in directory C:/Data/Projects/libpointmatcher
   used as include directory in directory C:/Data/Projects/libpointmatcher/examples
   used as include directory in directory C:/Data/Projects/libpointmatcher/examples
   used as include directory in directory C:/Data/Projects/libpointmatcher/examples
   used as include directory in directory C:/Data/Projects/libpointmatcher/examples
   used as include directory in directory C:/Data/Projects/libpointmatcher/examples
   used as include directory in directory C:/Data/Projects/libpointmatcher/examples
   used as include directory in directory C:/Data/Projects/libpointmatcher/utest

-- Configuring incomplete, errors occurred!
See also "C:/Data/Projects/libpointmatcher/CMakeFiles/CMakeOutput.log".
See also "C:/Data/Projects/libpointmatcher/CMakeFiles/CMakeError.log".
[... 183 lines cut ...]

// Installation directory for header files
INSTALL_INCLUDE_DIR:PATH=include

// Installation directory for libraries
INSTALL_LIB_DIR:PATH=lib

// MSVC: Build with shared runtime libs (/MD)
MSVC_SHARED_RT:BOOL=ON

// MSVC: Build with single-threaded static runtime libs (/ML until VS .NET 2003)
MSVC_STHREADED_RT:BOOL=OFF

// To build shared (true) or static (false) library
SHARED_LIBS:BOOL=true

// Set to ON to look for OpenCL, if your libnabo was compiled with CL support
USE_OPEN_CL:BOOL=false

// Use system version of yaml-cpp rather than one packaged with libpointmatcher
USE_SYSTEM_YAML_CPP:BOOL=OFF

// The directory containing a CMake configuration file for libnabo.
libnabo_DIR:PATH=C:/Data/Projects/libnabo/build
```

kwill, 2015-07-14:
It looks like some of the above are false negatives as a result of running cmake from the command line (for example, complaining about 32-bit compilers). Here's the full output from when I hit "Configure" in the cmake-gui, if it helps...

[image]

```
Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
Boost version: 1.58.0
Found the following Boost libraries:
  thread
  filesystem
  system
  program_options
  date_time
Boost version: 1.58.0
Found the following Boost libraries:
  thread
  filesystem
  system
  program_options
  date_time
  chrono
-- libnabo found, version 1.0.6
CMake Warning (dev) at CMakeLists.txt:191 (get_property):
  Policy CMP0026 is not set: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The LOCATION property should not be read from target "yaml-cpp-pm".  Use
  the target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

This warning is for project developers.  Use -Wno-dev to suppress it.

-- using built-in yaml-cpp, version 0.3.0
   -- text-based configuration enabled
CMake Error at CMakeLists.txt:262 (install):
  install Library TARGETS given no DESTINATION!

CMake Warning (dev) at CMakeLists.txt:330 (get_target_property):
  Policy CMP0026 is not set: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The LOCATION property should not be read from target "pointmatcher".  Use
  the target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

This warning is for project developers.  Use -Wno-dev to suppress it.

Configuring incomplete, errors occurred!
```

kwill, 2015-07-14:
I recompiled everything from scratch (see https://github.com/ethz-asl/libpointmatcher/issues/102 for documentation tweaks) and encountered the same problem. Thoughts:

* I am using Visual Studio 2015 RC. I selected the VS 2015 x64 generator (default native compiler) for all projects.

* Are my `include` path values correct? Should they be `<source folder>` or `<source folder>\include`? I have used the former. If the latter, what should I do if there is no `include` folder in the source folder?

* All the dependencies have SHARED_LIBS or BUILD_SHARED_LIBS or similarly named variables set to False. Only in _libpointmatcher_ is SHARED_LIBS True. If I set SHARED_LIBS to False in _libpointmatcher_, configuration and generation succeed, but msbuild fails (see below).

* I noticed that the configuration/build seems to be referencing an "internal" version of yaml-cpp rather than the one I built previously.

```
       "C:\Data\Projects\libpointmatcher\build\libpointmatcher.sln" (default ta
       rget) (1) ->
       "C:\Data\Projects\libpointmatcher\build\examples\pmicp.vcxproj.metaproj"
        (default target) (9) ->
       "C:\Data\Projects\libpointmatcher\build\examples\pmicp.vcxproj" (default
        target) (19) ->
       (Link target) ->
         LINK : fatal error LNK1104: cannot open file '..\contrib\yaml-cpp-pm\R
       elWithDebInfo\libyaml-cpp-pm.lib' [C:\Data\Projects\libpointmatcher\buil
       d\examples\pmicp.vcxproj]

       "C:\Data\Projects\libpointmatcher\build\libpointmatcher.sln" (default ta
       rget) (1) ->
       "C:\Data\Projects\libpointmatcher\build\examples\align_sequence.vcxproj.
       metaproj" (default target) (3) ->
       "C:\Data\Projects\libpointmatcher\build\examples\align_sequence.vcxproj"
        (default target) (18) ->
         LINK : fatal error LNK1104: cannot open file '..\contrib\yaml-cpp-pm\R
       elWithDebInfo\libyaml-cpp-pm.lib' [C:\Data\Projects\libpointmatcher\buil
       d\examples\align_sequence.vcxproj]

       "C:\Data\Projects\libpointmatcher\build\libpointmatcher.sln" (default ta
       rget) (1) ->
       "C:\Data\Projects\libpointmatcher\build\examples\list_modules.vcxproj.me
       taproj" (default target) (8) ->
       "C:\Data\Projects\libpointmatcher\build\examples\list_modules.vcxproj" (
       default target) (20) ->
         LINK : fatal error LNK1104: cannot open file '..\contrib\yaml-cpp-pm\R
       elWithDebInfo\libyaml-cpp-pm.lib' [C:\Data\Projects\libpointmatcher\buil
       d\examples\list_modules.vcxproj]

       "C:\Data\Projects\libpointmatcher\build\libpointmatcher.sln" (default ta
       rget) (1) ->
       "C:\Data\Projects\libpointmatcher\build\examples\icp_simple.vcxproj.meta
       proj" (default target) (7) ->
       "C:\Data\Projects\libpointmatcher\build\examples\icp_simple.vcxproj" (de
       fault target) (21) ->
         LINK : fatal error LNK1104: cannot open file '..\contrib\yaml-cpp-pm\R
       elWithDebInfo\libyaml-cpp-pm.lib' [C:\Data\Projects\libpointmatcher\buil
       d\examples\icp_simple.vcxproj]

       "C:\Data\Projects\libpointmatcher\build\libpointmatcher.sln" (default ta
       rget) (1) ->
       "C:\Data\Projects\libpointmatcher\build\utest\utest.vcxproj.metaproj" (d
       efault target) (11) ->
       "C:\Data\Projects\libpointmatcher\build\utest\utest.vcxproj" (default ta
       rget) (17) ->
       (ClCompile target) ->
         C:\Data\Projects\libpointmatcher\utest\ui\DataFilters.cpp(74): error C
       2593: 'operator =' is ambiguous [C:\Data\Projects\libpointmatcher\build\
       utest\utest.vcxproj]
         C:\Data\Projects\libpointmatcher\utest\ui\DataFilters.cpp(117): error
       C2593: 'operator =' is ambiguous [C:\Data\Projects\libpointmatcher\build
       \utest\utest.vcxproj]
         C:\Data\Projects\libpointmatcher\utest\ui\DataFilters.cpp(157): error
       C2593: 'operator =' is ambiguous [C:\Data\Projects\libpointmatcher\build
       \utest\utest.vcxproj]
         C:\Data\Projects\libpointmatcher\utest\ui\DataFilters.cpp(194): error
       C2593: 'operator =' is ambiguous [C:\Data\Projects\libpointmatcher\build
[... 33 lines cut ...]
       rget) (1) ->
       "C:\Data\Projects\libpointmatcher\build\examples\compute_overlap.vcxproj
       .metaproj" (default target) (5) ->
       "C:\Data\Projects\libpointmatcher\build\examples\compute_overlap.vcxproj
       " (default target) (22) ->
       (Link target) ->
         LINK : fatal error LNK1104: cannot open file '..\contrib\yaml-cpp-pm\R
       elWithDebInfo\libyaml-cpp-pm.lib' [C:\Data\Projects\libpointmatcher\buil
       d\examples\compute_overlap.vcxproj]

       "C:\Data\Projects\libpointmatcher\build\libpointmatcher.sln" (default ta
       rget) (1) ->
       "C:\Data\Projects\libpointmatcher\build\examples\build_map.vcxproj.metap
       roj" (default target) (4) ->
       "C:\Data\Projects\libpointmatcher\build\examples\build_map.vcxproj" (def
       ault target) (23) ->
         LINK : fatal error LNK1104: cannot open file '..\contrib\yaml-cpp-pm\R
       elWithDebInfo\libyaml-cpp-pm.lib' [C:\Data\Projects\libpointmatcher\buil
       d\examples\build_map.vcxproj]

    174900 Warning(s)
    20 Error(s)

Time Elapsed 00:13:04.24
```

kwill, 2015-07-14:
I can't find a way to include this as just an attachment. In case it helps, here are my notes from the second attempt at compiling (everything up to setting up CMake GUI for libpointmatcher).

----

GREP
====

Installed and verified without issue.

BOOST
=====

Compiled without issue.

LIBNABO
=======

CMAKE-GUI OUTPUT WHEN CREATING libnabo.sln:

Note: SHARED_LIBS is not ticked by default, I left it unticked.

Configuration and generation succeed, but with warnings (output below includes debug output and dev warnings):

```
Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
Boost version: 1.58.0
   Entering             C:/Data/Projects/libnabo/examples
   Returning to         C:/Data/Projects/libnabo
   Entering             C:/Data/Projects/libnabo/tests
ANN not found, disabling it in benchmarks
FLANN not found, disabling it in benchmarks
   Returning to         C:/Data/Projects/libnabo
   Entering             C:/Data/Projects/libnabo/python
Could NOT find PythonInterp (missing:  PYTHON_EXECUTABLE) 
Python version from interpreter is 
Could NOT find PythonLibs (missing:  PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS) 
Python libs or executable not found, skipping Python bindings
   Returning to         C:/Data/Projects/libnabo
CMake Warning (dev) at CMakeLists.txt:159 (get_property):
  Policy CMP0026 is not set: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The LOCATION property should not be read from target "nabo".  Use the
  target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

This warning is for project developers.  Use -Wno-dev to suppress it.

Configuring done
CMake Warning (dev) at CMakeLists.txt:124 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "doc" of target "nabo" does not exist.
This warning is for project developers.  Use -Wno-dev to suppress it.

Generating done
```

msbuild of .sln succeeds with 20844 warnings and 0 errors.

YAML-CPP
========

Note: BUILD_SHARED_LIBS is not ticked by default, I left it unticked.

Configure, generate and msbuild without issue. (Builds with 10 warnings, 0 errors.)

GTEST
=====

Note: BUILD_SHARED_LIBS not ticked, left unticked.

Configure and generate succeeded without Python interpreter.

```
Could NOT find PythonInterp (missing:  PYTHON_EXECUTABLE) 
Configuring done
Generating done
```

Configure, generate and msbuild without issue. (Builds with 0 warnings, 0 errors.)

LIBPOINTMATCHER
===============

Note: SHARED_LIBS is ticked, left ticked.

Configuration fails, cannot proceed to generation or msbuilding. Configuration output with debug output and dev warnings:

```
Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
   Entering             C:/Data/Projects/libpointmatcher/contrib
   Entering             C:/Data/Projects/libpointmatcher/contrib/yaml-cpp-pm
   Returning to         C:/Data/Projects/libpointmatcher/contrib
   Returning to         C:/Data/Projects/libpointmatcher
Boost version: 1.58.0
Found the following Boost libraries:
  thread
  filesystem
  system
  program_options
  date_time
Boost version: 1.58.0
Found the following Boost libraries:
  thread
  filesystem
  system
  program_options
  date_time
  chrono
-- libnabo found, version 1.0.6
CMake Warning (dev) at CMakeLists.txt:191 (get_property):
  Policy CMP0026 is not set: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The LOCATION property should not be read from target "yaml-cpp-pm".  Use
  the target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

This warning is for project developers.  Use -Wno-dev to suppress it.

-- using built-in yaml-cpp, version 0.3.0
   -- text-based configuration enabled
CMake Error at CMakeLists.txt:262 (install):
  install Library TARGETS given no DESTINATION!

   Entering             C:/Data/Projects/libpointmatcher/examples
   Returning to         C:/Data/Projects/libpointmatcher
   Entering             C:/Data/Projects/libpointmatcher/evaluations
   Returning to         C:/Data/Projects/libpointmatcher
   Entering             C:/Data/Projects/libpointmatcher/utest
   Returning to         C:/Data/Projects/libpointmatcher
CMake Warning (dev) at CMakeLists.txt:330 (get_target_property):
  Policy CMP0026 is not set: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The LOCATION property should not be read from target "pointmatcher".  Use
  the target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

This warning is for project developers.  Use -Wno-dev to suppress it.

Configuring incomplete, errors occurred!
See also "C:/Data/Projects/libpointmatcher/build/CMakeFiles/CMakeOutput.log".
See also "C:/Data/Projects/libpointmatcher/build/CMakeFiles/CMakeError.log".
```

Error message that persists without debug output and dev warnings:

```
-- libnabo found, version 1.0.6
-- using built-in yaml-cpp, version 0.3.0
   -- text-based configuration enabled
CMake Error at CMakeLists.txt:262 (install):
  install Library TARGETS given no DESTINATION!
```

kwill, 2015-08-03:
This can be fixed by https://github.com/kwill/libpointmatcher/commit/b26e4d4d34e92eedff412b8d4885eae1bd23fd76

HannesSommer, 2015-08-03:
This looks good to me. 
Would you like to issue a pull request for this commit only (i.e. it has to be rebased or cherry-picked) maybe plus some of your other commits that improve mingw compatibility)? 
I can do that otherwise for this commit.

kwill, 2015-08-03:
I have had to switch back to Visual Studio for other reasons, so I've pushed my MinGW rough work for now. Due to project deadlines I'm not sure when I'll get a chance to make some clean pull requests, but I'll try :) Anyone with the time to get there first should feel welcome.

pomerlef, 2015-09-09:
This was fixed with the commit d0453a50b6a8ba4804202e2aa1e5e1cca9d874fa (although the commit message says otherwise...) following kwill patch.

closed 2015-09-09 by pomerlef