norlab-ulaval/libpointmatcher #104: Windows/MSVC: Cannot open file 'contrib\yaml-cpp-pm\RelWithDebInfo\libyaml-cpp-pm.lib'

kwill, 2015-08-03:
**Steps to reproduce:**

Building libpointmatcher with MSVC following the current instructions, excluding the instructions to use an external yaml-cpp and external gtest

**What should happen:**

Build succeeds (C++ compatibility issues anticipated)

**What happens instead:**

I get the following error:

       "C:\Users\username\Documents\registration_builds\libpointmatcher-msbuil
       d\libpointmatcher.sln" (default target) (1) ->
       "C:\Users\username\Documents\registration_builds\libpointmatcher-msbuil
       d\pointmatcher.vcxproj.metaproj" (default target) (10) ->
       "C:\Users\username\Documents\registration_builds\libpointmatcher-msbuil
       d\pointmatcher.vcxproj" (default target) (16) ->
       (Link target) ->
         LINK : fatal error LNK1104: cannot open file 'contrib\yaml-cpp-pm\RelW
       ithDebInfo\libyaml-cpp-pm.lib' [C:\Users\username\Documents\registratio
       n_builds\libpointmatcher-msbuild\pointmatcher.vcxproj]

    133065 Warning(s)
    1 Error(s)

I have a file `<build>\contrib\yaml-cpp-pm\RelWithDebInfo\libyaml-cpp-pmmd.lib` not `libyaml-cpp-pm.lib`

**Workaround:**

Make a copy of `libyaml-cpp-pmmd.lib` with the correct name, and build again.

HannesSommer, 2015-08-03:
To contributed a yaml copy is only a temporary hack. This is why we haven't cleaned up (or even tested ?) the yaml-cpp-pm/CMakeLists.txt file for windows. 

The line https://github.com/ethz-asl/libpointmatcher/blob/master/contrib/yaml-cpp-pm/CMakeLists.txt#L116 is apparently introducing this suffix. The best way is probably to get rid of this fine-grained support of different co-existing output files and rather always use the same (without any extra suffix). What happens if you just remove the two ```if(MSVC)``` blocks? Does it work for you?

kwill, 2015-08-04:
That worked! I've included it in my clean branch.

HannesSommer, 2015-08-04:
Great! And thanks for testing!

pomerlef, 2015-09-09:
Should I remove those two blocks on the master?

kwill, 2015-09-10:
I can confirm that it worked for me. Apologies for not submitting my branch as a pull request yet, but feel free to take anything useful from it :)

closed 2015-09-10 by pomerlef