norlab-ulaval/libpointmatcher #105: Windows/MSVC: Error 'operator =' is ambiguous

kwill, 2015-08-03:
**Steps to reproduce:**

Build on Windows/MSVC:

**What should happen:**

Build successful.

**What happens instead:**

    "C:\Users\username\Documents\registration_builds\libpointmatcher-msbuil
    d\libpointmatcher.sln" (default target) (1) ->
    "C:\Users\username\Documents\registration_builds\libpointmatcher-msbuil
    d\examples\build_map.vcxproj.metaproj" (default target) (4) ->
    "C:\Users\username\Documents\registration_builds\libpointmatcher-msbuil
    d\examples\build_map.vcxproj" (default target) (23) ->
      LINK : fatal error LNK1104: cannot open file '..\RelWithDebInfo\pointm
    atcher.lib' [C:\Users\username\Documents\registration_builds\libpointma
    tcher-msbuild\examples\build_map.vcxproj]

    49592 Warning(s)
    20 Error(s)

This is because of several C++ compatibility errors like this one:

    Error	396	error C2593: 'operator =' is ambiguous	C:\Users\username\Documents\registration\libpointmatcher-vs\utest\ui\Matcher.cpp	55	1	utest

**Fix (from François Pomerleau via email):**

The second type of error is "error C2593: 'operator =' is ambiguous". This happen for example here in the source code: https://github.com/ethz-asl/libpointmatcher/blob/master/utest/ui/DataFilters.cpp#L74

They are all trigger on the same kind of semantic which looks like that:

Original source:

    params = map_list_of<string,string>
    		("dim","0")
    		("maxDist", toParam(6.0))
    	;

You can try the following to see if it resolve the error for the line 74:

    params = PM::Parameters(
    		map_list_of<string,string>
    		("dim","0")
    		("maxDist", toParam(6.0))
    	);

If it does, you can do the same to all error lines and submit a merge request. I'll test it on Ubuntu and MacOs.

pomerlef, 2015-09-09:
The commit a021a1ab482bbd49faed52b2e25e554dc1a98cba should fix this problem.

The proposed fix was triggering another error on ubuntu so it was simpler to just explicitly call the map.

```
params = PM::Parameters();
params["dim"] = "0";
params["maxDist"] = toParam(6.0);
```

@kwill if you could confirm that it work on Windows, I'll close it.

kwill, 2015-09-10:
It'll be a little while till I can confirm. I've bookmarked this issue and will come back to it.

pomerlef, 2016-01-19:
Since somebody else was able to build it on Windows, I'll assume that this is resolved.

closed 2016-01-19 by pomerlef

kwill, 2016-01-25:
Hurrah!