norlab-ulaval/libpointmatcher #106: Error in align_sequence example

norouzzadeh, 2015-08-12:
While running the align_sequence example, i got two errors:
The first one is
./align_sequence ../../examples/data/default.yaml ../../examples/data/carCloudList.csv 
---------------------
Loading: ../../examples/data/car_cloud400.csv
---------------------
Loading: ../../examples/data/car_cloud401.csv
writing to pointmatcher-run1-iterationInfo.csv
match ratio: 0.75008
terminate called after throwing an instance of 'PointMatcherSupport::Parametrizable::InvalidParameter'
  what():  Parameter binSize for module SurfaceNormalDataPointsFilter was set but is not used
Aborted (core dumped)

I changed the line 108 from ("binSize", "10") to ("knn", "10") and the first error was solved.
Then the following error occured:

./align_sequence ../../examples/data/default.yaml ../../examples/data/carCloudList.csv 
---------------------
Loading: ../../examples/data/car_cloud400.csv
---------------------
Loading: ../../examples/data/car_cloud401.csv
writing to pointmatcher-run1-iterationInfo.csv
match ratio: 0.75008
outputFileName: ./align_sequence_1
terminate called after throwing an instance of 'std::runtime_error'
  what():  saveAnyFormat(): Unknown extension "" for file "./align_sequence_1", extension must be either ".vtk" or ".csv"
Aborted (core dumped)

Where i changed the line 151 from outputFileNameIter << outputFileName << "_" << i;
to outputFileNameIter << outputFileName << "_" << i << ".vtk";

Now it works.

pomerlef, 2015-08-12:
Great, thanks for reporting. Indeed, this example was outdated. I'll do the modification in the next hour.

If you find more solutions, you can also directly submit a merge request.

pomerlef, 2015-08-12:
Fixed with commit ea51643ca53f53509dd59984b76b837e024cd3ac

closed 2015-08-12 by pomerlef