norlab-ulaval/libpointmatcher #111: Building on Windows

aramgev, 2015-08-31:
I try build on windows, but get following errors:

Make Error at CMakeLists.txt:25 (MESSAGE):
  32 bits compiler detected.  Libpointmatcher is only supported in 64 bits.

Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
Boost version: 1.49.0
Found the following Boost libraries:
  thread
  filesystem
  system
  program_options
  date_time
Boost version: 1.49.0
Found the following Boost libraries:
  thread
  filesystem
  system
  program_options
  date_time
  chrono
CMake Error at CMakeLists.txt:134 (find_package):
  By not providing "Findlibnabo.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "libnabo", but
  CMake did not find one.

  Could not find a package configuration file provided by "libnabo" with any
  of the following names:

    libnaboConfig.cmake
    libnabo-config.cmake

  Add the installation prefix of "libnabo" to CMAKE_PREFIX_PATH or set
  "libnabo_DIR" to a directory containing one of the above files.  If
  "libnabo" provides a separate development package or SDK, be sure it has
  been installed.

kwill, 2015-09-03:
Quick comments:
* You must build with a 64-bit compiler. It should be called something like _Visual Studio 12 2013 Win64_
* It looks like you haven't generated a MSVC project for libnabo. Once you have done this `libnabo_DIR` should be set automatically to the libnabo MSVC project folder. If it isn't set it to that folder yourself (the folder containing `libnabo.sln`)
* libnabo and libpointmatcher have not been updated for MSVC recently. Please feel free to use my work-in-progress branches [1] [2] that include relevant fixes. If they work for you and you have time to submit them as a clean pull-request to the upstream branch, please feel free to do so (or not).

[1] https://github.com/kwill/libnabo/tree/visual-studio-attempt

[2] https://github.com/kwill/libpointmatcher/tree/visual-studio-attempt

aramgev, 2015-09-05:
Thank you, for answer.

I make your fixes, but now I get this error:

-- using built-in yaml-cpp, version 0.3.0
   -- text-based configuration enabled
CMake Error at CMakeLists.txt:262 (install):
  install Library TARGETS given no DESTINATION!

[image]

kwill, 2015-09-06:
See https://github.com/ethz-asl/libpointmatcher/issues/101 (commit that fixes this referenced right at the end)

aramgev, 2015-09-07:
Thanks, it works)

pomerlef, 2015-09-07:
thanks @kwill for the support!

aramgev, 2015-09-08:
But I have another problem :(

When I build solution in VS2010, I get these errors:

Error	76746	error LNK1104: cannot open file 'pointmatcherlib.lib'	
Error	79277	error LNK1104: cannot open file '..\RelWithDebInfo\pointmatcher.lib'	
Error	80869	error LNK1104: cannot open file '..\RelWithDebInfo\pointmatcher.lib'	
Error	82252	error LNK1104: cannot open file '..\RelWithDebInfo\pointmatcher.lib'	
Error	100475	error LNK1104: cannot open file '..\RelWithDebInfo\pointmatcher.lib'	
Error	111526	error LNK1104: cannot open file '..\RelWithDebInfo\pointmatcher.lib'

pomerlef, 2015-09-09:
Check if the fix in #104 work for you.

aramgev, 2015-09-16:
I got rid of this problem, but know when building I get these errors:

LINK : fatal error LNK1104: cannot open file '..\RelWithDebInfo\pointmatcher.lib'

I think it's because, pointmatcher.sln after build generates .dll not .lib

aramgev, 2015-09-16:
I got rid of this problem, but when building I get these errors:

LINK : fatal error LNK1104: cannot open file '..\RelWithDebInfo\pointmatcher.lib'

It's because, pointmatcher.sln after build generates .dll not .lib changig it in configuration, it compiles.

kwill, 2015-09-16:
Make sure you have SHARED_LIBS *un*ticked.

syedharoonalam, 2016-01-12:
I facing same problem of: cannot open file '..\RelWithDebInfo\pointmatcher.lib'
I also tried changing .dll in configuration and untick SHARED_LIBS.

Please advice.

syedharoonalam, 2016-01-12:
[image]

I believe the reason projects are not able to find pointmatcher.lib is because of error "redefinition different type modifiers" in DataPoints.cpp, any suggestion regarding this and "CurrentBibliography: ambiguous symbol" in icp_advance_api ?

pomerlef, 2016-01-12:
Your linking problem is clearly related to the fact that pointmatcher.lib was not built so all the example executables cannot link.

Your first error "redefinition different type modifiers" is more concerning since it is related to a recent change in the code. Can your compiler provide more details about that error?

syedharoonalam, 2016-01-13:
For first error: "redefinition different type modifiers", it showed redefinition of 'getTimeViewByName' in DataPoints.cpp in line 648, 655. I changed "..TimeConstView PointMatcher<T>::DataPoints::getTimeViewByName" return type to 'const' as well so its now generating pointmatcher.lib hence all examples are now able to link it.

The second error: ambiguous symbol is I think confusion of compiler with CurrentBibliography and  ::CurrentBibliography, I simply changed typedef PointMatcherSupport::CurrentBibliography with different name and its usage in same file. Now project is compiling without any errors. 

However there is now crash in icp_simple and in other variations when feeding sample data. I can try debugging it and will keep you posted, if anyone has encounter it please post here as well.
[image]

pomerlef, 2016-01-13:
That's great that you manage to make it work on your side. Could you send a pull request or the exact changes (with line numbers) you did to make it work on your side without the executable modifs. I would like to test them on Ubuntu and Mac on my side, so I could add them to the main branch.

As for the test executables, if I understand well, it's the line

 [ref.save("test_ref.vtk");](https://github.com/ethz-asl/libpointmatcher/blob/master/examples/icp_simple.cpp#L75) 

and similar that make it crash on Windows. I would expect a clearer error message as we explicitly check is the file could open or not: [InspectorsImpl.cpp#L685](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/InspectorsImpl.cpp#L685). Is the error you received is equivalent to a segfault? If so, do you have access to a debugger telling you what line in the save function is faulty?

Thanks for pushing on your side for a working Windows version.

syedharoonalam, 2016-01-14:
I have pulled request of changes I have made.

Yes for saving its similar line where it crashes. I am not sure if its a segfault as I can save the file at same location given complete path e.g: "D:\\...\\test_ref.vtk" otherwise with only "test_ref.vtk" it gets save when debugging (hence not getting error message) but not when execute outside environment.

pomerlef, 2016-01-14:
The request was merged with some modifications. Thanks for digging into the code to make it work. 

Before I close this issue, could you send your system information (something in the range of [this table](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/CompilationWindows.md#in-short)) so we can track what work and what doesn't?

syedharoonalam, 2016-01-15:
Sure,

I am using Windows 10 64bit
Built libpointermatcher, libnabo from sources
Visual Studio 2010
Cmake 3.3 
Eigen 3.0.5
Boost 1.55
yaml-cpp 0.3.0
grep 2.5.4
gtest 1.7

closed 2016-01-15 by pomerlef