norlab-ulaval/libpointmatcher #113: Maximum sampling ratio

oleg-alexandrov, 2015-09-15:
I found it somewhat odd that in a few places the sampling ratio for filters is restricted to 0.9999999. Why not just use 1.0? What if the user wants to use all points, without artificially sampling?

I tried this change (https://github.com/oleg-alexandrov/libpointmatcher/commit/9082e5c99c1860e18873fed2e832e3de43aeab72) and all tests pass.

pomerlef, 2015-09-16:
I think the idea, at that time, was that the user should not use a subsampling filter if he wants to keep all the points. But we might have overthinking it and I can see some situations where it's easier just to set the sampling parameter to one.

Should we avoid to loop through the points if the sampling parameter is set to one?

oleg-alexandrov, 2015-09-16:
Probably allowing the sampling parameter to be 1 would make libpointmacher more robust if used as part of a larger project. 

The loop through the points may not be avoidable at all times, for example, SamplingSurfaceNormalDataPointsFilter also computes normals along the way that are used in point-to-plane alignment. 

I'd be happy to submit a patch if any agreement is achieved on how to proceed.

pomerlef, 2015-09-16:
You can a merge request from your personal commit. I'll accept it.

closed 2015-09-17 by pomerlef