norlab-ulaval/libpointmatcher #118: Building on windows fails: Determining if the _POSIX_TIMERS exist failed

mkaze, 2015-09-19:
I wanted to build the libnado according to instructions provided [here](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/CompilationWindows.md#build-libnabo). Unfortunately it failed at step 9. Here is the cmake output:

```
Boost version: 1.59.0
CMake Error at CMakeLists.txt:25 (message):
  Cannot grep version number: The system cannot find the file specified

Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
Boost version: 1.59.0
OpenCL disabled, not looking for it
ANN not found, disabling it in benchmarks
FLANN not found, disabling it in benchmarks
CMake Error at tests/CMakeLists.txt:95 (message):
  Compiling a test executable failed.
Call Stack (most recent call first):
  tests/CMakeLists.txt:105 (try_compile_cloud_types)

Configuring incomplete, errors occurred!
See also "C:/ICP/libnabo/build/CMakeFiles/CMakeOutput.log".
See also "C:/ICP/libnabo/build/CMakeFiles/CMakeError.log".
```

And here is CMakeError.log contents:

```
Determining if the _POSIX_TIMERS exist failed with the following output:
Change Dir: C:/ICP/libnabo/build/CMakeFiles/CMakeTmp

Run Build Command:"C:/Program Files (x86)/MSBuild/12.0/bin/MSBuild.exe" "cmTC_39dad.vcxproj" "/p:Configuration=Debug" "/p:VisualStudioVersion=12.0"
Microsoft (R) Build Engine version 12.0.31101.0

[Microsoft .NET Framework, version 4.0.30319.34209]

Copyright (C) Microsoft Corporation. All rights reserved.

Build started 9/19/2015 5:07:53 PM.

Project "C:\ICP\libnabo\build\CMakeFiles\CMakeTmp\cmTC_39dad.vcxproj" on node 1 (default targets).

PrepareForBuild:

  Creating directory "cmTC_39dad.dir\Debug\".

  Creating directory "C:\ICP\libnabo\build\CMakeFiles\CMakeTmp\Debug\".

  Creating directory "cmTC_39dad.dir\Debug\cmTC_39dad.tlog\".

InitializeBuildStatus:

  Creating "cmTC_39dad.dir\Debug\cmTC_39dad.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.

ClCompile:

  C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\bin\CL.exe /c /Zi /W3 /WX- /Od /Ob0 /Oy- /D WIN32 /D _WINDOWS /D _DEBUG /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Fo"cmTC_39dad.dir\Debug\\" /Fd"cmTC_39dad.dir\Debug\vc120.pdb" /Gd /TC /analyze- /errorReport:queue C:\ICP\libnabo\build\CMakeFiles\CMakeTmp\CheckSymbolExists.c

  Microsoft (R) C/C++ Optimizing Compiler Version 18.00.31101 for x86

  Copyright (C) Microsoft Corporation.  All rights reserved.

  

  cl /c /Zi /W3 /WX- /Od /Ob0 /Oy- /D WIN32 /D _WINDOWS /D _DEBUG /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /RTC1 /MDd /GS /fp:precise /Zc:wchar_t /Zc:forScope /Fo"cmTC_39dad.dir\Debug\\" /Fd"cmTC_39dad.dir\Debug\vc120.pdb" /Gd /TC /analyze- /errorReport:queue C:\ICP\libnabo\build\CMakeFiles\CMakeTmp\CheckSymbolExists.c

  

  CheckSymbolExists.c

C:\ICP\libnabo\build\CMakeFiles\CMakeTmp\CheckSymbolExists.c(2): fatal error C1083: Cannot open include file: 'unistd.h': No such file or directory [C:\ICP\libnabo\build\CMakeFiles\CMakeTmp\cmTC_39dad.vcxproj]

Done Building Project "C:\ICP\libnabo\build\CMakeFiles\CMakeTmp\cmTC_39dad.vcxproj" (default targets) -- FAILED.

Build FAILED.

"C:\ICP\libnabo\build\CMakeFiles\CMakeTmp\cmTC_39dad.vcxproj" (default target) (1) ->

(ClCompile target) -> 

  C:\ICP\libnabo\build\CMakeFiles\CMakeTmp\CheckSymbolExists.c(2): fatal error C1083: Cannot open include file: 'unistd.h': No such file or directory [C:\ICP\libnabo\build\CMakeFiles\CMakeTmp\cmTC_39dad.vcxproj]

    0 Warning(s)

    1 Error(s)

Time Elapsed 00:00:00.48

File C:/ICP/libnabo/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <unistd.h>
#include <time.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef _POSIX_TIMERS
  return ((int*)(&_POSIX_TIMERS))[argc];
#else
  (void)argc;
  return 0;
#endif
}

```

I am using Windows 8.1 pro(x64) and VS2013. I think the problem is caused by 'unistd.h' which is a header file in Unix, however I'm not sure if this is a bug or a problem with my configurations.

Note: CMake couldn't find the boost path (which resulted in an error like : "Boost_INCLUDE_DIR not found"), so I added these lines at the beginning of 'CMakeLists.txt':

```
SET(CMAKE_INCLUDE_PATH ${CMAKE_INCLUDE_PATH} "C:/ICP/boost_1_59_0")
SET(CMAKE_LIBRARY_PATH ${CMAKE_LIBRARY_PATH} "C:/ICP/boost_1_59_0/build")

FIND_PACKAGE(Boost)
IF (Boost_FOUND)
    INCLUDE_DIRECTORIES(${Boost_INCLUDE_DIR})
    ADD_DEFINITIONS( "-DHAS_BOOST" )
ENDIF()
```

pomerlef, 2015-09-20:
Hi mkase,

So I see three problems in your CMake outputs:

1. This error is reported from the line 25, but in the latest version of libnabo, it should be on [line 16](https://github.com/ethz-asl/libnabo/blob/master/CMakeLists.txt#L16). Do you have the latest version of libnabo? Could you check if that line match your local version?
  ```
  CMake Error at CMakeLists.txt:25 (message):
    Cannot grep version number: The system cannot find the file specified
  ```
2. Did you add manually a PATH variable named BOOST_ROOT through __CMake Gui__? This should avoid you to change the `CMakeFile.txt` manually. See [step 7](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/CompilationWindows.md#build-libnabo).

  > Note: CMake couldn't find the boost path (which resulted in an error like : "Boost_INCLUDE_DIR not found"), so I added these lines at the beginning of 'CMakeLists.txt':
3. The last one is interesting and it's related to cmake not being able to build `CheckSymbolExists.c`. I searched for that a bit and it's seems to be often related to another problem (like boost wrongly linked).

mkaze, 2015-09-20:
Thanks for your reply.

Regarding the problems you mentioned:

1. The difference in line numbers is due to the fact that I have added the lines for finding boost path at the beginning of 'CMakeLists.txt'.

2. Yes. I followed the instructions step by step and added the BOOST_ROOT variable but even by doing so, it resulted in the the error that I mentioned. I read about this solution [here](http://stackoverflow.com/questions/3808775/cmake-doesnt-find-boost).

3. I'm not sure about that either. I followed the steps in instruction carefully and did them from beginning more than five times but I couldn't make it.

Maybe building the boost library again solve the problem. I would try that and let you know if any changes happen.

pomerlef, 2015-09-22:
Could you report all the versions you have on your system based on [that table](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/CompilationWindows.md#in-short) and other you think would be useful. That might help others.

I'm seeing a lot of similar questions on stackoverflow without any answers. It might be something trigger by the latest version of CMake or different version of .NET framework, etc.  I'm also seeing `"C:/Program Files (x86)/MSBuild/12.0/bin/MSBuild.exe`, should it be `(x64)`? But all that are only speculations, I'm not expert enough on Windows system to help you directly.

drhalftone, 2015-12-15:
I'm seeing this exact issue on Mac OS X.  I got past it by commenting out the tests subdirectory, but it would to see this resolved.

pomerlef, 2016-01-19:
I just realized that this problem is related to [libnabo](https://github.com/ethz-asl/libnabo) and not libpointmatcher. The issue was transferred to them [here](https://github.com/ethz-asl/libnabo/issues/45) so I'll close that one on our side.

closed 2016-01-19 by pomerlef

jonntw, 2017-02-08:
@mkaze I have the same problem but I still have not found the solution to it.

CMake Error at tests/CMakeLists.txt:95 (message):
  Compiling a test executable failed.

I just want to know how did you solve this problem at the end? Thank you so much!

dagata-mining, 2020-08-19:
For anyone stock with the Build of Libnabo on windows with the "Cannot grep version number:" error, you can contour this error by 
removing the following lines in the CMakeLists.txt file of the Libnabo:
**
if (NOT GREP_VERSION_RESULT EQUAL 0)
	message(SEND_ERROR "Cannot grep version number: ${GREP_VERSION_RESULT}")
endif (NOT GREP_VERSION_RESULT EQUAL 0)
string(REGEX REPLACE ".*\"(.*)\".*" "\\1" PROJECT_VERSION "${PROJECT_VERSION}" )
** 

The reason is that GREP is not a recognized command in windows.