norlab-ulaval/libpointmatcher #126: Linking Projects to libpointmatcher using Eclipse

aman1300, 2015-10-16:
I followed the tutorial how to compile libpointmatcher on Ubuntu, I'm using Linux Mint and it compiled correctly on my system. But I'm having problem to link this project "libpointmatcher" to Eclipse. I created a new C++ project in Eclipse and configure the project settings. I saved the configuration and create the Demo.cpp. When I build the project, it didn't compiled succesfully. It gave me this error: 

/usr/local/include/pointmatcher/PointMatcher.h:43:27: fatal error: Eigen/StdVector: No such file or directory
 #include "Eigen/StdVector"

pomerlef, 2015-10-16:
I'm familiar with the Eclipse development tool, but I think you can generate an Eclipse project file through cmake. It seems that there is [many examples around](http://codeyarns.com/2014/01/15/how-to-create-eclipse-project-from-cmake-project/).

In your case, you're missing the dependency with Eigen include folder. There are some examples on how to link libpointmatcher to an external project using [cmake](https://github.com/ethz-asl/libpointmatcher/tree/master/examples/demo_cmake) and [QT creator](https://github.com/ethz-asl/libpointmatcher/tree/master/examples/demo_Qt). You can check there to make sure you have all the dependency. The file [demo.pro](https://github.com/ethz-asl/libpointmatcher/blob/master/examples/demo_Qt/demo.pro) is more explicite on how to list manually all the dependencies.

If you manage to make it work, don't hesitate to post back. It might help other. We could also add an demo_eclipse folder.

pomerlef, 2015-10-16:
ok, I just realized that we have a tutorial page on that and that you were referencing to it. It seems that there is missing information about Eigen dependency (only for the `C++ Compiler > Includes`).

If that work, could you send the correction of the text to make sure it is accurate?

aman1300, 2015-10-29:
Hi François,

Thank you very much for the fast replay. I'm sorry that I didn't answer
sooner, I was a little busy.

I followed the tutorial of the Pointmatcher libary because I'm working on a
school project and I need a fast ICP that
can works on 2D.

I have found some algorithms but it wasn't complete or the outcome wasn't
right. So that's why I
found the pointmatcher library.

I don't know if I can use ur program, it is a realy big program and it come
with a lot of dependency.

Do you have a smaller program for ICP that can works for 2D? Or can you
help me pointing
to the right direction. I'm feeling I'm on the dead end here.

Thanks for your attention and time,

Yours Sincerely,

Mark Amante

pomerlef, 2015-10-29:
The library has one of the smallest footprint in term of dependencies when compared to other point cloud libraries. Eigen is now standard for anything implicating math in C++ and we had to rely on Boost for the multi platform capabilities.

That being said, the library is intended for developers and scientifics working on improving registration solutions. There are many options and parameters to tune to achieve a precise and fast solution, which require a good background knowledge. For school project, you might want to some of the [Matlab scripts available around](http://www.mathworks.com/matlabcentral/fileexchange/27804-iterative-closest-point).

Good luck with your project!

closed 2015-10-29 by pomerlef