norlab-ulaval/libpointmatcher #129: Segfault while loading PCD files

hirav, 2015-12-06:
Instead of using csv file I tried to load PCD files directly using 
const DP ref(DP::load(argv[1]));

and then also using 
const DP ref(PMIO::loadPCD(argv[1]));

Both give segfault with .pcd files passed as arguments but works with .csv and .vtk files

pomerlef, 2015-12-06:
Could you give me a short test file to replicate the problem on my side?

hirav, 2015-12-06:
This zip contains a simple pcd file in ASCII format and sample code from my project which works for csv and vtk files. Also it doesnt work on binary .pcd files.
[test.zip](https://github.com/ethz-asl/libpointmatcher/files/53265/test.zip)

hirav, 2015-12-06:
Dear Francois,

I have provide the short hand file on github. I need to complete my project
in next 24 hours, so please let me know the fix or any other alternative to
load pcd file(in binary format) either by converting them or any other way.

Regards,
Hirav

pomerlef, 2015-12-06:
The download link doesn't work.

Also, as stated [here](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/ImportExport.md), we only support ascii format.

hirav, 2015-12-06:
Even ASCII doesn't work. I am able to access the link but let me upload the files again
[test.zip](https://github.com/ethz-asl/libpointmatcher/files/53321/test.zip)

pomerlef, 2015-12-06:
I received the file. I'll have a look at the problem tomorrow morning.

hirav, 2015-12-07:
Thanks for the quick response 
BTW can you tell me if my syntax is correct for the below 2 lines

pomerlef, 2015-12-07:
There was indeed a bug. It is fix with the commit 2fabd0dea9c1294ba22fd2512d17d848de536180. I added your PCD example in our unit test to avoid this problem in the future. 

Both lines are doing the same thing:
- `DP::load(...)` looks at the extension of the file to define which specific function to call. This should be the prefered call for most of the users.
-  `PMIO::loadPCD(...)`  parses a file assuming that it is a PCD file. It is considered a low level call.

Thanks for reporting.

closed 2015-12-07 by pomerlef