norlab-ulaval/libpointmatcher #130: Building point matcher on Mac OS X...

drhalftone, 2015-12-15:
In order to get libnabo to build past cmake, I commented out the tests directory. Moving forward and compiling point matcher, I got the following error:

Scanning dependencies of target icp_advance_api
[ 76%] Building CXX object examples/CMakeFiles/icp_advance_api.dir/icp_advance_api.cpp.o
In file included from /Users/dllau/Libraries/libpointmatcher/examples/icp_advance_api.cpp:36:
In file included from /usr/local/include/pointmatcher/PointMatcher.h:43:
In file included from /usr/local/include/eigen3/Eigen/StdVector:14:
In file included from /usr/local/include/eigen3/Eigen/Core:373:
/usr/local/include/eigen3/Eigen/Eigen2Support:20:2: warning: "Eigen2 support is deprecated in Eigen 3.2.x and it will be removed in
      Eigen 3.3. (Define EIGEN_NO_EIGEN2_DEPRECATED_WARNING to disable this warning)" [-W#warnings]
#warning "Eigen2 support is deprecated in Eigen 3.2.x and it will be removed in Eigen 3.3. (Define EIGEN_NO_EIGEN2_DEPRECATED...
 ^
/Users/dllau/Libraries/libpointmatcher/examples/icp_advance_api.cpp:208:78: error: 'getMatchedPoints' is a protected member of
      'PointMatcher<float>::ErrorMinimizer'
        const PM::ErrorMinimizer::ErrorElements matchedPoints = icp.errorMinimizer->getMatchedPoints( data_out, ref, matches...
                                                                                    ^
/usr/local/include/pointmatcher/PointMatcher.h:491:18: note: declared protected here
                ErrorElements& getMatchedPoints(const DataPoints& reading, const DataPoints& reference, const Matches& matche...
                               ^
1 warning and 1 error generated.
make[2]: *** [examples/CMakeFiles/icp_advance_api.dir/icp_advance_api.cpp.o] Error 1
make[1]: *** [examples/CMakeFiles/icp_advance_api.dir/all] Error 2
make: *** [all] Error 2

drhalftone, 2015-12-16:
Sorry, the compiler was linking to headers from a previous installation.  So by deleting the old headers files in the /usr/local/include directory, all my errors went away.

closed 2015-12-16 by pomerlef

drhalftone, 2015-12-16:
Is it because I copied extra headers to my usr/local/include directory that I ran into this error or would this have happened simply because I had a previous installation of pointmatcher?

Closed #130 <https://github.com/ethz-asl/libpointmatcher/issues/130>.

pomerlef, 2015-12-16:
I don't try the installation procedure often. I'm assuming that it should fetch the local files in priority, but I might be wrong.