norlab-ulaval/libpointmatcher #131: ICP for only slightly overlapping clouds

85pando, 2015-12-16:
I have the following problem:

My setting is, that I have two Sensors which create a 2D point cloud and only overlap in a small part of their data points. I can not be completely sure about the exact positions of the sensors and therefore not about the translation. Now when I try to do an ICP it tries match the complete cloud. Is there a way to only try to match roughly the "expected to match" part of the clouds and then apply the transformation to the whole second cloud?

At the moment my result is like this, which is worse than doing no matching at all (corresponding [config file](https://github.com/ethz-asl/libpointmatcher/files/63867/output1.yaml.txt)):
[image]

When I try to use a BoundingBoxDataPointsFilter to limit the area which I assume could be matching, the result is still quite skewed (the yellow box represents the area for the bounding box filter), even though it is now much better than before but still worse than doing no ICP (corresponding [config file](https://github.com/ethz-asl/libpointmatcher/files/63868/output2.yaml.txt)):
[image]

My question is now, am I missing something here, that allows the kind of matching I am trying with better results?

gawela, 2015-12-16:
Looking at your config files and the problem at hand my best guess would be to change the errorMinimizer from ```PointToPointErrorMinimizer``` to ```PointToPlaneErrorMinimizer```.
Can you try this out?

85pando, 2015-12-16:
I've had some trouble with the ``PointToPlaneErrorMinimizer``. After inserting a ``SurfaceNormalDataPointsFilter`` to get surface normals i get the following error (
[config](https://github.com/ethz-asl/libpointmatcher/files/63946/myIcp.yaml.txt))
:

```
* KDTreeMatcher: initialized with knn=1, epsilon=0, searchType=1 and maxDist=inf
Applying 2 DataPoints filters - 326 points in
* KDTreeMatcher: initialized with knn=10, epsilon=0, searchType=1 and maxDist=inf
WARNING: Matrix C needed for eigen decomposition was degenerated in 290 points over 326 (88.9571 %)
* SurfaceNormalDataPointsFilter - 326 points out (-0%)
* BoundingBoxDataPointsFilter - 65 points out (-80.0613%)
Applied 2 filters - 65 points out (-80.0613%)
PointMatcher::icp - reference pre-processing took 0.0333833 [s]
Applying 2 DataPoints filters - 220 points in
* KDTreeMatcher: initialized with knn=10, epsilon=0, searchType=1 and maxDist=inf
WARNING: Matrix C needed for eigen decomposition was degenerated in 166 points over 220 (75.4545 %)
* SurfaceNormalDataPointsFilter - 220 points out (-0%)
* BoundingBoxDataPointsFilter - 79 points out (-64.0909%)
Applied 2 filters - 79 points out (-64.0909%)
PointMatcher::icp - reading pre-processing took 0.0293146 [s]
terminate called after throwing an instance of 'PointMatcher<float>::ConvergenceError'
  what():  abs rotation norm not a number
[1]    11939 abort (core dumped)  pmicp --config myIcp.yaml frontArt.csv rightArtTrans.csv
```

Adding a ``OrientNormalsDataPointsFilter`` produces:
```
terminate called after throwing an instance of 'PointMatcher<float>::DataPoints::InvalidField'
  what():  OrientNormalsDataPointsFilter: Error, cannot find observation directions in descriptors.
[1]    11987 abort (core dumped)  pmicp --config myIcp.yaml frontArt.csv rightArtTrans.csv
```

pomerlef, 2015-12-16:
Couple of points:
- your core problem is about outlier management more than error function.
- the bounding box is a good idea, but you could simply by limiting how far a match is authorized by using:
```
matcher:
  KDTreeMatcher:
    knn: 1
    epsilon: 0 
    maxDist: 0.25
``` 
- there will be some extra outliers to handle, so adding a trim outlier filter should do it:
```
outlierFilters:
  - TrimmedDistOutlierFilter:
      ratio: 0.85
```
- you are working with simulated data (no noise) and this why `SurfaceNormalDataPointsFilter` is producing the warning about the degenerated points. Either add a small amount of noise to your simulation or use `pointToPointErrorMinimizer` with the outlier filters I described.
- if you want to understand what is going on with your solution, you should use the vtkInspector:
```
 inspector:
 VTKFileInspector:
     baseFileName: debug- 
     dumpDataLinks: 1 
     dumpReading: 1
     dumpReference: 1 
 ```
This will drop a bunch of vtk file per iteration that you can open in Paraview. A bit like in that [Figure 6](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/ICPIntro.md). Otherwise, you will just blindly guess the value of the parameters.

pomerlef, 2015-12-16:
The fact that `nan` can crawl to the code is concerning though.

85pando, 2015-12-17:
The ``maxDist`` value in combination with the ``outlierFilter`` now produces reasonable results with the ``PointToPointErrorMinimizer``:

[image]

Adding noise to the data gets rid of the degeneration warnings as promised, but now there are problems with the ``TrimmedDistOutlierFilter`` 

```
* KDTreeMatcher: initialized with knn=1, epsilon=0, searchType=1 and maxDist=0.25
Applying 1 DataPoints filters - 326 points in
* KDTreeMatcher: initialized with knn=10, epsilon=0, searchType=1 and maxDist=inf
* SurfaceNormalDataPointsFilter - 326 points out (-0%)
Applied 1 filters - 326 points out (-0%)
PointMatcher::icp - reference pre-processing took 0.036638 [s]
Applying 1 DataPoints filters - 220 points in
* KDTreeMatcher: initialized with knn=10, epsilon=0, searchType=1 and maxDist=inf
* SurfaceNormalDataPointsFilter - 220 points out (-0%)
Applied 1 filters - 220 points out (-0%)
PointMatcher::icp - reading pre-processing took 0.02609 [s]
terminate called after throwing an instance of 'PointMatcher<float>::ConvergenceError'
  what():  no outlier to filter
[1]    4240 abort (core dumped)  pmicp --config myIcp.yaml frontArtNoise.csv rightArtTransNoise.csv
```

For me all is well now, as I have a working version with the ``PointToPointErrorMinimizer`` and will  try the tip with dumping the single steps and find good parameters as soon as I can use the actual system. If you want to further hunt down the problem with ``nan``s

 I will of course be there to further try and break the system.

In any case, here are the data files I used:

* Reference (black clouds):
    * [frontCloud.csv](https://github.com/ethz-asl/libpointmatcher/files/65373/frontArt.csv.txt)
    * [frontCloudNoise.csv](https://github.com/ethz-asl/libpointmatcher/files/65376/frontArtNoise.csv.txt)

* Reading (cyan clouds):
    * [rightCloud.csv](https://github.com/ethz-asl/libpointmatcher/files/65375/rightArt.csv.txt)
    * [rightCloudTrans.csv](https://github.com/ethz-asl/libpointmatcher/files/65372/rightArtTrans.csv.txt)
    * [rightCloudTransNoise.csv](https://github.com/ethz-asl/libpointmatcher/files/65374/rightArtTransNoise.csv.txt)

* [config.yaml](https://github.com/ethz-asl/libpointmatcher/files/65384/myIcp.yaml.txt)

85pando, 2016-01-17:
Using Data from the actual robot, the result is now very satisfying. Thanks again for the help.

[image]

closed 2016-01-17 by 85pando

YoshuaNava, 2022-05-19:
@pomerlef Following-up on your reply [here](https://github.com/ethz-asl/libpointmatcher/issues/131#issuecomment-165229927), is there a way to dump VTK logs from libpointmatcher's ICP with different filenames per call of ICP? 

I ask because I'm currently collecting logs, and my vtk files get overwritten on every call to ICP registration.

Thank you in advance!

pomerlef, 2022-05-21:
Out of the box no, but I see the interest ( at the expense of loading a hard drive ;) ).

One would need to add a member to [`VTKFileInspector`](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/InspectorsImpl.h#L178) initialized at 0 and that increment at every call to [`finish(...)`](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/InspectorsImpl.cpp#L743). The variable `baseFileName` could then be also modified in `finish(...)` to concatenate the ICP call id.

It would worth it to open a new issue about this.