norlab-ulaval/libpointmatcher #136: Installation instructions for Windows should reflect specific directory structure...

drhalftone, 2016-01-25:
I tried building the library following the Windows instructions and made it to the very end, except that VS kept trying to link to Daniel.lib.  This was caused by the fact that my home directory is Daniel Lau.  So the libnado directory had a space in it.  This caused VS to see two separate libraries inside of one long pathname to a single library.  So I threw away all my work and started over.  This time, I created the folder C:/Data/Projects and built all my libraries there instead of in my Downloads folder.

Might be worth adding a note to the installation instructions.

pomerlef, 2016-01-25:
I'm a bit in the dark when it comes to VS and the instructions to build on Windows. Most of the instructions for Windows as they are now were modified by external contributors.

If you want me to add specific sentences, I will be glad to do it. Otherwise, I can link the documentation to this issue.

drhalftone, 2016-01-25:
Up to you. Aside from this issue it was pretty straight forward.

pomerlef, 2016-01-25:
I added a link to this issue

Glad to hear that it is working! For archival, on what versions of Windows and VS did you make it compiled?

pomerlef, 2016-01-29:
The [documentation on Windows](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/CompilationWindows.md#reporting-issues) was quickly updated with a link to this issue.
From another communication, it is possible to build the library on Windows with the following:

> All my code was compiled in Windows 10 using Visual Studio 2013 Express with what I believe is the x64 VC12 compiler.  I did all my compiling inside the x64 cross compiler command window.  I also used the latest and greatest Eigen and Boost libraries.

closed 2016-01-29 by pomerlef