norlab-ulaval/libpointmatcher #139: eigen_assert(rows() == other.rows() && cols() == other.cols)

drhalftone, 2016-01-28:
In release mode, everything works great.  I compiled in Debug mode, and then I get this error.  I traced the error back to line 79 of the ErrorMinimizersImpl.cpp file where:

    const Vector& w = mPts.weights();

In running the unit tests, I get a failure created the Identity transform. The log says I should have 16-bytes in my result but I'm getting 32.  So I'm suspicious of my debug compiling from the unit tests.  I realize I'm off the beaten trail, but any help would be greatly appreciated.

drhalftone, 2016-01-28:
My immediate fix is to disable Eigen assertions using the EIGEN_NO_DEBUG directive.  Reading online, it looks like this gets set when compiling in release mode.  I can now run my application, compiled in Qt Creator, in either release or debug modes without any issue. However, is this smart?

pomerlef, 2016-01-29:
I recently encountered a similar problem on Ubuntu. I'll investigate that in a near future.

pomerlef, 2016-02-03:
That seems to solve the problem I had on Ubuntu also. I added the fix and will monitor if it causes other problems.

closed 2016-02-03 by pomerlef