norlab-ulaval/libpointmatcher #141: Extension: Similarity transformation for Point-to-Point Error Minimizer

bkueng, 2016-01-29:
Hi

I extended the Point-to-Point error minimizer by a scaling factor. 

The implementation is in the branch similarity_transformation: https://github.com/bkueng/libpointmatcher/tree/similarity_transformation

It works but requires proper integration into the library (create a separate transformation class based on [Transformation](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/TransformationsImpl.h#L54) and add a new SimilarityErrorMinimizer, ...), but I currently don't have the time to do it.

Derivation is based on Umeyama1991 "Least-Squares Estimation of Transformation Parameters Between Two Point Patterns", which describes the minimization without weights. The following shows my derivation with weights, using the Lemma from Umeyama1991:
[image]

pomerlef, 2016-01-29:
Great, that will give a good head start once we move forward in extending the transformation capabilities of the library.

oleg-alexandrov, 2016-03-12:
I tested that this functionality works. 

I would be willing to make a carefully prepared pull request for merging this into the main library. Is there interest in that? 

I am thinking of making a new class, called say PointToPointSimilarityErrorMinimizer, analogous to PointToPointErrorMinimizer and PointToPlaneErrorMinimizer. 

In principle something like this can work for PointToPlane as well, but I don't have time to go into the math. If bkueng could do the math, I could do the integration.

pomerlef, 2016-03-14:
I don't have a current use case for it, but I see many interesting applications in Robotics. I can support the review process, but hardly the development.

A new class named `PointToPointSimilarityErrorMinimizer` would make sense. You will need a new `Transformation` class also, which might highlight some missing features in the library as we only tested with the `RigidTransformation`.

I've the feeling that all the points collapsing into a single location is also a valid solution. Maybe a toy example with points on spheres with two different radius would help and could be added in the unit test data.

oleg-alexandrov, 2016-03-14:
I tested the current implementation quite a lot, before trying to integrate it. The code works, and gives correct results. Of course, just as plain point-to-point ICP, it does not converge if the two clouds have very different scales/translations/rotations. In fact, the similarity point-to-point ICP is even somewhat more sensitive than regular point-to-point ICP to how close the clouds are. But this is not news, and more likely people will use this code for clouds which are not far off to start with. 

I will of course add a unit test for this and add a new transformation class.

bkueng, 2016-03-15:
Yes the additional DOF makes it somewhat less stable if point correspondences are mismatched. That's why I added the scaling factor.

Sorry, currently I don't have the time to further work on this.

oleg-alexandrov, 2016-03-16:
I integrated the similarity transform. I added a new PointToPointSimilarityErrorMinimizer class, and a new SimilarityTransform class. I added a unit test which checks if the scale is what is expected.

oleg-alexandrov, 2016-03-16:
I put in a pull request. All tests pass.

oleg-alexandrov, 2016-03-17:
I added a blurb in the tutorial about this new fancy ability as well. This was merged. I think it is safe to close this.

pomerlef, 2016-03-17:
Great, as soon as I've a bit of time, I'll be delighted to play with this.

closed 2016-03-17 by pomerlef