norlab-ulaval/libpointmatcher #142: Compilation with Eigen 3.0.x now fails

pomerlef, 2016-02-02:
This is caused by [those lines](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/ErrorMinimizersImpl.cpp#L81):
```
const Vector meanReading =
		(mPts.reading.features.topRows(dimCount-1).array().rowwise() * w.array().transpose()).rowwise().sum() * w_sum_inv;
const Vector meanReference =
		(mPts.reference.features.topRows(dimCount-1).array().rowwise() * w.array().transpose()).rowwise().sum() * w_sum_inv;
```

The `rowwise()` multiplication was not yet defined in Eigen 3.0.x, but it is in more recent versions of Eigen. This is clear on Ubuntu 12.04. 

@bkueng if you have a quick fix that doesn't used a `rowwise` multiplication that would be great. Otherwise, I'll have a deeper look later.

bkueng, 2016-02-03:
oh I didn't see that. Yes, this one should work:

	OutlierWeights& w = mPts.weights;
	const Vector meanReading = mPts.reading.features.topRows(dimCount-1).cwiseProduct(w.replicate(dimCount-1, 1)).rowwise().sum() / w.sum();
	const Vector meanReference = mPts.reference.features.topRows(dimCount-1).cwiseProduct(w.replicate(dimCount-1, 1)).rowwise().sum() / w.sum();

(if you use `const Vector& w = mPts.weights`, then you need to use `w.transpose()` afterwards)

However it uses `replicate`, which I assume does memory allocations. Thus the current solution would be preferable once Eigen is updated on all distros.

pomerlef, 2016-02-03:
Great I'll surround the other lines with `#if EIGEN_MAJOR_VERSION > 0`.

Any reasons why using `.../ w.sum()` instead of `... * w_sum_inv` which is already precomputed?

bkueng, 2016-02-03:
the precomputed value is fine, I just took my initial version of the code.

closed 2016-02-03 by pomerlef