norlab-ulaval/libpointmatcher #144: Installation Error

leolx90, 2016-03-09:
Hi,
I'm following the Mac instructions to install the library but it constantly give me these error: 

-- Could NOT find LATEX (missing:  LATEX_COMPILER) 
CMake Warning (dev) at UseDoxygen.cmake:112 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "doc".
Call Stack (most recent call first):
  CMakeLists.txt:40 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Boost version: 1.58.0
-- Try OpenMP C flag = [ ]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [/openmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [-Qopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [-openmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [-xopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [+Oopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [-qsmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [-mp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [ ]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [/openmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [-Qopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [-openmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [-xopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [+Oopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [-qsmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [-mp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Could NOT find OpenMP (missing:  OpenMP_C_FLAGS OpenMP_CXX_FLAGS) 
OpenCL disabled, not looking for it
ANN not found, disabling it in benchmarks
FLANN not found, disabling it in benchmarks
-- Looking for _POSIX_TIMERS
-- Looking for _POSIX_TIMERS - found
CMake Error at tests/CMakeLists.txt:95 (message):
  Compiling a test executable failed.
Call Stack (most recent call first):
  tests/CMakeLists.txt:105 (try_compile_cloud_types)

The versions of dependencies I'm using are following: 

airbears2-10-142-32-121:build Leo$ sw_vers -productVersion

10.11.3

airbears2-10-142-32-121:build Leo$ gcc --version

Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/c++/4.2.1

Apple LLVM version 7.0.2 (clang-700.1.81)

Target: x86_64-apple-darwin15.3.0

Thread model: posix

airbears2-10-142-32-121:build Leo$ brew --version

0.9.5 (git revision 1ebf; last commit 2015-09-24)

airbears2-10-142-32-121:build Leo$ brew info boost

boost: stable 1.58.0 (bottled), HEAD

Collection of portable C++ source libraries

http://www.boost.org

/usr/local/Cellar/boost/1.58.0 (10718 files, 478M) *

  Poured from bottle

From: https://github.com/Homebrew/homebrew/blob/master/Library/Formula/boost.rb

==> Dependencies

Optional: icu4c ✘

==> Options

--c++11

	Build using C++11 mode

--universal

	Build a universal binary

--with-icu4c

	Build regexp engine with icu support

--with-mpi

	Build with MPI support

--without-single

	Disable building single-threading variant

--without-static

	Disable building static library variant

--HEAD

	Install HEAD version

airbears2-10-142-32-121:build Leo$ git --version

git version 2.5.4 (Apple Git-61)

airbears2-10-142-32-121:build Leo$ cmake --version

cmake version 3.3.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).

airbears2-10-142-32-121:build Leo$ brew info eigen

eigen: stable 3.2.5 (bottled), HEAD

C++ template library for linear algebra

http://eigen.tuxfamily.org/

/usr/local/Cellar/eigen/3.2.5 (367 files, 4.2M) *

  Poured from bottle

From: https://github.com/Homebrew/homebrew/blob/master/Library/Formula/eigen.rb

==> Dependencies

Build: cmake ✔

==> Options

--universal

	Build a universal binary

--HEAD

	Install HEAD version

airbears2-10-142-32-121:build Leo$ brew info doxygen

doxygen: stable 1.8.10 (bottled), HEAD

Generate documentation for several programming languages

http://www.doxygen.org/

/usr/local/Cellar/doxygen/1.8.10 (8 files, 13M) *

  Poured from bottle

From: https://github.com/Homebrew/homebrew/blob/master/Library/Formula/doxygen.rb

==> Dependencies

Build: cmake ✔

Optional: graphviz ✘

==> Options

--with-doxywizard

	Build GUI frontend with qt support.

--with-graphviz

	Build with dot command support from Graphviz.

--with-libclang

	Build with libclang support.

--HEAD

	Install HEAD version

pomerlef, 2016-03-10:
Indeed, that might be missing from the dependency list. If I read properly, you're missing latex. I looked in the cmake file and we didn't add an explicit flag to disable doxygen (this is for building the documentation).

You quickest option is to instead latex through MacTex:
https://tug.org/mactex/mactex-download.html

If you're will to wait up to this weekend, I can add a flag to explicitly disable the documentation.

jonsomi, 2016-05-12:
Hi,
I am a complete newbie trying to install your library. I am also running into the same errors when trying to install your library. Installing MacTex made the LaTex error disappear, but OpenMP its sadly still not found. 
Do you have any ideas on how to resolve this issue?
I hope the information below is enough.

Here is the output of the terminal:

`$ cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo ${SRC_DIR}`

```
CMake Warning (dev) at UseDoxygen.cmake:112 (get_target_property):
  Policy CMP0045 is not set: Error on non-existent target in
  get_target_property.  Run "cmake --help-policy CMP0045" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  get_target_property() called with non-existent target "doc".
Call Stack (most recent call first):
  CMakeLists.txt:40 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Boost version: 1.60.0
-- Try OpenMP C flag = [-fopenmp=libomp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [ ]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [/openmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [-Qopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [-openmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [-xopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [+Oopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [-qsmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP C flag = [-mp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [-fopenmp=libomp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [ ]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [/openmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [-Qopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [-openmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Failed
-- Try OpenMP CXX flag = [-xopenmp]
[... 94 lines cut ...]

```
brew info oxygen
doxygen: stable 1.8.11 (bottled), HEAD
Generate documentation for several programming languages
http://www.doxygen.org/
/usr/local/Cellar/doxygen/1.8.11 (8 files, 13.7M) *
  Poured from bottle on 2016-05-12 at 19:12:53
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/doxygen.rb
==> Dependencies
Build: make ✔
Optional: graphviz ✘
==> Options
--with-doxywizard
	Build GUI frontend with qt support.
--with-graphviz
	Build with dot command support from Graphviz.
--with-libclang
	Build with libeling support.
--HEAD
	Install HEAD version

```

Regards, jonsomi

pomerlef, 2016-07-05:
It seems that you are reporting an error from libnabo compilation, but try this to see if it helps

In your build folder run:
```
grep "USE_OPEN_CL" CMakeCache.txt 
```

If the output shows something like:
```
USE_OPEN_CL:BOOL=TRUE
```

Try this cmake line:
```
$ cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -D USE_OPEN_CL=FALSE ${SRC_DIR}
```

pomerlef, 2016-10-25:
The latex and doxygen dependencies are now optional, see [this comment](https://github.com/ethz-asl/libpointmatcher/issues/171#issuecomment-256127582).

closed 2016-10-25 by pomerlef