norlab-ulaval/libpointmatcher #152: Performing ICP onto multiple pairs of point clouds diles is not possible?

jaejunlee0538, 2016-03-21:
Performing ICP onto multiple pairs of point clouds files is not possible with a binary?

For example,  
```
$ ./batch_icp --folder ../dataset/pcds
```

If would be nice, if we can perform ICP algorithm onto tons of files(e.g. 0000.pcd, 0001.pcd....) with single binary.

pomerlef, 2016-03-21:
You can have a look at [align_sequence.cpp](https://github.com/ethz-asl/libpointmatcher/blob/master/examples/align_sequence.cpp) although it was not used for a while.

You might have to dig into the source to a bit as those features are not well documented. This is a good place to start:
https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/IO.cpp#L200

jaejunlee0538, 2016-03-21:
Great. Thanks @pomerlef .
It works well!

closed 2016-12-15 by pomerlef