norlab-ulaval/libpointmatcher #153: Dynamically change baseFileName of VTKFileInspector

Ellon, 2016-04-04:
Hello,

Is that possible to dynamically change the `baseFileName` attribute of a `VTKFileInspector` (available through inheritance from `PerformanceInspector`), so I can add a number suffix that will allow me to inspect several ICP being run by the same program but with different initial transformations? I was thinking that I would be able to dynamic_cast the inspector pointer to `PerformanceInspector` and then change `baseFileName`, but `PerformanceInspector` seems not to be available through the installed header files (it is defined into InspectorsImpl.h).

HannesSommer, 2016-04-05:
I'm currently using something like the following for that purpose :

```
  Parametrizable::Parameters param;
  boost::assign::insert(param) ("baseFileName", baseFileName);
  boost::assign::insert(param) ("dumpDataLinks", "1");
  boost::assign::insert(param) ("dumpReading", "1");
  boost::assign::insert(param) ("dumpReference", "1");
  inspector.reset(PM::get().REG(Inspector).create("VTKFileInspector", param));
```
where ```inspector``` is a public member of ICPChainBase and therefore also ICP (see https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/PointMatcher.h#L644).

So instead of changing baseFileName this just creates a new VTKFileInspector with the correct baseFileName. This isn't very expensive and shouldn’t hurt therefore unless it is done at high frequency.  Does that help?

Ellon, 2016-06-14:
(Sorry for the late reply.)

Yes. Reseting the pointer to a new inspector did the trick. Thank you!

closed 2016-06-14 by Ellon