norlab-ulaval/libpointmatcher #155: Why libpointmatcher works bad?

sunrainljy, 2016-05-11:
I download the indoor "Apartment" dataset of "Challenging data sets for point cloud registration algorithms". Then, I use libpointmatcher to align the point clouds taken by Hokuyo. We use the default parameters and set good initial guess based on the ground truth. However, for many pairs, the libpointmatcher performs very bad. The reference point clouds after alignment are just moved a litter compared with the reference point clouds before  alignment. Is the parameters not suitable? and the dataset of "NIFTi: multi-floor mapping:" or demo code is publicly available?

pomerlef, 2016-05-11:
Interesting title for the issue. 

The quick answer would be: because you didn't tuned the ICP solution to the application.

Long answer:

The default parameters are very _light_ and were tuned to do the bare minimum with the basic unit test we have. They work also with the example code:
```
$ ./pmicp ../example/data//cloud.00000.vtk ../example/data//cloud.00001.vtk
```

The goal of the library is to provide flexibility in its configuration so it can fit to different applications, not to provide one solution that should fit all problems. 
- If you want to learn what the different modules are doing, you can start with [our tutorials](http://libpointmatcher.readthedocs.io/en/latest/) in the beginner section. 
- If you're looking for classical ICP solutions, you can find some [parameter files here](https://github.com/ethz-asl/libpointmatcher/tree/master/evaluations/official_solutions). Those were evaluated using the _"Challenging data sets for point cloud registration algorithms"_ in [this paper](https://www.researchgate.net/publication/235631701_Comparing_ICP_variants_on_real-world_data_sets).

This library produces scan to scan alignment and is intended for developers who want to build there own solution on top of it. Building a large consistent map is a larger topic. The _"NIFTi: multi-floor mapping"_ video, and most of the continuous mapping videos are from [this code](https://github.com/ethz-asl/ethzasl_icp_mapping). The content of this repository is used for our research needs and is not supported for the greater public in general. The dataset was recorded under [ROS](http://www.ros.org/about-ros/) format and was not aimed at public usage.

I hope it clarify the goals of the library and its intended use.

pomerlef, 2016-07-05:
If there is no follow up on this, I'll close the issue.

closed 2016-07-06 by pomerlef