norlab-ulaval/libpointmatcher #156: Getting errors  while compiling a simple ICP for car_clouds using libpointmatcher.

nmohank, 2016-06-08:
when I compile using the following command
gcc ../libpointmatcher/examples/icp_simple.cpp .../examples/data/car_cloud400.csv  ../car_cloud401.csv
 
Getting a error as follows 
/PointMatcher.h:47:27: fatal error: Eigen/StdVector: No such file or directory #include "Eigen/StdVector".

currently Im using 
Name 		| Version
----------------|-------------------------------
ubuntu: 	| Description: Ubuntu 14.04.4 LTS
architecture: 	| 64-bit
gcc: 		| gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
git: 		| git version 1.9.1
cmake: 		| cmake version 2.8.12.2
boost: 		| Version: 1.54.0.1ubuntu1
eigen3: 	| Version: 3.2.0-8
doxygen: 	| Version: 1.8.6-2

More over when i ran the following command 
utest/utest --path [libpointmatcher root]/examples/data/   

I encounter some failure 
[==========] Running 58 tests from 13 test cases.
[----------] Global test environment set-up.
[----------] 5 tests from icpTest
[ RUN      ] icpTest.icpTest
/root/Libraries/libpointmatcher/utest/utest.cpp:158: Failure
Expected: (rel_err) < (0.03), actual: 0.18699 vs 0.03
This error was caused by the test file:
   examples/data/icp_data/defaultPointToPointMinDistDataPointsFilter.yaml
/root/Libraries/libpointmatcher/utest/utest.cpp:158: Failure
Expected: (rel_err) < (0.03), actual: 0.179957 vs 0.03
This error was caused by the test file:
   examples/data/icp_data/defaultSimilarityPointToPointMinDistDataPointsFilter.yaml
[  FAILED  ] icpTest.icpTest (3716 ms)
[ RUN      ] icpTest.icpSingular
[       OK ] icpTest.icpSingular (0 ms)
[ RUN      ] icpTest.icpIdentity
[       OK ] icpTest.icpIdentity (108 ms)
[ RUN      ] icpTest.similarityTransform
/root/Libraries/libpointmatcher/utest/utest.cpp:242: Failure
Expected: (std::abs(scale - 1.04)) < (0.001), actual: 0.241112 vs 0.001
Expecting the similarity transform scale to be 1.04.
[  FAILED  ] icpTest.similarityTransform (1110 ms)
[ RUN      ] icpTest.icpSequenceTest
writing to pointmatcher-run1-iterationInfo.csv
writing to pointmatcher-run1-iterationInfo.csv
[       OK ] icpTest.icpSequenceTest (277 ms)
[----------] 5 tests from icpTest (5211 ms total)

[----------] 1 test from GenericTest
[ RUN      ] GenericTest.ICP_default
[       OK ] GenericTest.ICP_default (135 ms)
[----------] 1 test from GenericTest (135 ms total)

[----------] 4 tests from IOTest
[ RUN      ] IOTest.loadYaml
[       OK ] IOTest.loadYaml (2 ms)
[ RUN      ] IOTest.loadCSV
[       OK ] IOTest.loadCSV (0 ms)
[ RUN      ] IOTest.loadPLY
[       OK ] IOTest.loadPLY (0 ms)
[ RUN      ] IOTest.loadPCD
[       OK ] IOTest.loadPCD (0 ms)
[----------] 4 tests from IOTest (2 ms total)

[----------] 5 tests from IOLoadSaveTest
[ RUN      ] IOLoadSaveTest.VTK
writing to examples/data/unit_test.vtk
[       OK ] IOLoadSaveTest.VTK (1 ms)
[... 110 lines cut ...]
[----------] 13 tests from PointCloudTest (8 ms total)

[----------] 2 tests from Inspectors
[ RUN      ] Inspectors.PerformanceInspector
[       OK ] Inspectors.PerformanceInspector (0 ms)
[ RUN      ] Inspectors.VTKFileInspector
[       OK ] Inspectors.VTKFileInspector (0 ms)
[----------] 2 tests from Inspectors (0 ms total)

[----------] 1 test from Loggers
[ RUN      ] Loggers.FileLogger
[       OK ] Loggers.FileLogger (0 ms)
[----------] 1 test from Loggers (0 ms total)

[----------] Global test environment tear-down
[==========] 58 tests from 13 test cases ran. (15529 ms total)
[  PASSED  ] 56 tests.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] icpTest.icpTest
[  FAILED  ] icpTest.similarityTransform

 2 FAILED TESTS.

Please help me. 
Thanx in advance.

HannesSommer, 2016-06-08:
Your compile command is pretty wrong. Is this a copy and paste bug? The actual error is due to missing include paths (`-I`). The `*.csv` files don't belong there. They should be arguments of the compiled program. I'd recommend some unix c++ tutorials to understand better how to use the compiler or try to use cmake to compile the examples / your upcoming program. Actually the examples should be already compiled with the libpointmatcher library.

The failures when running utest are certainly unrelated but should not happen. @pomerlef  , do you have an idea what could have happened to these tests?

pomerlef, 2016-07-05:
Yes, it seems that you already compile the library and its API examples. You don't need to use gcc to run an executable.

Both of the unit test errors I seeing are related to the `similarity transform` module. Here's the useful part:

```
[ RUN ] icpTest.icpTest
/root/Libraries/libpointmatcher/utest/utest.cpp:158: Failure
Expected: (rel_err) < (0.03), actual: 0.18699 vs 0.03
This error was caused by the test file:
examples/data/icp_data/defaultPointToPointMinDistDataPointsFilter.yaml
/root/Libraries/libpointmatcher/utest/utest.cpp:158: Failure
Expected: (rel_err) < (0.03), actual: 0.179957 vs 0.03
This error was caused by the test file:
examples/data/icp_data/defaultSimilarityPointToPointMinDistDataPointsFilter.yaml

[ RUN ] icpTest.similarityTransform
/root/Libraries/libpointmatcher/utest/utest.cpp:242: Failure
Expected: (std::abs(scale - 1.04)) < (0.001), actual: 0.241112 vs 0.001
Expecting the similarity transform scale to be 1.04.
[ FAILED ] icpTest.similarityTransform (1110 ms)
```

I'm looking at that right now, but it seems that I've a segfault on MacOS with the latest version... I'll investigate more on the unit tests this week.

pomerlef, 2016-07-06:
Everything is fine on my side. Maybe you need to update your version to the latest modif:
```
$ git pull
```

Feel free to reopen if something doesn't work on your side.

closed 2016-07-06 by pomerlef