norlab-ulaval/libpointmatcher #167: What are TransformFeatures and TransformNormals in compute_overlap example?

Ellon, 2016-07-13:
I can't find an explanation of what is TransformFeatures and TransformNormals parameter passed to `PM::get().TransformationRegistrar.create()` in `compute_overlap.cpp` example, lines [79](https://github.com/ethz-asl/libpointmatcher/blob/master/examples/compute_overlap.cpp#L79) and [82](https://github.com/ethz-asl/libpointmatcher/blob/master/examples/compute_overlap.cpp#L82). 

From what I saw [here](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/TransformationsImpl.cpp#L80) I believe that `RidigTransformation::compute()` already transforms the normals and observation vectors, but then I found the lines I mentioned in `compute_overlap.cpp` and now I'm confused.

Could someone explain this?

Best,

pomerlef, 2016-07-13:
Those lines seem to date from 2012. You're right, it should be `RidigTransformation` used there.

pomerlef, 2016-07-19:
This has been corrected. Thanks for reporting!

closed 2016-07-19 by pomerlef