norlab-ulaval/libpointmatcher #170: Libpointmatcher is only supported in 64 bits.

christophk86, 2016-09-04:
Hello everybody,

is it possible, that libpointmatcher supports 32 bit compiler. The reason is to run it on a ARM architecture.

Best
Chris

pomerlef, 2016-09-04:
Hi @christophk86,

We had a problem in the early days of libpointmatcher with Eigen alignment not working as expected for 32 bit. Because of a lack of time, we simply decided to not investigate this and just catch the architecture in the CMakeFiles.txt to inform the user.

I can see the advantage of supporting compilation for ARM, but I'm not equipped to test that and don't have time to be the main investigator. If you want to invest your time in the port, I can guide you in what I think needs to be  addressed and make sure that your changes get properly integrated in the main branch.

Is it possible that now ARM architecture can support 64 bit ([reference](https://en.wikipedia.org/wiki/ARM_architecture#64.2F32-bit_architecture))?  If it's possible for you, it might be easier to change your hardware.

christophk86, 2016-09-05:
Hi @pomerlef,
I successfully compiled libpointmatcher for 32 bit platform. 
If I start the unit-test with utest --path ../../../examples/data/
following error occurs:
[==========] Running 59 tests from 13 test cases.
[----------] Global test environment set-up.
[----------] 5 tests from icpTest
[ RUN      ] icpTest.icpTest
[       OK ] icpTest.icpTest (12045 ms)
[ RUN      ] icpTest.icpSingular
[       OK ] icpTest.icpSingular (2 ms)
[ RUN      ] icpTest.icpIdentity
D:\Daten\PTX\SVN_dev\software_development\trunk\Windows_SW\Tests\Reconstruction\libpointmatcher\utest\utest.cpp(218): error: Value of: PM::Matrix::Identity(4,4)
  Actual: 16-byte object <56-4C 27-68 04-00 00-00 04-00 00-00 07-00 00-00>
Expected: curT
Which is: 12-byte object <80-A6 3A-02 04-00 00-00 04-00 00-00>
Expecting identity transform.

[  FAILED  ] icpTest.icpIdentity (348 ms)

Is this what you mean by the eigen-problem you had in the past?

Best,
Chris

pomerlef, 2016-09-05:
It might be, but it as been many years now.

Here some readings that could be useful: https://eigen.tuxfamily.org/dox/group__DenseMatrixManipulation__Alignement.html

pomerlef, 2016-12-20:
This issue is linked with #30.