norlab-ulaval/libpointmatcher #171: Does libpointmatcher is available on Ubuntu 14.04 Trusty ?

ArthurVanpoucke, 2016-09-06:
Hi i'm trying to install libpointmatcher on Ubuntu 14.04 to use it on ROS indigo, i'm having trouble with align_sequence.

I'm getting this error when i build the libpointmatcher lib directory : 

`align_sequence.cpp:153 : undefined reference to « PointMatcher<float>::DataPoints::save(std::string const&) const »
`
Do you have any ideas why ? The libs Eigen, libnabo and boost seem to work fine.

Thank you for your time and help

pomerlef, 2016-09-06:
Did you compile from source? `align_sequence` work well on on my 14.04 machine.

Follow [those instructions](https://github.com/ethz-asl/libpointmatcher/blob/master/doc/Compilation.md#having-problems) to report a problem with compilation. It's quicker on our side to understand what is going on with all your versions.

ArthurVanpoucke, 2016-09-06:
Here is my config : 

Name 	| Version
-------------|-------------------------------
ubuntu: 	| Description: Ubuntu 14.04.4 LTS
architecture: 	| 64-bit
gcc: 		| gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
git: 		| git version 1.9.1
cmake: 	| cmake version 2.8.12.2
boost: 	| Version: 1.54.0.1ubuntu1
eigen3: 	| Version: 3.2.0-8
doxygen: 	| Version: 1.8.6-2

I can't get the command `sudo apt-get install libpointmatcher-dev` even with adding the ppa source. https://launchpad.net/~stephane.magnenat/+archive/ubuntu/trusty 
In this link i don't see the libpointmatcher package for Ubuntu Trusty, is that normal ? 

Yes i compile from source, I get it from github and execute these commands : 

```
mkdir build && cd build
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo ..
make
sudo make install
```
```
I get stuck here with make :

[ 36%] Built target doxygen
Scanning dependencies of target doc
[ 36%] Built target doc
Scanning dependencies of target pointmatcher
[ 37%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/Logger.cpp.o
[ 38%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/Exceptions.cpp.o
[ 40%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/DataPoints.cpp.o
[ 41%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/Matches.cpp.o
[ 43%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/ICP.cpp.o
[ 44%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/Registry.cpp.o
[ 45%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/Registrar.cpp.o
[ 47%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/DataPointsFilter.cpp.o
[ 48%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/Matcher.cpp.o
[ 50%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/OutlierFilter.cpp.o
[ 51%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/ErrorMinimizer.cpp.o
[ 52%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/Transformation.cpp.o
[ 54%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/TransformationChecker.cpp.o
[ 55%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/Inspector.cpp.o
[ 56%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/IO.cpp.o
[ 58%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/Bibliography.cpp.o
[ 59%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/Timer.cpp.o
[ 61%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/Histogram.cpp.o
[ 62%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/Parametrizable.cpp.o
[ 63%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/LoggerImpl.cpp.o
[ 65%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/DataPointsFiltersImpl.cpp.o
/home/jmc/Libraries/libpointmatcher/pointmatcher/DataPointsFiltersImpl.cpp: In member function ‘void DataPointsFiltersImpl<T>::GestaltDataPointsFilter::fuseRange(DataPointsFiltersImpl<T>::GestaltDataPointsFilter::BuildData&, DataPointsFiltersImpl<T>::DataPoints&, int, int) const [with T = float; DataPointsFiltersImpl<T>::DataPoints = PointMatcher<float>::DataPoints]’:
/home/jmc/Libraries/libpointmatcher/pointmatcher/DataPointsFiltersImpl.cpp:1881:7: warning: ‘cylindricality’ may be used uninitialized in this function [-Wmaybe-uninitialized]
       (*data.gestaltShapes)(1,data.indicesToKeep[i]) = cylindricality;
       ^
/home/jmc/Libraries/libpointmatcher/pointmatcher/DataPointsFiltersImpl.cpp:1880:7: warning: ‘planarity’ may be used uninitialized in this function [-Wmaybe-uninitialized]
       (*data.gestaltShapes)(0,data.indicesToKeep[i]) = planarity;
       ^
/home/jmc/Libraries/libpointmatcher/pointmatcher/DataPointsFiltersImpl.cpp: In member function ‘void DataPointsFiltersImpl<T>::GestaltDataPointsFilter::fuseRange(DataPointsFiltersImpl<T>::GestaltDataPointsFilter::BuildData&, DataPointsFiltersImpl<T>::DataPoints&, int, int) const [with T = double; DataPointsFiltersImpl<T>::DataPoints = PointMatcher<double>::DataPoints]’:
/home/jmc/Libraries/libpointmatcher/pointmatcher/DataPointsFiltersImpl.cpp:1881:7: warning: ‘cylindricality’ may be used uninitialized in this function [-Wmaybe-uninitialized]
       (*data.gestaltShapes)(1,data.indicesToKeep[i]) = cylindricality;
       ^
/home/jmc/Libraries/libpointmatcher/pointmatcher/DataPointsFiltersImpl.cpp:1880:7: warning: ‘planarity’ may be used uninitialized in this function [-Wmaybe-uninitialized]
       (*data.gestaltShapes)(0,data.indicesToKeep[i]) = planarity;
       ^
[ 66%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/MatchersImpl.cpp.o
[ 68%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/OutlierFiltersImpl.cpp.o
[ 69%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/ErrorMinimizersImpl.cpp.o
[ 70%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/TransformationsImpl.cpp.o
[ 72%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/TransformationCheckersImpl.cpp.o
[ 73%] Building CXX object CMakeFiles/pointmatcher.dir/pointmatcher/InspectorsImpl.cpp.o
Linking CXX shared library libpointmatcher.so
[ 73%] Built target pointmatcher
Scanning dependencies of target gtest
[ 75%] Building CXX object contrib/CMakeFiles/gtest.dir/gtest/gtest-all.cc.o
Linking CXX static library libgtest.a
[ 75%] Built target gtest
Scanning dependencies of target align_sequence
[ 76%] Building CXX object examples/CMakeFiles/align_sequence.dir/align_sequence.cpp.o
Linking CXX executable align_sequence
[... cut ...] align_sequence.dir/align_sequence.cpp.o : Dans la fonction « main » :
/home/jmc/Libraries/libpointmatcher/examples/align_sequence.cpp:153 : référence indéfinie vers « PointMatcher<float>::DataPoints::save(std::string const&) const »
collect2: error: ld returned 1 exit status
make[2]: *** [examples/align_sequence] Erreur 1
make[1]: *** [examples/CMakeFiles/align_sequence.dir/all] Erreur 2
make: *** [all] Erreur 2
jmc@jmcpc:~/Libraries/libpointmatcher/build$ 

```

Thank you for your fast answer !

pomerlef, 2016-09-06:
We stopped supporting `apt-get install` since a while. That's normal.

Could it be that you have an old version of libpointmatcher somewhere and `align_sequence` links to it?

Have a look there: `/usr/local/lib/libpointmatcher.so`

and/or run

```
$ VERBOSE=1 make align_sequence
```
and copy-paste the line under:
```
[100%] Building CXX object examples/CMakeFiles/align_sequence.dir/align_sequence.cpp.o
```

ArthurVanpoucke, 2016-09-07:
Not sure if is is what you ask for but : 

```
jmc@jmcpc:~/Libraries/libpointmatcher/build$ VERBOSE=1 make align_sequence >> ~/Bureau/debug.txt 
Warning: Tag `SYMBOL_CACHE_SIZE' at line 289 of file /home/jmc/Libraries/libpointmatcher/build/Doxyfile has become obsolete.
To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
Warning: Tag `SHOW_DIRECTORIES' at line 473 of file /home/jmc/Libraries/libpointmatcher/build/Doxyfile has become obsolete.
To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
Warning: Tag `HTML_ALIGN_MEMBERS' at line 799 of file /home/jmc/Libraries/libpointmatcher/build/Doxyfile has become obsolete.
To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
Warning: doxygen no longer ships with the FreeSans font.
You may want to clear or change DOT_FONTNAME.
Otherwise you run the risk that the wrong font is being used for dot generated graphs.
/home/jmc/Libraries/libpointmatcher/pointmatcher/DataPoints.cpp:69: warning: no matching class member found for 
  template < T >
  PointMatcher< T >::DataPoints::Labels::Labels(const PointMatcher::DataPoints::Label &label)
Possible candidates:
  PointMatcher< T >::DataPoints::Labels::Labels()
  PointMatcher< T >::DataPoints::Labels::Labels(const Label &label)

/home/jmc/Libraries/libpointmatcher/pointmatcher/DataPoints.cpp:102: warning: no matching class member found for 
  template < T >
  PointMatcher< T >::DataPoints::DataPoints(const PointMatcher::DataPoints::Labels &featureLabels, const PointMatcher::DataPoints::Labels &descriptorLabels, size_t pointCount)
Possible candidates:
  PointMatcher< T >::DataPoints::DataPoints()
  PointMatcher< T >::DataPoints::DataPoints(const Labels &featureLabels, const Labels &descriptorLabels, const size_t pointCount)
  PointMatcher< T >::DataPoints::DataPoints(const Labels &featureLabels, const Labels &descriptorLabels, const Labels &timeLabels, const size_t pointCount)
  PointMatcher< T >::DataPoints::DataPoints(const Matrix &features, const Labels &featureLabels)
  PointMatcher< T >::DataPoints::DataPoints(const Matrix &features, const Labels &featureLabels, const Matrix &descriptors, const Labels &descriptorLabels)
  PointMatcher< T >::DataPoints::DataPoints(const Matrix &features, const Labels &featureLabels, const Matrix &descriptors, const Labels &descriptorLabels, const Int64Matrix &times, const Labels &timeLabels)

/home/jmc/Libraries/libpointmatcher/pointmatcher/DataPoints.cpp:113: warning: no matching class member found for 
  template < T >
  PointMatcher< T >::DataPoints::DataPoints(const PointMatcher::DataPoints::Labels &featureLabels, const PointMatcher::DataPoints::Labels &descriptorLabels, const PointMatcher::DataPoints::Labels &timeLabels, size_t pointCount)
Possible candidates:
  PointMatcher< T >::DataPoints::DataPoints()
  PointMatcher< T >::DataPoints::DataPoints(const Labels &featureLabels, const Labels &descriptorLabels, const size_t pointCount)
  PointMatcher< T >::DataPoints::DataPoints(const Labels &featureLabels, const Labels &descriptorLabels, const Labels &timeLabels, const size_t pointCount)
  PointMatcher< T >::DataPoints::DataPoints(const Matrix &features, const Labels &featureLabels)
  PointMatcher< T >::DataPoints::DataPoints(const Matrix &features, const Labels &featureLabels, const Matrix &descriptors, const Labels &descriptorLabels)
  PointMatcher< T >::DataPoints::DataPoints(const Matrix &features, const Labels &featureLabels, const Matrix &descriptors, const Labels &descriptorLabels, const Int64Matrix &times, const Labels &timeLabels)

/home/jmc/Libraries/libpointmatcher/pointmatcher/DataPoints.cpp:131: warning: no matching class member found for 
  template < T >
  PointMatcher< T >::DataPoints::DataPoints(const Eigen::Matrix< T, Eigen::Dynamic, Eigen::Dynamic > &features, const PointMatcher::DataPoints::Labels &featureLabels)
Possible candidates:
  PointMatcher< T >::DataPoints::DataPoints()
  PointMatcher< T >::DataPoints::DataPoints(const Labels &featureLabels, const Labels &descriptorLabels, const size_t pointCount)
  PointMatcher< T >::DataPoints::DataPoints(const Labels &featureLabels, const Labels &descriptorLabels, const Labels &timeLabels, const size_t pointCount)
  PointMatcher< T >::DataPoints::DataPoints(const Matrix &features, const Labels &featureLabels)
  PointMatcher< T >::DataPoints::DataPoints(const Matrix &features, const Labels &featureLabels, const Matrix &descriptors, const Labels &descriptorLabels)
  PointMatcher< T >::DataPoints::DataPoints(const Matrix &features, const Labels &featureLabels, const Matrix &descriptors, const Labels &descriptorLabels, const Int64Matrix &times, const Labels &timeLabels)

/home/jmc/Libraries/libpointmatcher/pointmatcher/DataPoints.cpp:138: warning: no matching class member found for 
  template < T >
  PointMatcher< T >::DataPoints::DataPoints(const Eigen::Matrix< T, Eigen::Dynamic, Eigen::Dynamic > &features, const PointMatcher::DataPoints::Labels &featureLabels, const Eigen::Matrix< T, Eigen::Dynamic, Eigen::Dynamic > &descriptors, const PointMatcher::DataPoints::Labels &descriptorLabels)
Possible candidates:
  PointMatcher< T >::DataPoints::DataPoints()
  PointMatcher< T >::DataPoints::DataPoints(const Labels &featureLabels, const Labels &descriptorLabels, const size_t pointCount)
[... cut ...]
CMakeFiles/align_sequence.dir/align_sequence.cpp.o : Dans la fonction « main » :
/home/jmc/Libraries/libpointmatcher/examples/align_sequence.cpp:153 : référence indéfinie vers « PointMatcher<float>::DataPoints::save(std::string const&) const »
collect2: error: ld returned 1 exit status
make[3]: *** [examples/align_sequence] Erreur 1
make[2]: *** [examples/CMakeFiles/align_sequence.dir/all] Erreur 2
make[1]: *** [examples/CMakeFiles/align_sequence.dir/rule] Erreur 2
make: *** [align_sequence] Erreur 2

```

pomerlef, 2016-09-07:
ok, the output is not what I expected. I'm not sure that libpointmatcher is building at all given:
```
sh: 1: LATEX_COMPILER-NOTFOUND: not found
error: Problems running latex. Check your installation or look for typos in _formulas.tex and check _formulas.log!
sh: 1: dvips: not found
error: Problems running dvips. Check your installation!
```

Check if you have latex install, if not try:
```
sudo apt-get install doxygen-latex
```

This might be related to issue #144, where we don't detect properly if latex is installed and, if not, disable doxygen compilation.

Kailegh, 2016-10-25:
I did `sudo apt-get install doxygen-latex `and this error ```
sh: 1: LATEX_COMPILER-NOTFOUND: not found
error: Problems running latex. Check your installation or look for typos in _formulas.tex and check _formulas.log!
sh: 1: dvips: not found
```
error: Problems running dvips. Check your installation! still shows up

pomerlef, 2016-10-25:
**Short:**
Alright, I disabled the documentation by default. You should be able to `git pull` and build without the documentation generated by Doxygen.

**Long:**
The option to build with Latex was disabled, but it seems that there was some equations in the documentation forcing the use of Latex.

If you need the API documentation, you will have to install latex and its tools which is a lot of drive space if you're not using it for other things (it's usually used for scientific writing):
```
sudo apt-get install texlive-full
```

Once you have latex on your system, you can set the cmake variable `GENERATE_API_DOC` to `true`. You can use cmake GUI or the following command line:

```
cmake -D GENERATE_API_DOC=true .
```

I'll change the doc to say that Latex and doxygen are optional dependencies.

peci1, 2016-10-27:
Are you sure it is a latex problem? To me it seems sombody started using the DataPoints::save() function, which is defined, but not implemented (or at least I was unable to find its implementation on the master branch). Or is it a pure virutal function?

    CMakeFiles/align_sequence.dir/align_sequence.cpp.o : Dans la fonction « main » :
    /home/jmc/Libraries/libpointmatcher/examples/align_sequence.cpp:153 : référence indéfinie vers « PointMatcher<float>::DataPoints::save(std::string const&) const »
    collect2: error: ld returned 1 exit status

pomerlef, 2016-10-27:
It is defined in [IO.cpp](https://github.com/ethz-asl/libpointmatcher/blob/75044815d40ff934fe0bb7e05ed8bbf18c06493b/pointmatcher/IO.cpp#L888).

My interpretation is that the following error abort the compilation of libpointmatcher.
```
sh: 1: LATEX_COMPILER-NOTFOUND: not found
error: Problems running latex. Check your installation or look for typos in _formulas.tex and check _formulas.log!
sh: 1: dvips: not found
error: Problems running dvips. Check your installation!
```

So the library is not compiled at all and the examples [CMakeList.txt is triggered](https://github.com/ethz-asl/libpointmatcher/blob/master/CMakeLists.txt#L327) without the proper link.

peci1, 2016-10-27:
Weird. I get this linking error with the newest git version of libpointmatcher when I'm compiling laser_mapper. It seems latex plays no role in it.

My bad, the save function is really defined in IO.cpp. It even gets built:

    $ nm -Cg devel/lib/libpointmatcher.so | grep save | head -n2
    00000000001936c0 W PointMatcher<double>::DataPoints::save(std::string const&, bool) const
    0000000000193e70 W PointMatcher<float>::DataPoints::save(std::string const&, bool) const

But still, when I build the laser mapper, I get:

    Linking CXX executable laser_slam_ws/devel/.private/laser_mapper/lib/laser_mapper/laser_mapper_node
    laser_slam_ws/devel/.private/laser_mapper/lib/liblaser_mapper.so: undefined reference to `PointMatcher<float>::DataPoints::save(std::string const&) const'
    collect2: error: ld returned 1 exit status
    make[2]: *** [laser_slam_ws/devel/.private/laser_mapper/lib/laser_mapper/laser_mapper_node] Error 1
    make[1]: *** [CMakeFiles/laser_mapper_node.dir/all] Error 2

pomerlef, 2016-10-27:
@peci1 When you built libpointmatcher, did `align_sequence` compiled properly from the examples?

The original issue was having trouble even with the example compilation. I fear that the error 
```
undefined reference to `PointMatcher<float>::DataPoints::save(std::string const&) const'
collect2: error: ld returned 1 exit status
``` 
is just a symptom of a linking problem with `libpointmatcher.so`.

What is the output of:
```
$ ldd liblaser_mapper.so | grep libpointmatcher
```
and does the path pointing to libpointmatcher still exist on your machine?

HannesSommer, 2016-10-28:
The problem here is almost surely outdated headers somewhere because it is looking for a `DataPoints::save(std::string const&)` that does not exist anymore. Could run `make install` in WORKSPACE/build/libpointmatcher and try again.

pomerlef, 2016-10-28:
Well seen @HannesSommer. I was about to say that this function is as old as the library, but after checking, it changed in Jan 2016. It's still a long time ago though.

HannesSommer, 2016-10-28:
I had a big advantage here: I've changed it back then :). Just realizing that his is quite an old issue, I wonder why I totally missed its beginning. Maybe I could have saved some trouble here. But let's see first whether this is indeed the cause.

peci1, 2016-10-28:
@HannesSommer You're right, after a lot of fiddling with different kinds of environment cleaning, I was finally able to build it ;)

rdube, 2016-10-28:
@peci1 can you propose a fix? @arcgithub FYI you might experience the same issue.

peci1, 2016-10-28:
@rdube Make sure you properly clean your build environment. Delete build and devel in the libpointmatcher workspace. Do not source anything else than `/opt/ros/$ROS_DISTRO/setup.bash`. Thoroughly check the contents of `$LD_LIBRARY_PATH`, `$PKG_CONFIG_PATH`, `$ROS_PACKAGE_PATH` and `$CMAKE_PREFIX_PATH` to not contain any traces of some other older workspace. If you can, uninstall `ros-$ROS_DISTRO-libpointmatcher`.

closed 2016-11-01 by pomerlef