norlab-ulaval/libpointmatcher #172: Problem with pmicp tutorial

The-gimmickk, 2016-09-09:
Followed the following tutorial https://libpointmatcher.readthedocs.io/en/latest/ICPIntro/
It works well till I open cloud_0.vtk and cloud_1.vtk in PARAVIEW and at the next step of launching pmicp with an empty ICP configuration I get an error, I tried to copy paste the file from the tutorial, In that case the error is following:  

terminate called after throwing an instance of 'PointMatcher<float>::ConvergenceError'
  what():  ErrorMnimizer: no point to minimize
Aborted (core dumped)

and when I launched the same YAML in the package you provided, the output is very unexpected:
* KDTreeMatcher: initialized with knn=1, epsilon=0, searchType=1 and maxDist=inf
match ratio: 0.900055
ICP transformation:
   0.993545   -0.113204  0.00724467    0.608055
   0.113237    0.993559 -0.00431907 -0.00381928
-0.00670904   0.0051116    0.999964   0.0106469
          0           0           0           1

Please help how to correct this.
Ubuntu 15.10
Ros kinetic

pomerlef, 2016-09-11:
Thanks for trying out our library. It seems that the tutorial is slightly outdated. We added some safety checks to avoid computation problems when their is zero point that were matched. Also, the `FileLogger` doesn't seems to behave properly.

I'll update the tutorial and investigate what is happening with the `FileLogger` during the week.

Thanks for reporting!

pomerlef, 2016-12-14:
The tutorial has been updated with the commit 836393079cdba12020c79a8d32fcab05bb90e7eb

closed 2016-12-14 by pomerlef