norlab-ulaval/libpointmatcher #174: Can you add something like gitter.im to ask questions ?

Techmind, 2016-09-26:
Having weird results and unshure if bug or i am doing something wrong.

This should be 99% translation, created a gist instead:
https://gist.github.com/Techmind/d52fd6d01756ced2fd68dbc5ab4da354

Techmind, 2016-09-26:
Or maybe a mailing list or something.

Techmind, 2016-09-26:
Ok, probably a bug, added a spreadsheet to calculate difference (length between points) between ref & in, in & out, it became more, than original. 
Should always become less if i understand the purpose of ICP correctly(J47 > J23).

Spreadsheet: 
https://docs.google.com/spreadsheets/d/15IEtTkTp7KwdC0P09Vo7EiAYq7_vd25C9FoTAOUv5VU/edit#gid=1228734682

pomerlef, 2016-09-26:
For now, we answer directly here as everything is at the same place.

What is your configuration file (something.yaml)? If you use the the default configuration (no yaml file), it is most likely that outliers rejection is removing some of your matches. The default configuration was intended to work for the example files, but it is suggested to always setup a yaml file for your own application.

Techmind, 2016-09-26:
readingDataPointsFilters:
  - IdentityDataPointsFilter

referenceDataPointsFilters:
  - IdentityDataPointsFilter

matcher:
  KDTreeMatcher:
    knn: 1
    epsilon: 0.0000001
    searchType: 0

outlierFilters:
  - NullOutlierFilter

errorMinimizer:
  PointToPointErrorMinimizer

transformationCheckers:
  - CounterTransformationChecker:
      maxIterationCount: 4

inspector:
  NullInspector

logger:
  NullLogger

Techmind, 2016-09-26:
But maybe spreadsheet was created from default config(if y provide none), i think after playing with "knn" and epsilon i got better results.

Techmind, 2016-09-26:
Probably default config should be VERY restrictive.

Techmind, 2016-09-26:
With this config:
'''
readingDataPointsFilters:
  - IdentityDataPointsFilter

referenceDataPointsFilters:
  - IdentityDataPointsFilter

matcher:
  KDTreeMatcher:
    knn: 1
    epsilon: 0.0000001
    searchType: 0

outlierFilters:
  - NullOutlierFilter

errorMinimizer:
  PointToPointErrorMinimizer

transformationCheckers:
  - CounterTransformationChecker:
      maxIterationCount: 4

inspector:
  NullInspector

logger:
  NullLogger
'''
i get this: http://s10.postimg.org/m799tc9ax/image.png, which is actually better.

pomerlef, 2016-09-26:
Don't hesitate to use the VTKInspector to debug your solution. See this tutorial:
https://github.com/ethz-asl/libpointmatcher/blob/master/doc/ICPIntro.md#a-real-icp-configuration

```
inspector:
  VTKFileInspector:
     dumpReading : 1
     dumpReference : 1
     dumpDataLinks : 1
```

The default configuration will always be wrong for an application or an other. That's why we tried to simplify the tuning of a solution.

closed 2016-09-26 by pomerlef