norlab-ulaval/libpointmatcher #176: Using PCL and LibPointMatcher together

Kailegh, 2016-10-25:
Hi, this may looks like a silly question, I am starting to use the LibPointMatcher functions, but I would like to work both with them and the PCL functions. I mean, I have 2 pointclouds,and want to use both functions in those. Is there any way I can convert from one format (PointCloud) to the other (PointMatcher) directly? Or is it impossible to work with both of them at the same time? Because everything I have seen until now consists in loading the clouds from a file.
Thanks a lot for your help!

pomerlef, 2016-10-25:
Not a silly question at all. There is no bidding directly implemented in libpointmacher. You can have a look here for ROS conversion:  [libpointmatcher_ros](https://github.com/ethz-asl/ethzasl_icp_mapping/tree/master/libpointmatcher_ros)

If you want to send a patch, we would welcome it. We handle external communications by two ways:

1. Directly in libpointmatcher: no external dependency are added, basic implementation providing a limited set of options is provided to the user (this is the case with the save/load functions for cvs, ply, vtk, etc.)
2. An external repository that would be named `libpointmatcher_pcl` that include both headers and provided a larger set of functionalities.

Kailegh, 2016-10-27:
Yeah I had given a look at this code: https://github.com/ethz-asl/ethzasl_icp_mapping/blob/master/libpointmatcher_ros/src/point_cloud.cpp it seems to make a conversion, so I would probably have to use that to make a conversion. But my question is, how do everyone who use that to build a map while moving ( I have seen that your library is very used to do that kind of stuff) capture the pointcloud from their sensor? There is no sense in saving and loading each pointcloud, they must captured them in libpointmatcher format directly or something.
I am starting now with this library, if it works for me, I can try to make that conversion function and give it to you. I am not an expert, but I can try to do it.
Thanks a lot for your fast answers and thanks for your great work!

pomerlef, 2016-10-27:
Libpointmatcher is only the library containing registration algorithms. Mapping of live point cloud are often done with ROS. 

An example application is the [ethzasl_icp_mapper](https://github.com/ethz-asl/ethzasl_icp_mapping/tree/master/ethzasl_icp_mapper) node, which is available for all to use or get inspired, but not supported as a stable release (by lack of time).

Kailegh, 2016-10-27:
Yeah yeah, I know that, but your registration algorithms only work with your specific point cloud data type, right?
I don't intent only to do a map, I also want to develop other applications, like localisation in that map using point clouds and I have seen that your functions may be very useful to do that

pomerlef, 2016-12-15:
I'm not sure what is your question.

Yes, our registration library only work with our specific point cloud data type (for obvious reasons). You need to convert other formats to `DataPoints`. We tried to keep that format as simple as possible. If you dig in it a bit, you will see that it's essentially some Eigen matrices with extra information. 

I personally use directly `DataPoints` for segmentation, localization, etc.

closed 2018-10-10 by pomerlef

cardboardcode, 2020-11-26:
Hi all, this is to build off what has been discussed here and construct a quick solution to **use PCL and LibPointMatcher**.

## Install [ libnabo ]

Follow instructions to install libnabo.
```
cd $HOME
git clone https://github.com/ethz-asl/libnabo.git
cd $HOME/libnabo
mkdir build && cd build
cmake ..
sudo make install
```

## Install [ libpointmatcher ]

Follow instructions to install **libpointmatcher**.
```
cd $HOME
git clone https://github.com/ethz-asl/libpointmatcher.git
cd $HOME/libpointmatcher
mkdir build && cd build
cmake ..
sudo make install
```

## Set Up [ libpointmatcher_ros ]
Follow the instructions below to download and set up the Pseudo-PCL Wrapper library [ libpointmatcher_ros ].

Ensure that you have installed ROS1 on your workstation. Tested on ROS1 Melodic.
```
cd $HOME
mkdir -p libpointmatcher_ros1_ws/src && cd libpointmatcher_ros1_ws/src
git clone https://github.com/ethz-asl/ethzasl_icp_mapping
cd $HOME/libpointmatcher_ros1_ws
catkin build libpointmatcher_ros
```

## Modify Your Application ROS1 package to use libpointmatcher_ros

1. Modify `CMakeLists.txt` to include libpointmatcher_ros ROS1 package dependency.
```
find_package(catkin REQUIRED COMPONENTS
...
libpointmatcher_ros
libpointmatcher
...
)

...
target_link_libraries(${ros_name_executable_name} ${catkin_LIBRARIES})
...
```

2. Include the following header file in the source files in your Application ROS1 package.
```
#include "pointmatcher/PointMatcher.h"

#include "pointmatcher_ros/point_cloud.h"
#include "pointmatcher_ros/transform.h"
#include "pointmatcher_ros/get_params_from_server.h"
#include "pointmatcher_ros/ros_logger.h"

```

3. Include the following function calls to use the ICP in ROS.

```
sensor_msgs::PointCloud2 scene_Cloud_libpointmatcher;
PointMatcher<float>::DataPoints scene = PointMatcher_ros::rosMsgToPointMatcherCloud<float>(scene_Cloud_libpointmatcher, false);
sensor_msgs::PointCloud2 obj_Cloud_libpointmatcher;
PointMatcher<float>::DataPoints object = PointMatcher_ros::rosMsgToPointMatcherCloud<float>(obj_Cloud_libpointmatcher, false);

PointMatcher<float>::ICP icp;
icp.setDefault();
PointMatcher<float>::TransformationParameters T = icp(object, scene);

std::cout << "Transformation Matrix = \n" << T << std::endl;
PointMatcher<float>::DataPoints transformed_object(object);
icp.transformations.apply(transformed_object, T);

sensor_msgs::PointCloud2 transformed_pcd = PointMatcher_ros::pointMatcherCloudToRosMsg<float>(transformed_object, "/camera_frame_id", ros::Time::now());

```

For use as a PointCloud Data, you can use ROS1 Library to convert **sensor_msgs::PointCloud2** to **pcl::PointCloud**.

pomerlef, 2020-11-26:
That's great @cardboardcode! I'll add a link to your post in our tutorial section.

YoshuaNava, 2020-11-26:
This is super cool. I'd also add that when you convert between PCL and Libpointmatcher that can result in data loss, as pcl point types are usually limited to some fields known a-priori.

It can be useful if you want to use a certain algorithm from PCL, but the representation is not as flexible as libpointmatcher's.