norlab-ulaval/libpointmatcher #177: Some recent change in libpointmatcher breaks compilation of ethzasl_icp_mapping

skohlbr, 2016-10-26:
I have libpointmatcher  and ethzasl_icp_mapping in a workspace (14.04, indigo, build using catkin_tools). This worked fine, but recently I started getting:
<pre>
__________________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << libpointmatcher_ros:make /home/kohlbrecher/argo/logs_isolated/libpointmatcher_ros/build.make.001.log                                                                                                
In file included from /home/kohlbrecher/argo/src/external/ethzasl_icp_mapping/libpointmatcher_ros/src/point_cloud.cpp:1:0:
/home/kohlbrecher/argo/src/external/ethzasl_icp_mapping/libpointmatcher_ros/include/pointmatcher_ros/point_cloud.h:4:39: fatal error: pointmatcher/PointMatcher.h: No such file or directory
 #include "pointmatcher/PointMatcher.h"
                                       ^
compilation terminated.
make[2]: *** [CMakeFiles/pointmatcher_ros.dir/src/point_cloud.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/kohlbrecher/argo/src/external/ethzasl_icp_mapping/libpointmatcher_ros/src/transform.cpp:1:0:
/home/kohlbrecher/argo/src/external/ethzasl_icp_mapping/libpointmatcher_ros/include/pointmatcher_ros/transform.h:4:39: fatal error: pointmatcher/PointMatcher.h: No such file or directory
 #include "pointmatcher/PointMatcher.h"
                                       ^
compilation terminated.
make[2]: *** [CMakeFiles/pointmatcher_ros.dir/src/transform.cpp.o] Error 1
make[1]: *** [CMakeFiles/pointmatcher_ros.dir/all] Error 2
make: *** [all] Error 2
cd /home/kohlbrecher/argo/build_isolated/libpointmatcher_ros; catkin build --get-env libpointmatcher_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..................................................................................................................................................................................................................
Failed     << libpointmatcher_ros:make                                          [ Exited with code 2 ]                                                                                                            
Failed    <<< libpointmatcher_ros                                               [ 1.8 seconds ]                
</pre>

Going back to d199aaaf97964409aeba910308b02845be48e9aa, things build fine again, so it looks like some commit past that accidentally broke things. I don't have the resources to investigate this further, but wanted to report as others might experience this also.

pomerlef, 2016-10-26:
Some compiliation flags were changed in ethzasl_icp_mapping. I suspect more those than the changes in libpointmatcher (as the commit you pointed at dates from 2015).

I'll have a look during the week.

skohlbr, 2016-10-27:
Yeah, wasn't sure where to file the issue, but the fact I could get things back to working by changing the libpointmatcher revision to a previous one made me think I should file it here. What I forgot to mention: I'm using the `reintegrate/master_into_indigo_devel` of ethzasl_icp_mapping in my now working setup.

pomerlef, 2016-12-15:
I lost track of what happen regarding this issue. Does it work for your.

The suggested way to install the library is from source with `sudo make install` outside a catkin environment.

closed 2016-12-20 by pomerlef