norlab-ulaval/libpointmatcher #178: Singular matrix while minimizing point to plane distance

rdube, 2016-10-29:
Hi @pomerlef ! We are getting the following warning when using lpm with NiFTI's rotating laser scanner:

PointMatcher::icp - encountered almost singular matrix while minimizing point to plane distance. QR solution was too inaccurate. Trying more accurate approach using double precision SVD. 

Any tip / hint on how to avoid this situation? Thanks!

pomerlef, 2016-10-30:
@HannesSommer is your guy for this warning.

I saw that problem when the reading and reference are exactly the same points or when the points only lie on a "perfect" plane or line.

Ellon, 2016-10-30:
@rdube This warning (that was a throw before) was discussed in #158 

I still have this warning sometimes when using clouds from the KITTI dataset in my experiments. The reasons for the singularity are still unknown for me. But since I don't see any degradation in the result of the ICP I decided to just live with it.

HannesSommer, 2016-10-30:
I agree, this warning isn't more than a hint that  **in case** something goes wrong tells you where you might want to start searching.

Technically it means that in some iteration (and therefore point-match configuration) of your point to plane ICP there was a symmetry in the cost : a direction in which the two point clouds could be moved w.r.t. each other without changing the cost significantly AND the current algorithm had `minor difficulties` finding an appropriate step in that case. Still the step can very well be good enough.
And even if it wasn't that good this symmetry depends on the matching in that iteration and after the next update this might be very different and the symmetry might vanishes. So possibly no harm done in such cases. 
If the symmetry remains then it is a systematic problem of the point clouds themselves: for example you have two **planar** regular grids and the smaller one is currently somewhere in the middle: it is genuinely unclear how to step along the common plane and possibly how to turn around its normal.

pomerlef, 2016-11-01:
Can we close this issue?

rdube, 2016-11-02:
I see! Thanks for the explanation! We reduced the number of points in the reference map and that somehow resolved the issue. We can discuss further when we are back in town. Thanks and cheers from Dortmund!

closed 2016-11-02 by rdube