norlab-ulaval/libpointmatcher #180: Distance from cloud to reference

Kailegh, 2016-11-09:
Is there any way of knowing how well the ICP has been? I have seen the `pointUsedRatio` and the `getWeightedPointUsedRatio`() in the `errorMinimezer` class. However that only gives me the ratio of used points for the registration, do you include any algorithm for computing the distance between the two clouds?

Thanks a lot!

pomerlef, 2016-11-09:
You can have a look at the issue #125 for general information.

There is now a method called `getResidualError()` introduce with the pull request #154.

Kailegh, 2016-11-10:
thanks for the help, I saw those issues and I ended up in this tutorial:
`https://github.com/ethz-asl/libpointmatcher/blob/master/examples/icp_advance_api.cpp`
that is what I think I was looking for, thanks a lot

pomerlef, 2016-11-10:
no problem!

closed 2016-11-10 by pomerlef