norlab-ulaval/libpointmatcher #181: problem in utests

juliasanchez, 2016-12-08:
Hello,
I have the same issue with the unit tests as in https://github.com/ethz-asl/libpointmatcher/issues/156

Name 		| Version
----------------|-------------------------------
ubuntu: 	| Description: Ubuntu 16.10
architecture: 	| 64-bit
gcc: 		| gcc (Ubuntu 6.2.0-5ubuntu12) 6.2.0 20161005
git: 		| git version 2.9.3
cmake: 		| cmake version 3.5.2 CMake suite maintained and supported by Kitware (kitware.com/cmake).
boost: 		| Version: 1.61.0.2
eigen3: 	| Version: 3.3~beta2-2ubuntu2
doxygen: 	| Version: 1.8.11-3

hp@HP-Z800:~/libpointmatcher/libpointmatcher/build$ utest/utest --path ../examples/data/

[==========] Running 59 tests from 13 test cases.
[----------] Global test environment set-up.
[----------] 5 tests from icpTest
[ RUN      ] icpTest.icpTest
/home/j/libpointmatcher/libpointmatcher/utest/utest.cpp:158: Failure
Expected: (rel_err) < (0.03), actual: 0.279621 vs 0.03
This error was caused by the test file:
   ../examples/data/icp_data/defaultSimilarityPointToPointMinDistDataPointsFilter.yaml
/home/j/libpointmatcher/libpointmatcher/utest/utest.cpp:158: Failure
Expected: (rel_err) < (0.03), actual: 0.253851 vs 0.03
This error was caused by the test file:
   ../examples/data/icp_data/defaultRobustWelschOutlierFilter.yaml
/home/j/libpointmatcher/libpointmatcher/utest/utest.cpp:158: Failure
Expected: (rel_err) < (0.03), actual: 0.258459 vs 0.03
This error was caused by the test file:
   ../examples/data/icp_data/defaultPointToPointMinDistDataPointsFilter.yaml
/home/j/libpointmatcher/libpointmatcher/utest/utest.cpp:158: Failure
Expected: (rel_err) < (0.03), actual: 0.254899 vs 0.03
This error was caused by the test file:
   ../examples/data/icp_data/RobustWelschOutlierFilterApproximation.yaml
[  FAILED  ] icpTest.icpTest (5569 ms)
[ RUN      ] icpTest.icpSingular
[       OK ] icpTest.icpSingular (1 ms)
[ RUN      ] icpTest.icpIdentity
[       OK ] icpTest.icpIdentity (169 ms)
[ RUN      ] icpTest.similarityTransform
/home/j/libpointmatcher/libpointmatcher/utest/utest.cpp:240: Failure
Expected: (std::abs(scale - 1.04)) < (0.001), actual: 0.04 vs 0.001
Expecting the similarity transform scale to be 1.04.
[  FAILED  ] icpTest.similarityTransform (569 ms)
[ RUN      ] icpTest.icpSequenceTest
[       OK ] icpTest.icpSequenceTest (350 ms)
[----------] 5 tests from icpTest (6659 ms total)

[----------] 1 test from GenericTest
[ RUN      ] GenericTest.ICP_default
[       OK ] GenericTest.ICP_default (166 ms)
[----------] 1 test from GenericTest (166 ms total)

[----------] 4 tests from IOTest
[ RUN      ] IOTest.loadYaml
[       OK ] IOTest.loadYaml (2 ms)
[ RUN      ] IOTest.loadCSV
[       OK ] IOTest.loadCSV (0 ms)
[ RUN      ] IOTest.loadPLY
[       OK ] IOTest.loadPLY (1 ms)
[ RUN      ] IOTest.loadPCD
[       OK ] IOTest.loadPCD (0 ms)
[----------] 4 tests from IOTest (3 ms total)

[----------] 5 tests from IOLoadSaveTest
[ RUN      ] IOLoadSaveTest.VTK
[       OK ] IOLoadSaveTest.VTK (1 ms)
[ RUN      ] IOLoadSaveTest.VTKBinary
[... 125 lines cut ...]
[----------] 13 tests from PointCloudTest (10 ms total)

[----------] 2 tests from Inspectors
[ RUN      ] Inspectors.PerformanceInspector
[       OK ] Inspectors.PerformanceInspector (0 ms)
[ RUN      ] Inspectors.VTKFileInspector
[       OK ] Inspectors.VTKFileInspector (0 ms)
[----------] 2 tests from Inspectors (0 ms total)

[----------] 1 test from Loggers
[ RUN      ] Loggers.FileLogger
[       OK ] Loggers.FileLogger (0 ms)
[----------] 1 test from Loggers (0 ms total)

[----------] Global test environment tear-down
[==========] 59 tests from 13 test cases ran. (18546 ms total)
[  PASSED  ] 56 tests.
[  FAILED  ] 3 tests, listed below:
[  FAILED  ] icpTest.icpTest
[  FAILED  ] icpTest.similarityTransform
[  FAILED  ] ErrorMinimizerTest.PointToPointErrorMinimizer

 3 FAILED TESTS

Some idea about it? Thanks in advance.

pomerlef, 2016-12-08:
Thanks for reporting. The pattern of what it failing is less clear as in #156.

Here are the tests failing:

- icpTest.icpTest
  - defaultSimilarityPointToPointMinDistDataPointsFilter.yaml
  - defaultRobustWelschOutlierFilter.yaml
  - defaultPointToPointMinDistDataPointsFilter.yaml
  - RobustWelschOutlierFilterApproximation.yaml
-  icpTest.similarityTransform
- ErrorMinimizerTest.PointToPointErrorMinimizer

And of course all tests are passing on my machine and I guess on the test server. What is the output of this command?
```
echo `git log -1 --pretty=format:"commit: %h | date: %cD"` "| branch:" `git rev-parse --abbrev-ref HEAD`
```
You should have something like this:
```
commit: aa47dae | date: Tue, 1 Nov 2016 09:34:51 -0400 | branch: master
```

juliasanchez, 2016-12-09:
That's exactly what I have...

tejaswid, 2016-12-09:
I have exactly the same commit and the same tests are **failing** on my machine too.

My configuration is
ubuntu:	Ubuntu 16.04, 64bit
gcc:	gcc (Ubuntu 6.2.0-5ubuntu12) 6.2.0 20161005
git:	git version 2.7.4
cmake:	cmake version 3.5.1
boost:	Version: 1.58
eigen3:	Version: 3.2.92

On the other hand on another machine with the following config and **everything works**.
ubuntu: Ubuntu 14.04, 64bit
gcc:	gcc (Ubuntu 4.9.4-2ubuntu1) 4.9.4
git:	git version 2.11
cmake:	cmake version 2.8.12.2
boost:	Version: 1.54.0
eigen3:	Version: 3.2.8

juliasanchez, 2016-12-09:
I tried on another PC with these specifications : 
Name | Version
----------------|-------------------------------
ubuntu: | Description: Ubuntu 16.04.1 LTS
architecture: | 64-bit
gcc: | gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
git: | git version 2.7.4
cmake: | cmake version 3.5.1 CMake suite maintained and supported by Kitware (kitware.com/cmake).
boost: | Version: 1.58.0.1ubuntu1
doxygen: | Version: 1.8.11-1

the new version of eigen 3.3.1 was installed in an extern way in local.

I got the same errors in utests. 

_________________________________________________________________________________

However, I tried on Mac and the utests succeeded with yosemite and failed with sierra.

_________________________________________________________________________________

I tried to change my gcc version from 6 to 5 on one hand and the eigen version on the other hand but it didn't work. 

Could I try something else?

pomerlef, 2016-12-09:
It is working on my side with those specs:

Name 		| Version
----------------|-------------------------------
ubuntu: 	| Description: Ubuntu 14.04.5 LTS
architecture: 	| 64-bit
gcc: 		| gcc (Ubuntu 4.9.4-2ubuntu1~14.04.1) 4.9.4
git: 		| git version 1.9.1
cmake: 		| cmake version 2.8.12.2
boost: 		| Version: 1.54.0.1ubuntu1
eigen3: 	| Version: 3.2.0-8
doxygen: 	| Version: 1.8.6-2

Which seems to be coherent with @tejaswid. 

The usual suspect is often boost. Is there an easy way for you to try with boost 1.54?

HannesSommer, 2016-12-10:
@pomerlef , I've just created a Jenkins multi config job to provide more samples: http://129.132.38.183:8080/job/pointmatcher/. Same situation there: OSX and Trusty build fine. Xenial fails with at least very similar failures. 
The failing Xenial has 

Name 		| Version
----------------|-------------------------------
ubuntu: 	| Description: Ubuntu 16.04.1 LTS
architecture: 	| 64-bit
gcc: 		| gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
boost: 		| Version: 1.58.0.1ubuntu1
eigen3: 	| Version: 3.3~beta1-2

Trusty is:

Name 		| Version
----------------|-------------------------------
ubuntu: 	| Description: Ubuntu 14.04.5 LTS
architecture: 	| 64-bit
gcc: 		| gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
boost: 		| Version: 1.54.0.1ubuntu1
eigen3: 	| Version: 3.2.0-8

I would actually guess that eigen is making the difference here. I'll try with a different eigen on the same system.

HannesSommer, 2016-12-10:
Yep, I can confirm: on the very same Xenial setup but with eigen version 3.2.10 the unit tests are green. It seemed kind of fishy from the very beginning that Xenial has a beta version of eigen installed. Maybe this is an actual bug. Maybe it is just some specification margin that these tests do not tolerate.

tejaswid, 2016-12-10:
Yes, I can confirm what @HannesSommer said. On my Xenial setup with Eigen version 3.2.10 all the unit tests are green.

pomerlef, 2016-12-10:
Thanks everyone for searching the bug. It's getting there. Also, good catch for the beta version of Eigen.

It seems that we are way beyond a tolerance problem:
` actual: 0.279621 vs 0.03`

juliasanchez, 2016-12-11:
I want to clarify that I had already changed the beta version to the last version of eigen (3.3.1) and it didn't work. I will try with the 3.2.10 now.

HannesSommer, 2016-12-12:
Okay, this was even nastier to fix than to find :). The underlying problem was an assignment matrix (weights) to vector, which apparently changed it's (maybe unspecified) meaning over the progress of Eigen.

Unfortunately I didn't find a better replacement without copying for this  `const Vector w = mPts.weights.row(0);`  (withing my limited time budget). Therefore this fix should make the PointToPoint minimizer slightly slower (probably not much).

Interestingly this conversion to a vector type is not even required for this deactivated code : https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/ErrorMinimizersImpl.cpp#L217-L218. Maybe this is a good occasion to go one step forward?

For who ever want to take over from here: pleas be aware that the buggy code has been bluntly copied in th past! So what ever you do, do ti twice, or even better: deduplicate first if possible :).

pomerlef, 2016-12-12:
That might highlight an actual bug. 

The member `weights` in the struct `ErrorElements` should be of type `Vector` as all the elements are flatten. This would avoid implementation problem on the long run.

The patch of @HannesSommer produces the same behavior as before, so I would suggest the following to avoid wasting too much time:
- accept the patch #183 as soon as possible so people can move on with their projects
- open an issue with the type of `weights` being changed to `Vector` (enhancement)
- open an issue about the code redundancy in `ErrorMinimizersImp.cpp` (enhancement)

pomerlef, 2016-12-13:
If it compile for everybody, this issue can be close.

closed 2016-12-15 by pomerlef