norlab-ulaval/libpointmatcher #182: Problem with PLY read

arunabh-coderep, 2016-12-10:
Hi, there is a problem with the loadPLY function as it doesn't contain the option to read a PLY with intensity even though the documentation says so, I checked the code and the property intensity is not written into the loadPLY function hence it gives a segmentation fault when I try to read it

Thank you!

arunabh-coderep, 2016-12-10:
[iso_sweep1.ply.zip](https://github.com/ethz-asl/libpointmatcher/files/643478/iso_sweep1.ply.zip)

pomerlef, 2016-12-12:
I did a simple fix to load PLY files with intensity.

The support for PLY is a bit limited for now and its implementation need to be cleaned. I opened an issue more generic asking for a cleanup, but for now, that's the time I've.

pomerlef, 2016-12-13:
It should work in the master branch. Just do `git pull`.

arunabh-coderep, 2016-12-13:
Thank you sir. That worked.

closed 2016-12-13 by pomerlef