norlab-ulaval/libpointmatcher #188: Transform between 2 points clouds

BrunoDuisit, 2016-12-15:
Hi,

I have a point cloud of a statue made by photogrammetry so 360 ° and a point cloud by structured light so only one face. The two points clouds don't have the same scale.

I tried the method used by the unit test "similarityTransform" but it find a scale near 0 to flatten the geomtry!

Do you have a solution to do that considering the different scale?

I can send the point coulds if needed!

Thank you

pomerlef, 2016-12-15:
I didn't play with the similarityTransform yet, but your problem seems general to all point cloud registrations. If your prior transformation is too far, you will converge to a local minima. In the case of a transformation with scale, an obvious one is the zero scale.

You can use a software like Paraview to pre-align your point cloud manually. I would suggest you to use the inspector `VTKFileInspector` at first to understand what happen during the minimization process.

You should use your own calibration file to customize your solution with that part in it:
```
inspector:
 VTKFileInspector:
     dumpDataLinks: 1
     dumpReading: 1
     dumpReference: 1
```

There are tons of reasons why a point cloud registration could fail. The best advice I can give you is to learn how to debug your solution with the visual tools we implemented in the library.

Cheers!

pomerlef, 2016-12-20:
Can we close this issue?

BrunoDuisit, 2016-12-22:
Thank you for your reply! Yes we can close it!

BrunoDuisit, 2016-12-28:
I have one more question related to that issue. 

Can we avoid the algorithm to use one point more than once?! Because it find me a solution with a reprojection error near to 0.00000001, but with a scale near to 0.0000001. Every points in the second point cloud are scale to match on the same point in the first point cloud! 

Can we avoid it?

Thank you

BrunoDuisit, 2017-01-05:
Hello, I still have my problem! 

I tried serveral starts with different angle (Every 22.5° in each direction so 729 starts) and i want to keep the best solution. But the best (with the lowest getMeanError), is wrong. The scale is so small, every point are match on only one point in the second point cloud. 
Can we avoid this? 
Or at least to bounded the scale between 2 values. 
For exemple I have 2 points cloud nearly to the same scale but the higher scale it find me is 0.3
Is there some parameters I can configure to fix this?

Thank you!

pomerlef, 2017-01-05:
You can send your point clouds as attachments and I will check what is going on.

From what I understand, there is no parameters for the [PointToPointSimilarityErrorMinimizer](https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/ErrorMinimizersImpl.h#L85) implementation.

You can also play with the `knn` parameter for the matching step. 
```
matcher:
  KDTreeMatcher:
    knn: 10
    epsilon: 0 
```

The parameter `knn` controls the number of points to be matched. This could reduce the probability of all point from the reading to match with a single point from the reference.

BrunoDuisit, 2017-01-06:
This is my point clouds : 
[data.zip](https://github.com/ethz-asl/libpointmatcher/files/689345/data.zip)

I tried : 
matcher:
  KDTreeMatcher:
    knn: 10
    epsilon: 0 
but it crash!

Should I translate the two point clouds to be closed? or it is already done in icp?

Thank you

pomerlef, 2017-01-06:
Ok, that's what I suspected. Your statue is 0.8 m high and your second scan is roughly 3 m away and flipped by 180 degrees. ICP in general is a local solution, not a global one. You need either a human, another sensor or another algorithm to roughly position your scans not too far away.

Prior too far away:
[image]

Better prior achieved by using Paraview to manually align the scans:
[image]

It's even more critical for a registration with scale as the other degree of freedom produce even more local minimas during the minimization. In the following animation you see that the head first move toward the shoulder before scaling properly. This is an indication that the registration almost converge to a local minima. The animation was done using `VTKFileInspector` in the yaml file.
[image]

The complet yaml file for this registration is:
```
readingDataPointsFilters:
  - RandomSamplingDataPointsFilter:
      prob: 0.5

referenceDataPointsFilters:
  - SamplingSurfaceNormalDataPointsFilter:
      knn: 10

matcher:
  KDTreeMatcher:
    knn: 3
    epsilon: 0

outlierFilters:
  - TrimmedDistOutlierFilter:
      ratio: 0.95

errorMinimizer:
  PointToPointSimilarityErrorMinimizer

transformationCheckers:
  - CounterTransformationChecker:
      maxIterationCount: 80
  - DifferentialTransformationChecker:
      minDiffRotErr: 0.08
      minDiffTransErr: 0.01
      smoothLength: 4

inspector:
 VTKFileInspector:
     baseFileName: pointmatcher-run1
     dumpPerfOnExit: 0
     dumpStats: 0
     dumpIterationInfo: 1
     dumpDataLinks: 1
     dumpReading: 1
     dumpReference: 1

logger:
  FileLogger
```

This issue is more in a range of technical support than a problem with libpointmatcher. I did it completely only because I did tried myself the new `PointToPointSimilarityErrorMinimizer` module. In general, we don't have time to support application specific problems.

Enjoy!

BrunoDuisit, 2017-01-09:
Thank you very much for your reply! 

I applied icp, but first I translated one point cloud on the other and rotated in every direction(bruteforce) and I kept the best solution.

I will try with your yaml!

closed 2017-01-12 by pomerlef

alexsmartens, 2017-03-10:
Thank you!
My result is way better, but it is still a bit off. I attached the screenshot:

[image]

Could you suggest the parameters I can play with for improving the results, please

pomerlef, 2017-03-10:
Hard for me to tell without doing it completely on my side. We cannot support individual tuning for all the applications using `libpointmatcher`. Please read through the tutorials to get some intuitions on what each modules are doing. We try to focus our efforts on bugs, compilation problems and math errors.

Also, your post has nothing to do with this issue "Similarity Transform leads to a zero scale".

That being said, I'm glad your trying the library with what it looks like a construction application.

alexsmartens, 2017-03-10:
Thank you.
Libpointmatcher is a great project with broad perspectives in many fields