norlab-ulaval/libpointmatcher #192: Most efficient way to check if ICP found the right transformation between two clouds

ArthurVanpoucke, 2017-01-17:
Hi !

I was wondering what was the best way to say if the step of the ICP achieved to get the "right" transformation in order to merge 2 point cloud ?

I saw here some posts, so I computed the mean distance and/or the hausdorff distance after the ICP step.

> https://github.com/ethz-asl/libpointmatcher/blob/master/examples/icp_advance_api.cpp

My guess was : if after the ICP step, two pointcloud get a distance approaching 0 the ICP went well, is that right ?

 But is this the best way ? Can i have your thoughts on this please ?

Thank you for your time :)

pomerlef, 2017-01-17:
That will most depend if you identified the outliers correctly. This might be tricky depending of your application.

ArthurVanpoucke, 2017-01-17:
I have a Velodyne VLP16 that i just want to move through a room in order to scan it. 
I also use a Inertial Measurement Unit which gives me the delta of the orientation between two scans.

In my case i could use the orientation to know if my scan went well.

And lets take the case where the outliers are removed correctly.

pomerlef, 2017-01-17:
ICP will minimize an alignment error, that how it was designed. Typical pitfalls are:
1. outliers drag the local minima somewhere else
2. the environment is under constrained (i.e., two planes can slide over each other)
3. sensor sampling is not uniform in the Euclidian space

The second point can happen often and won't be detected with a small residual error.

ArthurVanpoucke, 2017-01-23:
I understand that errors could happen.

Since my tests (with mean distance and/or the hausdorff distance ) 
I have seen that a more accurate distance can be computed with the function getResidualError. Is that correct ?
I saw it in the issue : ` https://github.com/ethz-asl/libpointmatcher/issues/154`

My problem is that the function is not known by the compiler, do i have to include some additional files ? 
And if it's possible for you, can you please write/send me an example of code/pseudo code where you compute that use the residual Error function.

Thank you :)

pomerlef, 2017-01-23:
The functions are clearely implemented (ex.: https://github.com/ethz-asl/libpointmatcher/blob/master/pointmatcher/ErrorMinimizersImpl.cpp#L120), but you need to feed some arguments.

Those functionalities were added by other users, so I cannot help much with the reste. Once I've access to more resources (human or time), I'll have to trigger a cleanup of the ErrorMinimizer files as it see to have drifted over time.

pomerlef, 2017-01-30:
Check issue #193 for an example.

closed 2017-01-30 by pomerlef