norlab-ulaval/libpointmatcher #195: Error when building  libnabo with CMake

jonntw, 2017-02-08:
I have seen some people also have this problem, but I still could not find the solution to it.
Does anyone know how to solve this problem? QQ

Here are the error messages in CMake:

 Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
Boost version: 1.63.0
OpenCL disabled, not looking for it
ANN not found, disabling it in benchmarks
FLANN not found, disabling it in benchmarks
CMake Error at tests/CMakeLists.txt:95 (message):
  Compiling a test executable failed.
Call Stack (most recent call first):
  tests/CMakeLists.txt:105 (try_compile_cloud_types)

pomerlef, 2017-02-08:
This doesn't seems to be the output of the CMake in libpointmatcher. Is that from libnabo?

If that's the case, you should report on their side. Also, you might want to use the flag `--debug-output` for a bit more details on the outputs.

jonntw, 2017-02-08:
@pomerlef ok! I will report to libnabo. Thank you!

closed 2017-03-20 by pomerlef