norlab-ulaval/libpointmatcher #198: ICP error local minima

alexsmartens, 2017-03-22:
Hi Francios,
I've played around with the libpointmatcher parameters, including
-- RandomSamplingDataPointsFilter & VoxelGridDataPointsFilter;
-- different numbers of knn for referenceDataPointsFilters;
-- different ration for TrimmedDistOutlierFilter;
-- different numbers for minDiffRotErr & minDiffTransErr.
I got the best improvement by decreasing the minDiffRotErr & minDiffTransErr to a certain point. The best results indicate my two models to be almost at the same scale and almost at the same location. Have a look at the following image 
[image]
At the first glance, the results look awesome, but I am still not able to compare these two point clouds. It seems to me that the program just finds the local minima and cannot overcome it. Have a look at the second attached picture.
[image]
The right part of the picture looks to be at the correct place, but the left part is not, it can be easily noticeable from the horizontal lines. Simple rotation around the right part of the white point cloud in meshlab set it in the correct place. But I would want to achieve the same result fully automatically

What do you think? Can you at least give me a suggestion where I can tune the settings for global minima? or any other suggestions?
Thank you!

pomerlef, 2017-03-23:
Hard to tell.

I suggest you to activate 
```
inspector:
 VTKFileInspector:
     baseFileName: pointmatcher-run1
     dumpPerfOnExit: 0
     dumpStats: 0
     dumpIterationInfo: 0 
     dumpDataLinks: 1 
     dumpReading: 1 
     dumpReference: 1 
```

This will drop a vtk file per iteration with the weights of the paired points draw as lines. You will be able to inspect which points are identified as outliers and see if your tuning is appropriate.

alexsmartens, 2017-03-25:
Thank you @pomerlef . I've managed to tune the parameters. Libpointmatcher performance is impressive

closed 2017-03-25 by alexsmartens